zigpy/bellows #273: Heiman smoke sensor don't stay paired

nerdosity, 2020-07-12:
Hello, I was a long time user of Zigate on Home assistant, first with developer custom component, then with ZHA HA integration.
I tried a more mature project so I bought a Bitronvideo BV 2010/10. It almost work natively with my installation (docker container on synology).
BUT, I have two major issue:

- it ignores my Xiaomi (https://github.com/zigpy/bellows/issues/166)
- bigger one: my two smoke sensors stay connected just an hour. Then, after that, they drop off and they never connect again automatically.
Version is SMOK_V16. 
I don't know if they are threated as a different product, since they have the "mains" power supply (false, battery powered) and they can "scream" (but in reality it's not user controllable, they just chime when there is smoke or when the test button is pressed).
How can I help?

https://pastebin.com/QmpKWgHv

Adminiuga, 2020-07-12:
Yeah, heiman does not report any activity and goes into deep sleep. In reality it doesn't really goes offline, but because it is not sending any data and does not reply to the request because of the deep sleep, ZHA considers it becoming unavailable.

Do you see any data coming in when you press test button?

nerdosity, 2020-07-12:
Yes, it became available again. But I can't trust it this way, since I don't know if it went offline or it's just in standby.

Adminiuga, 2020-07-12:
You won't be able to tell. Yes, they will be available in HA but if it really goes offline it still going to be online in zha. I'd recommend against this sensor. Alternatively, ask heinman to update firmware

closed 2020-09-04 by Adminiuga