zigpy/bellows #288: bellows2

MattWestb, 2020-08-02:
Then all is going well with the ongoing rework of the bellows, that is not a little code touch rather one master code refactoring of the old harigh code to enabling support for newer hardware and making it more compatible and flexible with old and new silabs hard and software.
Perhaps it's good to making one statement and giving it one new (updated)name so the users understand the magnitude of the work being done.
Its only one suggestion and the devs is in command.

belloes2 ?

Hedda, 2020-08-10:
If renaming bellows is now on-topic again then please see the request https://github.com/zigpy/bellows/issues/215

There I have suggested renaming the bellows library and its repository on GitHub to "zigpy-emberznet", "zigpy-ezsp", "zigpy-ember", or similar for clarity and consistency, to match all the other radio libraries for zigpy.

MattWestb, 2020-08-10:
For my the name is not important and it's up to the devs to deciding it and it can being historical / personal or some other reasons for not changing the name. 
I think its good pointing out that very soon belloes is very much refactored (not hundreds line changed more many thousands lines) to a multi version build and from my point of view its like "NG" (Next Generation) (https://github.com/zigpy/bellows/tree/multiversion).
If the devs naming it zigpy-ezsp or something else its good but in the end its not doing it better or worse and the same if they is saying with the bellows.
The most important is that the devs having the joy and being inspired then the end product is being great.

Hedda, 2020-08-10:
So your issue here is not then just in essence a duplicate of https://github.com/zigpy/bellows/issues/215 and can, therefore, be closed? I mean, why not post your original comment in https://github.com/zigpy/bellows/issues/215 instead?

@MattWestb  PS: Off-topic tip; hope you don't take offence take I would like to suggest that you install [Grammarly](https://www.grammarly.com/) to help with your spelling and grammar when writing. I'm not a native English speaker either and I think Grammarly helps me a lot!

Adminiuga, 2020-09-05:
nah, ATM there are no plans to rename the project. Too much hassle, better use the resources for development.

closed 2020-09-05 by Adminiuga

MattWestb, 2020-09-05:
I agree for 110% 👍