zigpy/bellows #328: CLI not working

pmkohn, 2020-09-01:
Hello, I just checked out latest versions of bellows and zigpy, but getting the following Traceback: 
`:~/bellows$ bellows devices
Traceback (most recent call last):
  File "/usr/local/bin/bellows", line 11, in <module>
    load_entry_point('bellows==0.20.0.dev0', 'console_scripts', 'bellows')()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2324, in load
    return self.resolve()
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/dist-packages/bellows-0.20.0.dev0-py3.6.egg/bellows/cli/__init__.py", line 2, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/dist-packages/bellows-0.20.0.dev0-py3.6.egg/bellows/cli/application.py", line 6, in <module>
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 656, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 626, in _load_backward_compatible
  File "/usr/local/lib/python3.6/dist-packages/bellows-0.20.0.dev0-py3.6.egg/bellows/config/__init__.py", line 2, in <module>
  File "/usr/local/lib/python3.6/dist-packages/zigpy-0.23.0.dev0-py3.6.egg/zigpy/config/__init__.py", line 4, in <module>
    from zigpy.config.defaults import (
  File "/usr/local/lib/python3.6/dist-packages/zigpy-0.23.0.dev0-py3.6.egg/zigpy/config/defaults.py", line 1, in <module>
    import zigpy.types as t
  File "/usr/local/lib/python3.6/dist-packages/zigpy-0.23.0.dev0-py3.6.egg/zigpy/types/__init__.py", line 2, in <module>
    from .named import *  # noqa: F401,F403
  File "/usr/local/lib/python3.6/dist-packages/zigpy-0.23.0.dev0-py3.6.egg/zigpy/types/named.py", line 53, in <module>
    class Channels(basic.bitmap32):
  File "/usr/lib/python3.6/enum.py", line 197, in __new__
    enum_member = __new__(enum_class)
TypeError: object.__new__(Channels) is not safe, use int.__new__()
`

What is wrong, is there any documention with required packages etc. that i might be missing?

Adminiuga, 2020-09-01:
system? environment?

Adminiuga, 2020-09-01:
does it work with `zigpy==0.22.2`

Adminiuga, 2020-09-01:
ah, and py37 is the minimum supported version

Adminiuga, 2020-09-05:
Latest changes in zigpy make it incompatible with py36. Py37 is the minimum required version.

closed 2020-09-05 by Adminiuga