zigpy/bellows #329: Cannot re-pair device: Skip initialization of existing device

bertouttier, 2020-09-01:
Hello,

I managed to add 2 SPZB0001 devices successfully in Home Assistant ZHA. However, I don't manage to add additional SPZB0001 devices due to this error:

```
2020-09-01 13:24:54 DEBUG (MainThread) [bellows.ezsp] Send command permitJoining: (60,)
2020-09-01 13:24:54 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'20102157540829bda37e'
2020-09-01 13:24:54 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'0310a157540815ded87e'
2020-09-01 13:24:54 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2020-09-01 13:24:54 DEBUG (MainThread) [bellows.ezsp] Application frame 34 (permitJoining) received: b'00'
2020-09-01 13:24:56 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1310b557540916b30169c208385492c4894e23f3f37e'
2020-09-01 13:24:56 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2020-09-01 13:24:56 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'2310b15754ee4d4f59944b79837e'
2020-09-01 13:24:56 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'83401b7e'
2020-09-01 13:24:56 DEBUG (MainThread) [bellows.ezsp] Application frame 35 (childJoinHandler) received: b'030158fd882d9201008d150004'
2020-09-01 13:24:56 DEBUG (MainThread) [bellows.zigbee.application] Received childJoinHandler frame with [3, <Bool.true: 1>, 0xfd58, 00:15:8d:00:01:92:2d:88, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
2020-09-01 13:24:56 DEBUG (MainThread) [bellows.ezsp] Application frame 196 (changeSourceRouteHandler) received: b'58fd000001'
2020-09-01 13:24:56 DEBUG (MainThread) [bellows.zigbee.application] Received changeSourceRouteHandler frame with [0xfd58, 0x0000, <Bool.true: 1>]
2020-09-01 13:24:57 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'3310b157540e4d4fd1b9d824aad887499d4e27ab1c447e'
2020-09-01 13:24:57 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8430fc7e'
2020-09-01 13:24:57 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received: b'58fd882d9201008d150001000000'
2020-09-01 13:24:57 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0xfd58, 00:15:8d:00:01:92:2d:88, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]
2020-09-01 13:24:57 INFO (MainThread) [zigpy.application] Device 0xfd58 (00:15:8d:00:01:92:2d:88) joined the network
2020-09-01 13:24:57 INFO (MainThread) [zigpy.device] [0xfd58] Requesting 'Node Descriptor'
2020-09-01 13:24:57 DEBUG (MainThread) [zigpy.util] Tries remaining: 2
2020-09-01 13:24:57 DEBUG (MainThread) [zigpy.device] [0xfd58] Extending timeout for 0x1b request
2020-09-01 13:24:57 DEBUG (MainThread) [bellows.zigbee.application] Extending timeout for 00:15:8d:00:01:92:2d:88/0xfd58
2020-09-01 13:24:57 DEBUG (MainThread) [bellows.ezsp] Send command setExtendedTimeout: (00:15:8d:00:01:92:2d:88, True)
2020-09-01 13:24:57 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'347d31215754549d9fcb954aa8bf559389057e'
2020-09-01 13:24:57 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'4411a1575454533c7e'
2020-09-01 13:24:57 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8520dd7e'
2020-09-01 13:24:57 DEBUG (MainThread) [bellows.ezsp] Application frame 126 (setExtendedTimeout) received: b''
2020-09-01 13:24:57 DEBUG (MainThread) [bellows.ezsp] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xfd58, <EmberApsFrame profileId=0 clusterId=2 sourceEndpoint=0 destinationEndpoint=0 options=320 groupId=0 sequence=27>, 28, b'\x1bX\xfd')
2020-09-01 13:24:57 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'45162157541e15eaa4944a27aa5592099d4e27b0f1cd7cd3007a517e'
2020-09-01 13:24:57 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'5516a157541e150f18157e'
2020-09-01 13:24:57 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8610be7e'
2020-09-01 13:24:57 DEBUG (MainThread) [bellows.ezsp] Application frame 52 (sendUnicast) received: b'00bd'
2020-09-01 13:24:59 DEBUG (MainThread) [bellows.ezsp] Send command nop: ()
2020-09-01 13:24:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'56172157542f1db97e'
2020-09-01 13:24:59 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'6617a157542fef0d7e'
2020-09-01 13:24:59 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'87009f7e'
2020-09-01 13:24:59 DEBUG (MainThread) [bellows.ezsp] Application frame 5 (nop) received: b''
2020-09-01 13:25:00 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'7617b557540916b30169c208385492c4894e2334e87e'
2020-09-01 13:25:00 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8070787e'
2020-09-01 13:25:00 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'0617b15754ee4d4f59944b460d7e'
2020-09-01 13:25:00 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2020-09-01 13:25:00 DEBUG (MainThread) [bellows.ezsp] Application frame 35 (childJoinHandler) received: b'030158fd882d9201008d150004'
2020-09-01 13:25:00 DEBUG (MainThread) [bellows.zigbee.application] Received childJoinHandler frame with [3, <Bool.true: 1>, 0xfd58, 00:15:8d:00:01:92:2d:88, <EmberNodeType.SLEEPY_END_DEVICE: 4>]
2020-09-01 13:25:00 DEBUG (MainThread) [bellows.ezsp] Application frame 196 (changeSourceRouteHandler) received: b'58fd000001'
2020-09-01 13:25:00 DEBUG (MainThread) [bellows.zigbee.application] Received changeSourceRouteHandler frame with [0xfd58, 0x0000, <Bool.true: 1>]
2020-09-01 13:25:00 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'1617b157540e4d4fd1b9d824aad887499d4e27ab13827e'
2020-09-01 13:25:00 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'82503a7e'
2020-09-01 13:25:00 DEBUG (MainThread) [bellows.ezsp] Application frame 36 (trustCenterJoinHandler) received: b'58fd882d9201008d150001000000'
2020-09-01 13:25:00 DEBUG (MainThread) [bellows.zigbee.application] Received trustCenterJoinHandler frame with [0xfd58, 00:15:8d:00:01:92:2d:88, <EmberDeviceUpdate.STANDARD_SECURITY_UNSECURED_JOIN: 1>, <EmberJoinDecision.USE_PRECONFIGURED_KEY: 0>, 0x0000]
2020-09-01 13:25:00 INFO (MainThread) [zigpy.application] Device 0xfd58 (00:15:8d:00:01:92:2d:88) joined the network
2020-09-01 13:25:00 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 00:15:8d:00:01:92:2d:88
```

I tried resetting the device multiple times in order to re-pair, but without success. Is there a cache that I need to clear somewhere? I am running Home Assistant ZHA with the ITEAD Sonoff ZBBridge (flashed with Tasmota firmware and EmberZNet 6.5.x.x)

Edit: it seems that the issue also occurs with other devices, like the IKEA control outlet.

bertouttier, 2020-09-01:
Thanks a lot @Adminiuga , that worked indeed!
Btw, I am running hassio, so I used the "custom deps deployment" addon to upgrade the bellows package, instead of using a custom component like you suggested.

closed 2020-09-01 by bertouttier

geeneo, 2020-09-10:
Can not pair with Sonoff ZigbeeZHA and Xiaomi WXKG01LM switch 

Ignoring message (b'18000a050042126c756d692e73656e736f725f6d6f74696f6e') on cluster 0: unknown endpoint or cluster id: 'No cluster ID 0x0000 on
Ignoring message (b'18030a00001801') on cluster 1030: unknown endpoint or cluster id:

Adminiuga, 2020-09-10:
Remove device. Re,- pair

geeneo, 2020-09-10:
> Remove device. Re,- pair from where... device is not present.

Adminiuga, 2020-09-10:
then re-pair. google how  pair xiaomi devices, they are super picky.

Adminiuga, 2020-09-10:
when pairing, keep the device awake.