zigpy/bellows #333: Integration in the Domoticz-Zigpy plugin

pipiche38, 2020-09-03:
I've started a project (for now its is more a Proof of Concept ) and as mentioned here https://github.com/pipiche38/Domoticz-Zigpy/issues/10 I'm trying to integrate the bellows Zigbee Radio layer .

In such I need to understand what are the parameters needed to start the Controller
In other worlds is that enought ?

```
        self.zigpyApp = await ControllerApplication.new(
            config=ControllerApplication.SCHEMA({
                "database_path": /tmp/zigpy.db,
                "device": {
                    "CONF_DEVICE_PATH":"/dev/ttyUSB0"
                }
            }),
            auto_form=True,
            start_radio=True,
        )
```

Do I need to add some additional parameters ? Does the bellows use only serial line or can it uses IP ?

pipiche38, 2020-09-03:
Sorry. I shouldn't have created.
problem solved

closed 2020-09-03 by pipiche38

Hedda, 2020-09-04:
Hopefully @Adminiuga will reply to this in https://github.com/pipiche38/Domoticz-Zigpy/issues/10 :D since pinged them with "@"