zigpy/bellows #338: [bellows] No pan_id and ext_pan_id

zoic21, 2020-09-10:
Hello,

I  just begin a dev for french home automation solution (so I'am a beginner) and I have some issue. First of all after zigbee start I can't find pan_id and ext_pan_id , when I do `application.pan_id()` result is None.

It's normal ? Are they any tips ? 

Also it's possible to retrieve ezsp version ?

Thank in advance

Adminiuga, 2020-09-10:
That is currently not implemented. However `pan_id` is different from the EZSP version. If you need EZSP version then you could call `res = await application._ezsp.get_board_info()`

to get pan-id, you would need to parse the result of `getNetworkParameters()` command, see [UG100](https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf)

e.g. `status, node_type, network_params = await application._ezsp.getNetworkParameters()`

zoic21, 2020-09-10:
Thank I didn't find `application._ezsp` tips.

It's work for `getNetworkParameters` but not for `application._ezsp.get_board_info()` I got error `get_board_info` is not in COMMANDS.

I replace by `version = await application._ezsp.getValue(0x11)` and this parsing `str(version[1][2])+'.'+str(version[1][3])+'.'+str(version[1][4])+'-'+str(version[1][6])`. 

It's seem ok.

Thank for your help

Adminiuga, 2020-09-10:
https://github.com/zigpy/bellows/blob/3b1ed934aa28039f4ba1db25688ccfbccfc2cdc9/bellows/ezsp/__init__.py#L231 it was recently added. Don't know which version are you using.

zoic21, 2020-09-10:
Perfect I move to last bellows version and it's work !!!!

Thank a lot.

closed 2020-09-10 by zoic21