zigpy/bellows #340: HA 0.115 beta Error and flakey Joining and Error

tube0013, 2020-09-10:
With Latest HA Beta (0.115b4)  I'm getting the following in logs when I trigger a join, (this is my prod system with a HUSBZB-1)

```
2020-09-10 09:59:59 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback ThreadsafeProxy.__getattr__.<locals>.func_wrapper.<locals>.check_result_wrapper() at /usr/local/lib/python3.8/site-packages/bellows/thread.py:97
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/bellows/thread.py", line 98, in check_result_wrapper
    result = call()
  File "/usr/local/lib/python3.8/site-packages/bellows/ezsp/__init__.py", line 229, in frame_received
    self._protocol(data)
  File "/usr/local/lib/python3.8/site-packages/bellows/ezsp/protocol.py", line 111, in __call__
    assert expected_id == frame_id
AssertionError
```

I was attempting to join some innr bulbs. it took a restart of HA before it would successfully join.
 Here was the log around the successfull join, 

https://paste.ubuntu.com/p/vRBwWzDm8g/

the error occurs twice, the error happen when hitting any add device button either on the main Zigbee Config panel or on a router device.

LordMike, 2020-09-13:
I get this too with HA `0.115b5`. I originally thought this was an issue with the devices, but a well-known-to-work device also failed joining. I get this error even when just running the `zha.permit` service, so not even trying to join anything.

```
2020-09-13 21:32:03 INFO (MainThread) [homeassistant.components.zha.api] Permitting joins for 60s
2020-09-13 21:32:03 DEBUG (MainThread) [bellows.ezsp.protocol] Send command addTransientLinkKey: (ff:ff:ff:ff:ff:ff:ff:ff, [90, 105, 103, 66, 101, 101, 65, 108, 108, 105, 97, 110, 99, 101, 48, 57])
2020-09-13 21:32:03 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'705d2107abd5ea4da66bb5daf03cf50bf92b66c781a706e59ea353b0420b7e'
2020-09-13 21:32:03 DEBUG (bellows.thread_0) [bellows.uart] Data frame: b'005da1f065f8df7e'
2020-09-13 21:32:03 DEBUG (bellows.thread_0) [bellows.uart] Sending: b'8160597e'
2020-09-13 21:32:03 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 88 (invalidCommand) received: b'31'
2020-09-13 21:32:03 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback ThreadsafeProxy.__getattr__.<locals>.func_wrapper.<locals>.check_result_wrapper() at /usr/local/lib/python3.8/site-packages/bellows/thread.py:97
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/bellows/thread.py", line 98, in check_result_wrapper
    result = call()
  File "/usr/local/lib/python3.8/site-packages/bellows/ezsp/__init__.py", line 229, in frame_received
    self._protocol(data)
  File "/usr/local/lib/python3.8/site-packages/bellows/ezsp/protocol.py", line 111, in __call__
    assert expected_id == frame_id
AssertionError
```

LordMike, 2020-09-13:
@Adminiuga - I noticed you perform Zigpy/bellows updates on HA, and contribute to those repos as well - do you have an idea about what could be wrong here?

Adminiuga, 2020-09-14:
Yeah, apparently the firmware on husbzb 1 is way too old.
I'll revert/refactor those changes for the older protocol versions.

LordMike, 2020-09-14:
Wow.. Should I consider a different device, like smartthings hub or one of the CCxxx chips?

Hedda, 2020-09-14:
@LordMike Checkout the Sonoff ZBBridge https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

Elelabs current generation devices have slightly older EFR32 MG1 chip but is still newer than HUSBZB-1 https://elelabs.com/shop/

Elelabs ships with older firmware but can be updated with their tool here https://github.com/Elelabs/elelabs-zigbee-ezsp-utility

Then there is the DIY approach with IKEA TRÅDFRI ICC-A-1 Module (also currently have a slightly older EFR32 MG1 chip) 

https://github.com/MattWestb/IKEA-TRADFRI-ICC-A-1-Module

Some more newer Silicon Labs EFR32 MG2 chip USB sticks are rumoured to become available on the market this fall and winter. 

The best alternative options is currently CC2652 with Z-Stack 3 firmware

https://www.tindie.com/products/slaesh/cc2652-zigbee-coordinator-or-openthread-router/

https://www.tindie.com/products/electrolama/zzh-cc2652r-multiprotocol-rf-stick/

dmulcahey, 2020-09-14:
You can update the firmware on the stick too.

Hedda, 2020-09-14:
> You can update the firmware on the stick too.

But isn't it true that only Silicon Labs EFR32 MG1 (EFR32MG1) and EFR32 MG2 (EFR32MG2) supports EmberZNet 6.7 based firmware and later which is need EZSP v8 support?

SiLabs Ember EM35x, EM358x, and EM359x device families (e.g. EM357 like the hardware chip inside the HUSBZB-1) and older generations that it does not support the very latest EmberZNet NCP application / SDK / firmware? And the same goes with Silicon Labs ETRX357 based adapters like Telegesis ETRX357USB?

If so that means it's practically impossible to get HUSBZB-1 to support very latest EZSP protocol even with slightly newer firmware?

tube0013, 2020-09-14:
> You can update the firmware on the stick too.

yes @walthowd has great instructions for updating the FW. I may go the route if migrating to a new coordinator..

https://github.com/walthowd/husbzb-firmware

Hedda, 2020-09-14:
> > You can update the firmware on the stick too.
> 
> yes @walthowd has great instructions for updating the FW. I may go the route if migrating to a new coordinator..
> 
> https://github.com/walthowd/husbzb-firmware

Firmware mentioned there is EmberZNet 6.6.3 (EmberZNet 6.6) so that would imply EZSP Protocol Version 7.

FYI, EmberZNet 6.7.0 (EmberZNet 6.7) added EZSP Protocol Version 8 (and Secure EZSP Protocol Version 2).

PS: The EZSP version changed from V4 to V5 in ZNet 5.9 and to EZSP V6 in ZNet 6.0, then from EZSP v6 to EZSP v7 in ZNet 6.4

Adminiuga, 2020-09-14:
EmberZNet 6.6 is fine to be used as a coordinator firmware. There were no changes in between EZSP protocol versions 6, 7, 8 offering a "must have" killer features. 
In other words, go ahead and update your HUSBZB-1 to EmberZNet 6.6.3. 

@LordMike if you are comfortable, follow WaltH guide on HUSBZB firmware upgrade. Otherwise the options are:
- Get Eleleabs ELU-13 stick -- uses EZSP, same family, much better hardware
- Don't get CC253x based sticks. You'd want something more modern, like [ZZH](https://electrolama.com/projects/zig-a-zig-ah/) or [Slaesh's stick](https://www.tindie.com/products/slaesh/cc2652-zigbee-coordinator-or-openthread-router/)

tube0013, 2020-09-14:
there are a lot of HUSBZB-1 users, upgrading FW or getting a new stick is great for me or other power users, but for the masses a fix for the older FW on the HUSBZB-1 is probably best. other wise 0.115 will be a breaking change for ZHA on HUSBZB-1.

Adminiuga, 2020-09-14:
it is going to be fixed for other users

closed 2020-09-14 by Adminiuga

LordMike, 2020-09-14:
@Adminiuga great work :)

I'm upgrading the firmware now. Thanks for all the suggestions, and especially for the pressing question of "is there anything I'm missing by not using something else" .. and the note on CCxx sticks.. - those answers aren't easily found.

As a curiosity, the upgrade scripts tells me I'm running `5.4.1-194` right now.

Hedda, 2020-09-17:
> * Don't get CC253x based sticks. You'd want something more modern, like [ZZH](https://electrolama.com/projects/zig-a-zig-ah/) or [Slaesh's stick](https://www.tindie.com/products/slaesh/cc2652-zigbee-coordinator-or-openthread-router/)

>  .. and the note on CCxx sticks.. - those answers aren't easily found.

Hopefully, the PR https://github.com/home-assistant/home-assistant.io/pull/14430 makes the difference between just a little clearer.

Once merged that will at least update the ZHA page at HA https://www.home-assistant.io/integrations/zha/

PR #344 (2020-09-14): Add Global TC link key as a transient link key only for EZSP v5 or later
Add Global TC link key as a transient link key only for EZSP v5 or later.
HUSBZB-1 comes with EmberZNet 5.4.1.0 based firmware and does not support transient link key command. 

Fixes #340