zigpy/bellows #347: [DOCUMENTATION] Links to how-to upgrade firmware on Telegesis ETRX357USB adapters?

Hedda, 2020-09-15:
Does someone have links to good information on exactly how-to upgrade firmware on Telegesis ETRX357USB adapters?

Plus links to publicly available firmware images with newer EmberZNet NCP application FW image files for them?

@walthowd has posted great instructions https://github.com/walthowd/EZSP-Firmware

PS: Submitted PR for updated README.md with a recommendation to upgrade EmberZNet firmware in https://github.com/zigpy/bellows/pull/346

PPS: I am asking this on behalf of the community as I do not actually own any Telegesis ETRX357 USB dongles myself.

Hedda, 2020-09-15:
Telegesis ETRX357 procedure for reference (with unknown firmware image):

https://community.home-assistant.io/t/eu-usb-sticks-for-the-new-zigbee-component/16718/10

> I can confirm that the ETRX357USB-LRS+8M can be flashed with an appropriate EZSP firmware without the use of any programming hardware.
> 
> So, without any guarantees (you might -but probably won’t- brick your device), here’s how:
> 
> Download the firmware. I used some random blob someone put on github that at least seemed to have the right name. What could possibly go wrong, right? :smile: https://github.com/yqyunjie/Zigbee-Project/blob/master/firmware/EmberZNet/EM35x-EZSP/build/em35x-ezsp-images/EM357/em357-ncp-uart-xon-xoff-use-with-serial-uart-bl-500.ebl?raw=true 311
> 
> (Install USB-to-serial drivers for the device. Linux will load the driver automatically when you plug in; not sure about other OSes.)
> 
> Install a serial port communication app that supports X-MODEM. (debian/ubuntu: sudo apt-get install minicom) Run it and configure it (sudo minicom -s) to use the fake serial port (/dev/ttyUSB0) at 19200 baud 8N1. Disable both hardware (RTS/CTS) and software (XON/XOFF) flow control.
> 
> Type ‘AT’ , you should get OK in response. Now type ‘AT+BLOAD’ . The device will reboot into the bootloader.
> 
> Change the baud rate in the serial port communication app setting to 115200 baud. (exit minicom using CTRL-A Q, run ‘sudo minicom -s’ again)
> 
> Pressing enter in the terminal should now show you a three-option boot loader menu. Choose option 1.
> 
> ‘C’ characters will start showing. Don’t wait for this to finish, but start an X-MODEM upload of the firmware you downloaded earlier (use CTRL-A S in minicom). You have 60 seconds to start the upload.
> 
> After the upload finished, you should return to the menu. Now select option 2, to reboot into the new firmware. You’re done.
> 
> Enjoy!

Hedda, 2020-09-16:
References on how-to upgrade EmberZNet NCP application firmware an some other Silicon Labs EM35x series Zigbee SOCs:

https://github.com/walthowd/husbzb-firmware/

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility

macfly92, 2020-11-23:
Hi, this is my report on my recent experience :

My first shot was to follow the HowTo you've quoted in your second post (with unknown firmware image). It worked well and led me on EmberZNet version: 5.0.0.0 build 95 on the dongle.

@Adminiuga gave me a link to an updater for HUSBZB-1 to try to find a newer firmware for my dongle ( https://www.github.com/walthowd/husbzb-firmware/tree/master/Dockerfile) and I've found it :
Download http://developer.silabs.com/studio/v4/control/stacks/PrivateGA/updates/binary/com.silabs.stack.znet.v6.4.feature_root_6.4.1.0 and extract the file :
developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/em357/ncp-uart-xon-xoff-use-with-serial-uart-btl-6.4.1.ebl

Use this file in place of the older one in the HowTo.

With this file the EmberZNet version is : 6.4.1.0 build 408

For me, after the first flash, I couldn't manage to get Serial AT command anymore, so I've adapted the Dockerfile to perform the firmware update of the ETRX357. This is what I've done :

Dockerfile :

```
FROM ubuntu:18.04

WORKDIR /tmp/silabs

RUN apt-get update \
  && apt-get install -y git wget python-pip python3-pip unzip jq qemu-user-static

RUN pip install xmodem pyserial

RUN mkdir -p /tmp/silabs

# Get firmware
RUN wget http://developer.silabs.com/studio/v4/control/stacks/PrivateGA/updates/binary/com.silabs.stack.znet.v6.4.feature_root_6.4.1.0
RUN unzip -p com.silabs.stack.znet.v6.4.feature_root_6.4.1.0 developer/sdks/gecko_sdk_suite/v2.4/protocol/zigbee/ncp-images/em357/ncp-uart-xon-xoff-use-with-serial-uart-btl-6.4.1.ebl > ncp-uart-xon-xoff-use-with-serial-uart-btl-6.4.1.ebl
RUN rm -f /tmp/silabs/com.silabs.stack.znet.v6.4.feature_root_6.4.1.0

# Get ncp.py
RUN wget http://devtools.silabs.com/solutions/apt/pool/main/s/silabs-zigbee-gateway/silabs-zigbee-gateway_2.5.0-3_armhf.deb
RUN ar x /tmp/silabs/silabs-zigbee-gateway_2.5.0-3_armhf.deb
RUN tar -xvf data.tar.gz ./opt/siliconlabs/zigbeegateway/tools/ncp-updater/ncp.py --strip-components 6 -C /tmp/silabs/
RUN rm -f silabs-zigbee-gateway_2.5.0-3_armhf.deb
RUN rm -f *.gz
RUN rm -f /tmp/silabs/debian-binary

CMD /bin/bash
```

Build the Docker container with :

`docker build -t updatetelegesis .`

Then run the container with the appropriate /dev path :

`docker run --rm -it device=/dev/ttyUSB0 updatetelegesis `

and, finally, perform the flash :

`python ncp.py flash -p /dev/ttyUSB0 -f ncp-uart-xon-xoff-use-with-serial-uart-btl-6.4.1.ebl`

If someone have a newer firmware, please provide it !
Thanks you.

walthowd, 2020-11-24:
@macfly92 Nice job! If you look around on Github, you can find some of the newer releases:

https://github.com/search?q=em357+ncp-uart-xon-xoff-use-with-serial-uart-btl+ncp-images+in%3Apath&type=Code

But flash with caution, I locked myself out of my EM3581 this way by flashing random firmware.

Adminiuga, 2020-11-24:
@macfly92 send me a message on HA discord server

macfly92, 2020-11-24:
Thanks you @walthowd for your dockerfile :) your commented lines were very helpfull.

@Adminiuga I've sent you a message on Discord.

walthowd, 2020-11-24:
Did some testing with @macfly92 and added support for the ETRX357USB to be flashed with the docker image. Modified ncp.py to talk to these sticks and added the 6.6.5 (EZSP7 NCP firmeware) 

https://github.com/walthowd/husbzb-firmware

closed 2020-12-17 by Adminiuga