zigpy/bellows #348: [DOCUMENTATION] Links to how-to upgrade firmware on Bitron Video/Smabit BV AV2010/10 USB-Stick?

Hedda, 2020-09-15:
Does anyone have links to information on exactly how-to upgrade firmware on Bitron Video/Smabit BV AV2010/10 USB-Stick?

Plus links to publicly available firmware images with newer EmberZNet NCP application FW image files for them?

@martin3000 added the Bitron Video Zigbee Stick to Home Assistant ZHA list https://www.home-assistant.io/integrations/zha/

Would be good to add links to step-by-step in bellows README.md file https://github.com/zigpy/bellows/blob/master/README.md

- Bitron Video/Smabit BV AV2010/10 USB-Stick with Silicon Labs Ember 3587

@walthowd has posted great instructions https://github.com/walthowd/EZSP-Firmware

PS: Submitted PR for updated README.md with a recommendation to upgrade EmberZNet firmware in https://github.com/zigpy/bellows/pull/346

PPS: I am only asking this on behalf of the community as I do not actually own any Bitron Video/Smabit USB dongles myself.

martin3000, 2020-09-15:
The official manuals page is https://manuals.smabit.eu/av2010_281.html but there is nothing written about updates. The stick is sold by german Telekom for their "magenta smarthome" system.

Hedda, 2020-09-15:
@martin3000 as I understand it, if it comes with a standard Ember bootloader then it should be possible for anyone with access to the EmberZNet PRO SDK to create an EmberZNet NCP application firmware for it which then can be upgraded/flashed via XMODEM.

@walthowd has for example posted great instructions on upgrading HUSBZB-1 https://github.com/walthowd/husbzb-firmware

The main problem with that is that there needs to be a personal interest to first create such firmware for your specific device as you really need to own one in order to figure out the GPIO pin layout (pins design, e.g. TX and RX port) that the chip uses before compiling the firmware so that the EmberZNet NCP application will communicate using the correct TXD and RXD ports.

https://www.silabs.com/community/wireless/zigbee-and-thread/forum.topic.html/how_to_use_the_ember-U1JQ

Also, to create and compile firm you need access to the EmberZNet PRO SDK through the Simplicity Studio software from Silicon Labs you need to have bought and registered an EFR32 Development Kit, like for example the EFR32xG22 Wireless Gecko Starter Kit:

https://www.cnx-software.com/2020/06/25/efr32xg22-wireless-gecko-starter-kit-targets-bluetooth-zigbee-and-proprietary-wireless-development/

This is basically the process what has been done for some other Silicon Labs devices, like; HUSBZB-1, Telegesis ETRX357USB, and Sonoff ZBBridge, which all have EmberZNet NCP application EZSP firmware available for them release by independent developers.

Hedda, 2020-09-15:
I don't speak german but it looks like at least some AV2010/10 ship with "Firmware 5.8.1" which could refer to old EmberZNet 5.8.1

https://telekomhilft.telekom.de/t5/Smart-Home/Neue-Osram-Lampe-angelernt-gt-Keine-Zigbee-Geraete-mehr/td-p/2996035

If that is the case then that in indeed shipped with very old firmware (similar to HUSBZB-1 which some even ship with old 5.7 FW).

Hedda, 2020-09-15:
> The official manuals page is https://manuals.smabit.eu/av2010_281.html but there is nothing written about updates. The stick is sold by german Telekom for their "magenta smarthome" system.

Do you happen to know the inside of these contain the same chip and firmware as Qivicon Funkstick for Magenta Smart Home?

https://www.expert.de/shop/unsere-produkte/computer-zubehor/netzwerk/smart-home/17790001805-smart-home-zigbee-stick-fuer-home-base.html

https://www.conrad.com/p/99922197-magenta-smarthome-usb-memory-stick-1384483

Someone did, however, comment a long time ago on OpenHAB that Bitron Video AV2010/10 works but a Qivicon Funkstick did not.

https://community.openhab.org/t/zigbee-dongle-compatible/10249/20

Their outside shell/enclosure looks different but many OpenHAB comments mention both those as "Funkstick" for "Qivicon".

Could of course be that they have the same chip but similar to Telegesis ETRX357USB it will not work unless flash a other FW.

Oh, someone did open up a Qivicon Zigbee Funkstick branded dongle and looks to be same hardware as Telegesis ETRX357

[image]

martin3000, 2020-09-15:
[image]
[image]
[image]
[image]
[image]
[image]

martin3000, 2020-09-15:
If someone builds a firmware, I'll buy a second stick and try it out.

Hedda, 2020-09-15:
Have you tried to connect to the Bootloader via XMODEM? That is probably a prerequisite before even building a firmware for it.

martin3000, 2020-09-15:
Using minicom, I do not find the correct baud rate. I only get strange characters.

Hedda, 2020-09-15:
Maybe start by testing the docker image that @walthowd provides an environment to see if can connect to check firmware info.

https://github.com/walthowd/husbzb-firmware

You do not need to flash the adapter (as the firmware provided is probably not right for chip)

https://github.com/zigpy/bellows/blob/dev/README.md

> It is worth noting that EM3588 devices that have an embedded USB core will likely work with any baud rate, where dongles using external USB interface (eg CP2102 used with an EM3581) will likely require a specific baud rate. Currently there are two main NCP images - one that supports hardware flow control with a baud rate of 115200, and one that supports software flow control with a rate of 57600.

From your pictures it looks like it has a CP2104 USB to Serial / UART chip from Silicon Labs so make sure have drivers for it

https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

https://www.silabs.com/community/interface/knowledge-base.entry.html/2016/12/30/downloading_cp210xd-ek07

Maybe try software that supports software "flow control" (try with and without) and also try both rate of 57600 and 115200 bps.

https://www.silabs.com/community/wireless/zigbee-and-thread/knowledge-base.entry.html/2012/06/29/getting_started_with-ZS9o

The default baud rate of the EZSP UART NCP application is as follows:

115,200 bps when using hardware flow control (RTS/CTS)
57,600 bps when using software flow (XON/XOFF)

"NOTE: Please be aware that hardware flow control is strongly recommended for the use of EZSP-UART, so if your hardware doesn't implement this, then you should use the Xon/Xoff (software flow control) version of the EZSP-UART firmware, or else make sure to keep the baud rate quite low (38400 or lower) to reduce the chances of communication errors."

> **HUSBZB-1 Firmware Recovery**
> 
> In the event of a bad flash or unexpected event, the bootloader for the EM3581 on the HUSBZB-1 can be accessed by resetting the stick and shorting TP17 to GND with a a serial connection (115200 8/N/1 no hw or sw flow control). On device startup, unshort TP17 and send a carriage return over the serial connection. You should be returned to the bootloader menu where a image can be uploaded via XMODEM.

If can probably trigger Recover / Bootload mode (Bootloader) by resetting the stick and shorting a specific pin to GND

https://github.com/walthowd/husbzb-firmware

Worse case you need to connect something like a JTAG debug adapter like a SEGGER J-Link to manually trigger Bootload on a GPIO

https://www.silabs.com/documents/public/application-notes/an706-ezsp-uart-host-interfacing-guide.pdf

Hedda, 2020-09-15:
Telegesis ETRX357 procedure for reference (with unknown firmware image):

https://community.home-assistant.io/t/eu-usb-sticks-for-the-new-zigbee-component/16718/10

> I can confirm that the ETRX357USB-LRS+8M can be flashed with an appropriate EZSP firmware without the use of any programming hardware.
> 
> So, without any guarantees (you might -but probably won’t- brick your device), here’s how:
> 
> Download the firmware. I used some random blob someone put on github that at least seemed to have the right name. What could possibly go wrong, right? :smile: https://github.com/yqyunjie/Zigbee-Project/blob/master/firmware/EmberZNet/EM35x-EZSP/build/em35x-ezsp-images/EM357/em357-ncp-uart-xon-xoff-use-with-serial-uart-bl-500.ebl?raw=true 311
> 
> (Install USB-to-serial drivers for the device. Linux will load the driver automatically when you plug in; not sure about other OSes.)
> 
> Install a serial port communication app that supports X-MODEM. (debian/ubuntu: sudo apt-get install minicom) Run it and configure it (sudo minicom -s) to use the fake serial port (/dev/ttyUSB0) at 19200 baud 8N1. Disable both hardware (RTS/CTS) and software (XON/XOFF) flow control.
> 
> Type ‘AT’ , you should get OK in response. Now type ‘AT+BLOAD’ . The device will reboot into the bootloader.
> 
> Change the baud rate in the serial port communication app setting to 115200 baud. (exit minicom using CTRL-A Q, run ‘sudo minicom -s’ again)
> 
> Pressing enter in the terminal should now show you a three-option boot loader menu. Choose option 1.
> 
> ‘C’ characters will start showing. Don’t wait for this to finish, but start an X-MODEM upload of the firmware you downloaded earlier (use CTRL-A S in minicom). You have 60 seconds to start the upload.
> 
> After the upload finished, you should return to the menu. Now select option 2, to reboot into the new firmware. You’re done.

martin3000, 2020-09-15:
stty /F /dev/ttyUSB0 tells mit it is 57600 with xon/xoff, but connection via minicom, I do not get an answer from AT command.

Hedda, 2020-09-16:
That probably means that they hardcoded the EmberZNet NCP application firmware to the baud rate of 57600 bps in that image.

Note that the bootloader might still have different baud rate if it comes to having to force reset bootload for firmware recovery.

So, in summary, sounds like it has basically same hardware chips on the board as Telegesis ETRX357USB but different firmware?

If want to risk it then you could now go ahead and try flashing the same firmware image as they use for Telegesis ETRX357USB, but as you mentioned that would likely mean that you would have to first buy another adapter to experiment on and lab with.

If you can flash that same firmware image as Telegesis ETRX357USB then could request a new image for it and this AV2010/10.

martin3000, 2020-09-16:
Elelabs_EzspFwUtility.py works a little bit, but with minicom I do not get the "OK" answer. So how to do the firmware update? Maybe I could try it with the elelabs tool.

Hedda, 2020-09-16:
> Elelabs_EzspFwUtility.py works a little bit, but with minicom I do not get the "OK" answer. So how to do the firmware update? Maybe I could try it with the elelabs tool.

> stty -F /dev/ttyUSB0 tells me it is 57600 with xon/xoff, but connecting via minicom, I do not get an answer from AT command.
> 
> After some modification to the tool, python3 Elelabs_EzspFwUtility.py probe -p /dev/ttyUSB1 -b 57600 returns firmware 5.8.1, EZSP v4

Yes you can try flashing with Elelabs tool as long as are aware that successfully flashing the wrong firmware might corrupt EmberZNet NCP application so that you then have to manually reset it to force it into bootload recovery mode to load the Bootloader. The firmware images that they provide is however not for EM357x series so need to try some other firmware image.

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility

If you are willing to risk it then could go ahead and try that same firmware image as the community use for Telegesis ETRX357:

https://community.home-assistant.io/t/eu-usb-sticks-for-the-new-zigbee-component/16718/10

em357-ncp-uart-xon-xoff-use-with-serial-uart-bl-500.ebl image file from:

https://github.com/yqyunjie/Zigbee-Project/blob/master/firmware/EmberZNet/EM35x-EZSP/build/em35x-ezsp-images/EM357/

As mentioned, before you might have to change the baud rate in the serial port communication to 115200 baud for Bootloader.

If that is successful then you can try asking the community if someone is willing to make a newer such image for EM357 using a later EmberZNet PRO stack that then is used by both these AV2010/10 USB-sticks and other Telegesis ETRX357USB products.

Even if you do flash the wrong EmberZNet NCP application firmware image then that should still be recoverable as the Bootloader firmware is separate and if you load/start that then you can try flashing an other EmberZNet NCP application firmware image, though if the EmberZNet NCP application firmware image has been corrupted then you need to force it into bootload recovery mode to make it reset and load/start the Bootloader (to do that you usually need to short-circuit a specific pad on the board with a gnd/ground-pad on the same board).

The normal worst-case scenario is probably if you manage to corrupt the Bootloader firmware too and in that case, you might need to connect something like a JTAG debug adapter like a SEGGER J-Link to flash a new Bootloader to recover it back to that. You should, of course, be warned that it is possible to fully brick adapters like these but it is unusual as they are designed to be recoverable with the right hardware, software tools, and expertise.

Hedda, 2020-09-16:
Again, could also try docker environment and instructions by @walthowd meant to HUSBZB-1 (Silicon Labs EM3581 / EM358)

https://github.com/walthowd/husbzb-firmware

https://github.com/walthowd/husbzb-firmware/blob/master/README.md

Hedda, 2020-09-16:
You could try with that "ncp-uart-sw-6.6.5.ebl" firmware file for husbzb-1 (EM3581) and hope that the pin layout matches AV2010/10

https://github.com/walthowd/husbzb-firmware/blob/master/ncp-uart-sw-6.6.5.ebl

https://github.com/walthowd/husbzb-firmware

martin3000, 2020-09-16:
The page https://www.texim-europe.com/getfile.aspx?id=101076 says:
> Telegesis ETRX357 AT Command Interface is not yet available for use with ETRX3587 modules
Maybe this means that the AT interface is not available on the EM3587.

Hedda, 2020-09-17:
> The page https://www.texim-europe.com/getfile.aspx?id=101076 says:
> 
> > Telegesis ETRX357 AT Command Interface is not yet available for use with ETRX3587 modules
> 
> Maybe this means that the AT interface is not available on the EM3587.

The interface is provided by the firmware, so if it is not in the firmware then no, if it is in the firmware then yes.

So likely the firmware on AV2010/10 does not have AT interface but then it should have some other interface.

martin3000, 2020-09-18:
I asked Smabit (Bitron Video) about firmware updates. They told me that updates are done automatically if you are using the stick together with the Magenta system. This means that there must be server hosting the firmware and that there are sticks with the up-to-date firmware.

Hedda, 2020-09-21:
@martin3000 OK, but just because they say that they release firmware updates does not mean that Smabit (Bitron Video) as the manufacturer releases major upgrades to much newer firmware. It might only mean that they release updates to an older version. It is common for manufacturers to stick with older firmware versions for historical reasons or they have no monetary gain in upgrading.

As you noted the firmware shipped with your own EM3587 / ETRX3587 adapter was EmberZNet 5.8.1 which is really old today.

Smabit (Bitron Video) as the manufacturer should if they wanted to be able to upgrade it to the latest EmberZNet 6.5.x.x and stay backwards compatible with EZSP v4 protocol plus with that version make it also be compatible with EZSP v5, EZSP v6, and EZSP v7.

EM3587 chip inside it does even support EmberZNet SDK 6.6.x.x but Smabit (Bitron Video) as the manufacturer might be more reluctant to upgrade to that EmberZNet version as it removes EZSP v4 backwards compatibility which would mean that it would also require all supported applications like Magenta SmartHome Telekom to also add support for EZSP v5, EZSP v6, or EZSP v7.

Again, independent developers who owns an EFR32 Development Kit, like for example the EFR32xG22 Wireless Gecko Starter Kit,  as as such has registered access to EmberZNet PRO SDK in Silicon Labs Simplicity Studio could also build EmberZNet 6.5.x.x and/or EmberZNet SDK 6.6.x.x firmware for EM3587 but they would need to know specific pins layout AV2010/10 USB-sticks.

https://github.com/zigpy/bellows/blob/dev/README.md

### _EmberZNet and EZSP Protocol Version_

_Silicon Labs do not currently have a consolidated list of changes by EmberZNet SDK or EZSP protocol version. The EZSP additions, changes and deletions have only ever been listed in the "Zigbee EmberZNet Release Notes" (EmberZNet SDK) under the "New items" section as well as the matching UG100 EZSP Reference Guide included with each EmberZNet SDK release._

_The largest change was between EZSP v4 (first added in EmberZNet 4.7.2 SDK) and EZSP v5 that was added in EmberZNet 5.9.0 SDK which requires the Legacy Frame ID 0xFF. The change from EZSP v5 to EZSP v6 was done in EmberZNet 6.0.0 SDK. The change from EZSP v6 to EZSP v7 was in EmberZNet 6.4.0 SDK. EmberZNet 6.7.0 SDK added EZSP v8 (and Secure EZSP Protocol Version 2)._

_Perhaps more important to know today is that EZSP v5, v6 and v7 (EmberZNet 6.6.x.x) use the same framing format, but EmberZNet 6.7.x.x/EZSP v8 introduced new framing format and expanded command id field from 8 bits to 16 bits._

martin3000, 2020-09-24:
I have a new stick for experiments.
I tried 
`python3 Elelabs_EzspFwUtility.py flash -f ELX0X3_MG13_6.0.3_ezsp_v6.gbl  -p /dev/ttyUSB0 -b 57600`
but it did not work. it is the wrong firmware file.

Hedda, 2020-09-25:
Suggest maybe post in https://github.com/zigpy/bellows/issues/243 and nicely there for someone to compile a new firmware for EM3587.

Best would be if you would first read the specification of the CP2104 USB to UART chip so know which pins are TX/TXD and RX/RXD on the serial/UART side, and then trace those on the circuit board to figure which pins on the EM3587 chip are used for TX/TXD and which is RX/RXD. Then you can read specification EM3587 chip to know which paths the firmware needs to be compiled for.

Hedda, 2020-11-25:
@martin3000 maybe talk to @walthowd and @macfly92 as they figured out a way to flash Telegesis ETRX357 adapters in https://github.com/zigpy/bellows/issues/347

martin3000, 2020-11-25:
I have the flash tool and it should work with the right firmware file.

maggu2810, 2020-12-31:
I own an AV2010/10 mysel, too.
I received an OVP yesterday. Not used myself yet.
Has there been any progress already?

martin3000, 2020-12-31:
> I received an OVP yesterday. Not used myself yet.

What is an OVP?

maggu2810, 2020-12-31:
Sorry! That is a german abbreviation.
I wanted to stay that it has been oroginal packaged and not opened before. So a new / unused one.

Hedda, 2021-01-22:
Did you guys manage to flash your Bitron Video / SMaBiT / Telekom Magenta USB-sticks based on Silicon Labs EM358x hardware?

FYI, there is a lot of recent discussion in the last week about EM35x firmware updates in https://github.com/walthowd/husbzb-firmware/issues/12

@grobasoz has uploaded new firmware images for both EM357 and EM358x in his repository here:

https://github.com/grobasoz/zigbee-firmware

https://github.com/grobasoz/zigbee-firmware/tree/master/EM358x

https://github.com/grobasoz/zigbee-firmware/tree/master/EM357

Try it at your own risk as this could brick the stick, though that might be a good thing as would be excuse to buy a new dongle ;)

PS: If you do manage to flash it then please update the documentation for any firmware tool and/or guide that you use

https://github.com/walthowd/husbzb-firmware

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/

Adminiuga, 2021-01-22:
AFAIK Bitron accepted standard builds which came with older emberznet versions, this means this firmware is not going to work. At least it doesn't work with HUSBZB-1

MattWestb, 2021-01-22:
@Adminiuga Have you trying to load the 6.7.8.0 for EM3581 / HUSBZB-1 and its not working ?

One user have loading the EM357 on Telegesis ETRX357USB and its loaded OK and looks working OK.

I think the EM358X its little more tricky then its one combo device.

MattWestb, 2021-02-13:
Its looks like the firmware cooker (Gary) have solving the problem with HUSBZB-1 and have getting EZSP 6.7.8.0 up and running on HUSBZB-1.

One thing with 6.7.X.X its using only the new protocol version 8 and is **not** backwards compatible with older protocols pre V8.

If its possible flashing one earlier 6.6.X.X EZSP it should being pretty secure trying the 6.7.8.0 and is its not working only flashing back the working 6.6.X.X.

ster1um, 2021-07-29:
I not understand,  it's possible upgrade the firmware on Bitron Video AV2010/10?

on this repository "https://github.com/grobasoz/zigbee-firmware/tree/master/EM358x" result deprecated...

MattWestb, 2021-07-29:
Siirlabs have flagged EM35X and EFR32 devices with 256 kb flash as deprecated and also most code for EM35X devices is deleted in the EZSP 6.9.2.x release.
[Zigbee EmberZNet SDK 6.9.2.0 GA](https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.9.2.0.pdf) 

But our master cooker have doing one working ESZP 6.7.8.0 for EF35X devices and also one for EM3581.

Some is saying the EM3581 shall working on EM3587 but no user have successful trying flashing it with our EM3581 firmware. One reason is we dont have any firmware to going back if it is flashed and not working (no have dumping the original firmware or getting one working upgrade file).

Feel free trying it but no one is leaving and grantee and you can braking your device if you not have (if its possible) dumped the original firmware before you is doing it and can re flashing it with J-Tag / SWD debugger.

MattWestb, 2021-07-29:
EZSP 6.7.X is not having any recompiled NCP for EM3587 only for EM3588 but i have "finding" 6.5.0.0 and 6.6.6.0 and attaching HW and SW versions in EBL and S37 format if some have use of them then flashing there device.
[ 6.5.0GA.zip](https://github.com/zigpy/bellows/files/6902840/6.5.0GA.zip)
[6.6.6.0GA.zip](https://github.com/zigpy/bellows/files/6902842/6.6.6.0GA.zip)

ster1um, 2021-07-30:
Ok thansks for a lot information!
seeing the situation I don't buy this dongle... 
I will wait August for buy the itead zigbbe efr2. 
From Itead confirmed to return the item in stock next month.

MattWestb, 2021-07-30:
If its the Itead / Sonoff USB Zigbee 3 dongle I can not recommend then its having large problems with the hardware and is barely working at all (the same problem with ZBB but worth).

Look in HA continuity forums and also Zigpy and bellows issues for getting info on Sonoff FER32 coordinators.

They is also releasing one EFR32 with Ethernet that looks OK on the Ethernet side (ESP32) but is using the same bad Zigbee module (SM-011) that making all one **NO GO!!** https://github.com/zigpy/zigpy/discussions/635#discussioncomment-976568

martin3000, 2021-07-30:
I would recommend cc2652r or cc2652rb or cc2652p

ster1um, 2021-08-01:
> If its the Itead / Sonoff USB Zigbee 3 dongle I can not recommend then its having large problems with the hardware and is barely working at all (the same problem with ZBB but worth).
> 
> Look in HA continuity forums and also Zigpy and bellows issues for getting info on Sonoff FER32 coordinators.
> 
> They is also releasing one EFR32 with Ethernet that looks OK on the Ethernet side (ESP32) but is using the same bad Zigbee module (SM-011) that making all one **NO GO!!** [zigpy/zigpy#635 (comment)](https://github.com/zigpy/zigpy/discussions/635#discussioncomment-976568)
Ok thanks

MattWestb, 2021-11-24:
Looks like Gary have cooked one working EZSP 6.7.8.0 by mistake that is working with AV2010/10 USB-Stick https://github.com/grobasoz/zigbee-firmware/issues/15.

I have not testing but 2 user have flagged its working.

Hedda, 2021-12-16:
> Looks like Gary have cooked one working EZSP 6.7.8.0 by mistake that is working with AV2010/10 USB-Stick [grobasoz/zigbee-firmware#15](https://github.com/grobasoz/zigbee-firmware/issues/15).
> 
> I have not testing but 2 user have flagged its working.

Nice! Also see new comment in https://github.com/grobasoz/zigbee-firmware/issues/15 that a user now managed to flash with walthowd docker image too.

martin3000, 2021-12-16:
something offtopic: are there modules which allow the creation of DIY zigbee sensors?

Hedda, 2021-12-16:
> something offtopic: are there modules which allow the creation of DIY zigbee sensors?

@martin3000 Better post such topics and new discussions here instead -> https://github.com/zigpy/zigpy/discussions

But here short summary as answer; yes there are. Texas Instruments CC2530 based radio modules from Ebyte and RF-star was most popular for DIY project until recently when more hobbyist started moving over to their newer CC2652 series (CC2650R or CC2652P). 

Examples modules are RF-BM-2652P1 and RF-BM-2652P2 modules from RF-Star, and E72-2G4M20S1E module from Ebyte.

What drives most those projects Zigbee Configurable Firmware tool by PTVO -> https://ptvo.info as it allows just about anyone to build firmware for Texas Instruments based radio modules -> https://ptvo.info/zigbee-configurable-firmware-features/ and that recently also added support for CC2652 and CC1352:

https://ptvo.info/zigbee-switch-configurable-firmware-v2-210/

https://ptvo.info/cc2652r1-cc2652rb-cc2652p-cc1352p2-configurable-router-firmware-748/

I think the most active community around such DIY Zigbee products is diyruz https://github.com/diyruz/ which I believe post most on https://modkam.ru/ and they have several open source projects posted for different Zigbee sensors.

Hedda, 2022-01-03:
> > Looks like Gary have cooked one working EZSP 6.7.8.0 by mistake that is working with AV2010/10 USB-Stick [grobasoz/zigbee-firmware#15](https://github.com/grobasoz/zigbee-firmware/issues/15).
> > I have not testing but 2 user have flagged its working.
> 
> Nice! Also see new comment in [grobasoz/zigbee-firmware#15](https://github.com/grobasoz/zigbee-firmware/issues/15) that a user now managed to flash with walthowd docker image too.
> 
> Note that this requires defining a baud rate of 115200 instead of 57600 which I believe is the baud rate on the default firmware.

FYI, EmberZNet 6.7.10 for EM3587 has now also been made available by grobasoz

https://github.com/grobasoz/zigbee-firmware/blob/master/EM3587/NCP_USW_EM3587_6710-LR.ebl

EmberZNet 6.7.10 has been confirmed working by Elix-g in a discussion about flashing here:

https://github.com/walthowd/husbzb-firmware/issues/38

Again,  this requires defining a baud rate of 115200 instead of 57600 which I believe is the baud rate on the default firmware.

closed 2022-01-10 by Hedda

javicalle, 2025-12-14:
In case it might be useful to someone, I'm sharing my experience on the matter here.

I updated the firmware (v5.8.0) of my Bitron Video 2010/10 adapter from a Windows 10 environment.

It was difficult to get all the components working just to run the updater.

Some of the problems I encountered were:

* The first problem was getting Windows to recognize the dongle. The dongle listed as 'Other devices' in the 'Device Manager' is the key to know that you need to 'update drivers'.
I only make it work with the v6.7 of the Silicon Labs drivers (CP210x USB to UART Bridge Virtual COM Port (VCP) drivers). The drivers v11.X didn't work for me.

* Once Windows detects the device, I have to make it visible to the WSL (Ubuntu WSL in my case), that can be done with a Microsoft utility `usbipd` but you need to have a image WSL v2:
  * https://learn.microsoft.com/windows/wsl/connect-usb

* Not sure if really needed but I have followed also that part (modprobe cp210x):
  * https://community.openhab.org/t/how-to-get-bitron-video-bv-2010-10-working/108265/3

****

The rest is fairly straightforward and has already been covered in most of the comments:

1.  you need to edit the `ncp.py` file to include the BitronVideo IDs (`10C4:8B34`) or you can download the one from the PR:
  - https://github.com/walthowd/husbzb-firmware/blob/c14b92f8d719212bf387e42af3e356e806917ebb/ncp.py
2. in your WSL image you need to execute the commands as `root`. If not you will get some `SerialException`, `permission denied`, etc
3. while you have a firmware v5.8.X the `BV2010_BAUD = 57600`
4. once you update to a firmware v6.7.X you need to update to `BV2010_BAUD = 115200`

<details><summary>Command outputs</summary>

`ncp_2.py` is the modified version of original `ncp.py`
`bitron_update` is my pthon venv 

```shell
(bitron_update) root@javicalle-PC:/tmp/bitron_update/husbzb-firmware# ./ncp_2.py scan
Connecting to.. /dev/ttyUSB0 57600 True False
{"ports": [{"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8B34", "deviceType": "zigbee", "stackVersion": "5.8.0-134"}]}
(bitron_update) root@javicalle-PC:/tmp/bitron_update/husbzb-firmware#
```

```shell
(bitron_update) root@javicalle-PC:/tmp/bitron_update/husbzb-firmware# ./ncp_2.py flash -p /dev/ttyUSB0 -f NCP_USW_EM3587_6710-LR.ebl
Restarting NCP into Bootloader mode...
BV2010 stick
EM3587 Serial Btl v5.8.0.0 b134

Successfully restarted into bootloader mode! Starting upload of NCP image...
Finished!
Rebooting NCP...
(bitron_update) root@javicalle-PC:/tmp/bitron_update/husbzb-firmware#
```

Once updated, I have to change to `BV2010_BAUD = 115200` to connect to device:

```shell
(bitron_update) root@javicalle-PC:/tmp/bitron_update/husbzb-firmware# ./ncp_2.py scan
Connecting to.. /dev/ttyUSB0 115200 True False
{"ports": [{"port": "/dev/ttyUSB0", "vid": "10C4", "pid": "8B34", "deviceType": "zigbee", "stackVersion": "6.7.10-423"}]}
(bitron_update) root@javicalle-PC:/tmp/bitron_update/husbzb-firmware#
```

</details>