zigpy/bellows #351: [COMPATIBILITY] Cortet ZM3588S-USB or ZM357S-USB EM35x USB Sticks by CEL?

Hedda, 2020-09-18:
Anyone tested if these "Cortet" EM3588 / EM357 based radios from CEL are compatible with bellows (EZSP) out-of-the-box?

EM35x based USB dongles in the "Cortet" series by CEL (California Eastern Laboratories):

***CEL Cortet ZM3588S-USB (EM3588)***

***CEL Cortet ZM357S-USB (EM357)*** 

Also, they have matching models come in LR (long-range) versions however they had some hint that may ship Thread firmware:

***CEL Cortet ZM3588S-USB-LR (EM3588)***

***CEL Cortet ZM357S-USB-LR (EM357)*** 

If they don't work out-of-the-box has anyone tried to flash them with newer Silabs EmberZNet firmware?

They are not expensive and seem to be sold all around the world by major electronics distributors. 

The documentation says they support Silicon Labs EmberZNet PRO ZigBee stack and serial interface:

http://www.cel.com/pdf/datasheets/EM357_USB_Sticks_DS.pdf

http://www.cel.com/product.do?command=viewOverview&group=6

I'm not asking for buying advice as already have adapters but requesting info for ZHA compatibility list:

https://www.home-assistant.io/integrations/zha/

PS: Found one mentioned in OpenHAB Supported Coordinators under their ZigBee Binding documentation:

https://www.openhab.org/addons/bindings/zigbee/

Hedda, 2020-11-25:
@walthowd Do you think these adapters could also be updated with your firmware updater image for Zigbee coordinators? 

https://github.com/walthowd/husbzb-firmware

Just a curious question as I do not actually own any of those myself and do not plan on buying them, nor recommend them.

walthowd, 2020-11-25:
I'm not sure -- It appears the `ncp.py` I have adapted and updated from Silicon Labs to query and load the firmware was originally written for CEL sticks, so I would hazard a guess it would work -- but someone would still need to test.

Hedda, 2022-01-10:
FYI, grobasoz tested his firmware image compiled EM3581 on a another EM3588 based adapter he had and it worked for him:

https://github.com/grobasoz/zigbee-firmware

https://github.com/walthowd/husbzb-firmware/

https://github.com/Elelabs/elelabs-zigbee-ezsp-utility/

closed 2022-01-10 by Hedda