zigpy/bellows #361: Trying to find a way to interact with the devices on the network.

chukpozohnToe, 2020-10-26:
Hi! I'm using zigpy and bellows to write my own code to interact my smart plugs. As of now I can successfully scan for the devices and see them on my network. My question is, when the devices are discovered, where are they stored? Where does zigpy/bellows store the devices?

INFO:ezsc:Setting up the Zigbee network ...
INFO:bellows.ezsp:EZSP Stack Type: 2, Stack Version: 24624
INFO:zigpy.application:Device 0x0000 (cc:cc:cc:ff:fe:18:08:67) joined the network
INFO:ezsc:Network setup complete!
INFO:bellows.zigbee.application:ZDO Device announce: 0x6edf, 90:fd:9f:ff:fe:61:92:c5
INFO:zigpy.application:Device 0x6edf (90:fd:9f:ff:fe:61:92:c5) joined the network

Solving this problem will bring me one step closer to finishing up my project. 

Thank you!

Adminiuga, 2020-10-26:
In a sqlite database, if the controller application was instantiated with a database file name

chukpozohnToe, 2020-10-29:
Thanks a lot! Knowing that helped me solved other problems in my code!

closed 2020-10-29 by chukpozohnToe

Hedda, 2020-11-18:
@chukpozohnToe this and any questions similar to this would probably be best asked and answered under zigpy issues:

https://github.com/zigpy/zigpy/issues

That way will also easier for other developers to find questions and answers about zigpy development in one location.