zigpy/bellows #363: What smart plugs are recommended?

chukpozohnToe, 2020-11-02:
Hi.

I'm having trouble sending the right command to my smart plugs. As you're aware I wrote my code using the zippy & bellows libraries. At this point I can reach my devices and interact with them. But when I send a command to turn them on/off it doesn't work. I'm getting an error saying that my devices can't accept on/off commands. I think it might be a problem with the devices I'm using. I'm currently using "Innr Zigbee Smart Pugs". Are there other Zigbee smart pugs that you would recommend? 

ERROR:__main__:Failed to control device!
Traceback (most recent call last):
  File "main.py", line 33, in control_device
    await ezsc_controller.send_command(device_ieee, command, "")
  File "/home/pi/PROJECTS/Brain-Computer-Interface/Python-code/ezsc.py", line 75, in send_command
    v = await getattr(self._get_cluster_by_command(device, command), command)(*params)
  File "/home/pi/PROJECTS/Brain-Computer-Interface/Python-code/ezsc.py", line 49, in _get_cluster_by_command
    raise Exception("Device does not support command %s!" % (command,))
Exception: Device does not support command on!

Thank you!

Hedda, 2020-11-18:
Personally, I would advice using Zigbee Alliance certified devices 

https://zigbeealliance.org/zigbee_products/?product_type=certified_product

I think that Home Assistant's ZHA integration documentation has a good explanation of why that is the case:

https://www.home-assistant.io/integrations/zha/#knowing-which-devices-are-supported

As you probably know/understand, there is a bugger chance that devices that do not follow Zigbee Alliance specifications will have to use "quirks" via the ZHA Device Handlers library.

https://www.home-assistant.io/integrations/zha/#zha-exception-and-deviation-handling

Hedda, 2020-11-18:
@chukpozohnToe Have you implemented zha-device-handlers in your integration implementation of zigpy?

https://github.com/zigpy/zha-device-handlers

https://github.com/zigpy/zha-device-handlers/blob/dev/README.md

Hedda, 2020-11-18:
@chukpozohnToe this and any questions similar to this would probably be best asked and answered under zigpy issues:

https://github.com/zigpy/zigpy/issues

That way will also easier for other developers to find questions and answers about zigpy development in one location.

PS: Please feel free to help improve the zigpy documentation while workiing on this, see example https://github.com/zigpy/zigpy/pull/477 and https://github.com/zigpy/zigpy/issues/471

closed 2020-12-17 by Adminiuga