zigpy/bellows #364: What commands control the devices

chukpozohnToe, 2020-11-09:
Hi, 

I'm having trouble sending the right command to the devices on my network. I'm writing my own code using zippy/bellows library. Please help

INFO:ezsc:sending command on to device 14757395478837987431
ERROR:__main__:Failed to control device!
Traceback (most recent call last):
  File "prof_test.py", line 24, in control_device
    await ezsc_controller.send_command(device_ieee, command, "")
  File "/home/pi/PROJECTS/Brain-Computer-Interface/Python-code/ezsc.py", line 75, in send_command
    v = await getattr(self._get_cluster_by_command(device, command), command)(*params)
  File "/home/pi/PROJECTS/Brain-Computer-Interface/Python-code/ezsc.py", line 49, in _get_cluster_by_command
    raise Exception("Device does not support command %s!" % (command,))
Exception: Device does not support command on

Adminiuga, 2020-11-09:
Command 'on' is only supported if the device has the "on_off" (cluster_id==6) input cluster.

closed 2020-11-09 by Adminiuga

chukpozohnToe, 2020-11-09:
It Looks like my cluster id is 242. Could you please tell me the commands associated with it? Also I read somewhere in the library that the commands of each cluster id was in the log but I couldn't find it. 

Also if possible, I would be interested in helping to documentations for the library on how to use it features. 

INFO:ezsc:Network setup complete!
[{'ieee': 14757395478837987431, 'nwk': 0x0000, 'endpoints': [{'id': 1, 'input_clusters': [], 'output_clusters': [], 'status': 'initialized'}]}, 

{'ieee': 10447682632402703045, 'nwk': 0xb3d9, 'endpoints': [{'id': 1, 'input_clusters': [0, 3, 4, 5, 6, 8, 2821, 4096, 64642], 'output_clusters': [10, 25], 'status': 'initialized'}, {'id': 242, 'input_clusters': [], 'output_clusters': [33], 'status': 'initialized'}]}, 

{'ieee': 10447682632402344369, 'nwk': 0x8664, 'endpoints': [{'id': 1, 'input_clusters': [0, 3, 4, 5, 6, 8, 2821, 4096, 64642], 'output_clusters': [10, 25], 'status': 'initialized'}, {'id': 242, 'input_clusters': [], 'output_clusters': [33], 'status': 'initialized'}]}]

Thank you!

Adminiuga, 2020-11-09:
check https://github.com/zigpy/zigpy/pull/477
There is no ZCL cluster id 242. There's endpoint_id 242 which designates Zigbee Green Power, but that currently does not work.
Both of `nwk': 0xb3d9`,    `'nwk': 0x8664` devices have endpoint id 1 with "cluster_id==6" present in the list of "input_clusters". "on" command would work only for those. 

I'd suggest using home-assistant version `0.77.3` as a reference implementation, as it is simpler for understanding than the current one.  e.g https://github.com/home-assistant/core/blob/4063b24ddb8aa1174527c6aa35ecc53fcccad077/homeassistant/components/light/zha.py#L115 turns on the device.
in your case something like `await zigpy_device_instance.endpoint[1].on_off.on()`

Hedda, 2020-11-18:
@chukpozohnToe this and any questions similar to this would probably be best asked and answered under zigpy issues:

https://github.com/zigpy/zigpy/issues

That way will also easier for other developers to find questions and answers about zigpy development in one location.

PS: Please feel free to help improve the zigpy documentation while workiing on this, see example https://github.com/zigpy/zigpy/pull/477 and https://github.com/zigpy/zigpy/issues/471

Hedda, 2021-02-15:
FYI, zoic21 made progress with ZGP (Zigbee Green Power) and is requesting regression testing of his pull request https://github.com/zigpy/zigpy/pull/656

> Can someone can test with this code if all it's ok ? No side effect. If it I think we can merge this first step (after I fix coveralls).

That is, he is asking people do functional and non-functional tests to ensure that his patch does not break any existing functions. If it does not break anything obvious in zigpy then it could possible be merged 'as-is' even if ZGP might be be fully production ready.