zigpy/bellows #366: Stability issues with Telegesis ETRX357USB adapters

macfly92, 2020-11-23:
Hi,

I've started to play with my Telegesis ETRX357USB adapter. Firt I have included some Xiaomi sensors without any issue for some days.
Next I decided to unbind 6 of my Tradfri Ikea Bulb from my Hue bridge to pair them with the Zigbee key.
The pairing was done, and I've created a group for them. The adapter stay stable as long as I dind't send any order to bulbs. But most of the time, when I send an order to bulbs, I catch this error :

```
2020-11-23 09:49:39 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2020-11-23 09:49:39 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 88 (invalidCommand) received: b'31'
2020-11-23 09:49:39 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback ThreadsafeProxy.__getattr__.<locals>.func_wrapper.<locals>.check_result_wrapper() at /usr/local/lib/python3.8/site-packages/bellows/thread.py:97
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/bellows/thread.py", line 98, in check_result_wrapper
    result = call()
  File "/usr/local/lib/python3.8/site-packages/bellows/ezsp/__init__.py", line 230, in frame_received
    self._protocol(data)
  File "/usr/local/lib/python3.8/site-packages/bellows/ezsp/protocol.py", line 111, in __call__
    assert expected_id == frame_id
AssertionError
```

Often, this error is followed by a "NCP Entered failed state".

```
2020-11-23 09:53:19 DEBUG (MainThread) [zigpy.zcl] [0x6596:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=45 command_id=Command.Read_Attributes_rsp>
2020-11-23 09:53:19 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'00f6'
2020-11-23 09:53:19 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xD029, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=51), 52, b'\x003\x00\x00\x00')
2020-11-23 09:53:19 ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2020-11-23 09:53:19 DEBUG (MainThread) [bellows.ezsp] /dev/ttyUSB0 connection lost unexpectedly: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2020-11-23 09:53:19 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
2020-11-23 09:53:19 DEBUG (MainThread) [bellows.zigbee.application] Received _reset_controller_application frame with ('Serial connection loss: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)',)
2020-11-23 09:53:19 DEBUG (MainThread) [bellows.zigbee.application] Resetting ControllerApplication. Cause: 'Serial connection loss: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)'
2020-11-23 09:53:20 DEBUG (MainThread) [bellows.ezsp] Resetting EZSP

```
Most of the time the system hang for some seconds and the states of some bulbs isn't correct.

I've flashed the adapter with the 8 years old firmware found at https://github.com/yqyunjie/Zigbee-Project and my first thought was to upgrade this one but despite lot of searchs, I dind't find a newer firmware for this dongle. The version is : EmberZNet version: 5.0.0.0 build 95

- My hardware is : https://www.gearbest.com/mini-pc/pp_1633397.html?wid=1433363.
- HA version is 118.0
- My Operating system is Debian 9 with Home Assistant core on docker.
- I've removed modem manager.
- Disk I/O wasn't very loaded. 
- Database size is approx 150mb.

I've also tried to swap the two USB port where stand my Z-wave and Zigbee adapters. The Z-wave adapter works well in both case.

I don't know what else to do, so my last hope is to ask some help here before give up and reverting back ...

For sure, I'll be happy to provide more informations or testing.

Thanks for reading.

Adminiuga, 2020-11-23:
Telegis has even older firmware than Husbzb. It is possible that it just doesn't support group commands and this is what the invite command frame/AssertionError error says.

Regarding entering the failed state -- depends on hardware. Usually happens on raspberry PIs when:
- high disk io
- recorder db is larger than 300-400MB
- sd card going bad
- bad power adapter

Adminiuga, 2020-11-23:
Dog, it's morning and I missed the details you already provided

macfly92, 2020-11-23:
Thanks for the reply,

Sending a command to the group perform the action correctly, but some bulbs doesn't report its state correctly and the Zigbee system become unresponsive. (After a OFF command to the group, some bulbs (but not whole) turn back ON again on HA, but not in the "real life" ). Must I remove the Zigbee group and play with HA Group instead ? Or Must I forget this dongle and look for another one ?

As I writed before, I'm not running HA on a Pi device, but on a fanless mini PC.

Adminiuga, 2020-11-23:
If you want to play with format upgrade, check https://www.github.com/walthowd/husbzb-firmware/tree/master/Dockerfile inside it has a URL for sdk which should have a newer firmware for EM357

The NCP failed state could be the result of the unknown command and actually it could be the read counter commands. 
Do you get the AssertionError if you disable debug for bellows.ezsp?

macfly92, 2020-11-23:
I've removed the Zigbee group and played with regular HA groups and it seem to be more stable but I noticed somes AssertionError and I've got some NCP failed test if I stress it a little bit.

Setting log to bellows.ezsp: info, and while HA is restarting, i saw this error : 
```
 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback ThreadsafeProxy.__getattr__.<locals>.func_wrapper.<locals>.check_result_wrapper() at /usr/local/lib/python3.8/site-packages/bellows/thread.py:97
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/asyncio/events.py", line 81, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.8/site-packages/bellows/thread.py", line 98, in check_result_wrapper
    result = call()
  File "/usr/local/lib/python3.8/site-packages/bellows/ezsp/__init__.py", line 230, in frame_received
    self._protocol(data)
  File "/usr/local/lib/python3.8/site-packages/bellows/ezsp/protocol.py", line 111, in __call__
    assert expected_id == frame_id
AssertionError
```
This is my log config :     
    homeassistant.components.zha: debug
    bellows.zigbee.application: debug
    bellows.ezsp: info
    zigpy: debug
    zigpy_cc: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug  

Must I completly remove the debug config ?

And thanks you so much for the firmware link, I found something for the EM357, I will check how to update that and report back.

Best regards

Adminiuga, 2020-11-23:
```
homeassistant.components.zha: debug

bellows.ezsp: info
...
```
My bad, to disable "counter reading" you need to disable debug for `bellows.zigbee.application: info` or alternatively edit https://github.com/zigpy/bellows/blob/c12cbfecb296c70adb12f0db765dff397da589cc/bellows/zigbee/application.py#L629 to be `if True:` so you could still have the debug, but not reading the counters.

macfly92, 2020-11-23:
Ok, with the help of your link, I've managed to update EmberZNet from version: 5.0.0.0 build 95 to version: 6.4.1.0 build 408, thanks you.

Now, the ReadCounter command doesn't throw an error anymore :

```
 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 241 (readCounters) received: b'29000d000e000400000000000000000004000200000000000200060000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000'
 DEBUG (MainThread) [bellows.zigbee.application] EZSP Counters: COUNTER_MAC_RX_BROADCAST: 41,COUNTER_MAC_TX_BROADCAST: 13,COUNTER_MAC_RX_UNICAST: 14,COUNTER_MAC_TX_UNICAST_SUCCESS: 4,COUNTER_MAC_TX_UNICAST_RETRY: 0,COUNTER_MAC_TX_UNICAST_FAILED: 0,COUNTER_APS_DATA_RX_BROADCAST: 0,COUNTER_APS_DATA_TX_BROADCAST: 0,COUNTER_APS_DATA_RX_UNICAST: 4,COUNTER_APS_DATA_TX_UNICAST_SUCCESS: 2,COUNTER_APS_DATA_TX_UNICAST_RETRY: 0,COUNTER_APS_DATA_TX_UNICAST_FAILED: 0,COUNTER_ROUTE_DISCOVERY_INITIATED: 2,COUNTER_NEIGHBOR_ADDED: 6,COUNTER_NEIGHBOR_REMOVED: 0,COUNTER_NEIGHBOR_STALE: 0,COUNTER_JOIN_INDICATION: 0,COUNTER_CHILD_REMOVED: 0,COUNTER_ASH_OVERFLOW_ERROR: 0,COUNTER_ASH_FRAMING_ERROR: 0,COUNTER_ASH_OVERRUN_ERROR: 0,COUNTER_NWK_FRAME_COUNTER_FAILURE: 0,COUNTER_APS_FRAME_COUNTER_FAILURE: 0,COUNTER_UTILITY: 0,COUNTER_APS_LINK_KEY_NOT_AUTHORIZED: 0,COUNTER_NWK_DECRYPTION_FAILURE: 0,COUNTER_APS_DECRYPTION_FAILURE: 0,COUNTER_ALLOCATE_PACKET_BUFFER_FAILURE: 0,COUNTER_RELAYED_UNICAST: 0,COUNTER_PHY_TO_MAC_QUEUE_LIMIT_REACHED: 0,COUNTER_PACKET_VALIDATE_LIBRARY_DROPPED_COUNT: 0,COUNTER_TYPE_NWK_RETRY_OVERFLOW: 0,COUNTER_PHY_CCA_FAIL_COUNT: 0,COUNTER_BROADCAST_TABLE_FULL: 0,COUNTER_PTA_LO_PRI_REQUESTED: 0,COUNTER_PTA_HI_PRI_REQUESTED: 0,COUNTER_PTA_LO_PRI_DENIED: 0,COUNTER_PTA_HI_PRI_DENIED: 0,COUNTER_PTA_LO_PRI_TX_ABORTED: 0,COUNTER_PTA_HI_PRI_TX_ABORTED: 0,COUNTER_ADDRESS_CONFLICT_SENT: 0
```

Unfortunatly, it isn't more stable, there are some errors like :

```
 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'008b'
DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xE011, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=225), 226, b'\x00\xe1\x00\x00\x00')
ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

```
```
 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=83), 255, -61, 0x8b95, 255, 255, b'\x08\xfa\x0b\x01\x00']
DEBUG (MainThread) [zigpy.zcl] [0x8b95:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=250 command_id=Command.Default_Response>
 ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```

As you said for Raspberry, it's maybe a power problem, I'll try to find a self powered USB Hub for test just in case.

In the mean time, don't hesitate if I can dig out more tests.

Adminiuga, 2020-11-23:
You still may have more than one issue. I was thinking that assertion error would cause USB disconnects, but I cannot reproduce it with HUSBZB-1. Even when I force it to throw the error because of the `invalidCommand` serial connection still stays there and in your cases you mentioning that you get
```
2020-11-23 09:53:19 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xD029, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=51), 52, b'\x003\x00\x00\x00')
2020-11-23 09:53:19 ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2020-11-23 09:53:19 DEBUG (MainThread) [bellows.ezsp] /dev/ttyUSB0 connection lost unexpectedly: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2020-11-23 09:53:19 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
```

Can you check `dmesg`, `syslog` kernel log on the host to see if there's any correlation between "Lost serial connection" and anything usb related?

macfly92, 2020-11-23:
Yep, each time I've got an NCP Reset, I've got an USB disconnect on Dmesg

```
[63152.589752] cp210x ttyUSB0: usb_serial_generic_read_bulk_callback - urb stopped: -32
[63152.602375] cp210x ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
[63152.606405] cp210x ttyUSB0: usb_serial_generic_write_bulk_callback - nonzero urb status: -71
[63152.657821] usb 1-4.2: USB disconnect, device number 50
[63152.658020] cp210x ttyUSB0: failed set request 0x7 status: -19
[63152.660100] cp210x ttyUSB0: failed set request 0x12 status: -19
[63152.662097] cp210x ttyUSB0: failed set request 0x0 status: -19
[63152.664445] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
[63152.664513] cp210x 1-4.2:1.0: device disconnected
[63152.891747] usb 1-4.2: new full-speed USB device number 51 using xhci_hcd
[63153.013674] usb 1-4.2: New USB device found, idVendor=10c4, idProduct=8293
[63153.013692] usb 1-4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[63153.013702] usb 1-4.2: Product: Telegesis USB Device
[63153.013710] usb 1-4.2: Manufacturer: Silicon Labs
[63153.013717] usb 1-4.2: SerialNumber: 04000B47
[63153.016180] cp210x 1-4.2:1.0: cp210x converter detected
[63153.023864] usb 1-4.2: cp210x converter now attached to ttyUSB0

```

So ... what is your tought ? One more point for powered problem ?

Adminiuga, 2020-11-23:
Ok. Either set logging for  `bellows.zigbee.application: info` to info or make modifications to the local file as outlined above.
This should eliminate the assertion errors and monitor for disconnects. I still think the NCP resets are caused by the serial disconnects.

macfly92, 2020-11-23:
I didn't notice any assertion error since EmberZNet update.

Here is my logging setup now : 
logger:
  default: info
  logs:
#    homeassistant.core: debug
    homeassistant.components.zha: debug
    bellows.zigbee.application: info
    bellows.ezsp: debug
    zigpy: debug
    zigpy_cc: debug
    zigpy_deconz.zigbee.application: debug
    zigpy_deconz.api: debug
    zigpy_xbee.zigbee.application: debug
    zigpy_xbee.api: debug
    zigpy_zigate: debug
    zigpy_znp: debug
    zhaquirks: debug  

And this is an exemple when an error throw with some context before if it help : 

```
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:46 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'0df80538aafeff570b00ffc800'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'00040108000101400100000dffc0d3bcffff0818550100000020fe02'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [zigpy.zcl] [0xbcd3:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=85 command_id=Command.Read_Attributes_rsp>#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000401060001014001000070ffcc11e0ffff08184b01000000100102'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [zigpy.zcl] [0xe011:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=75 command_id=Command.Read_Attributes_rsp>#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'00d3bc0401080001014001000063560000'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xBCD3:1:0x0008]: received attribute: 0 update with value: 254#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000401080001014001000078ffc80df8ffff0818570100000020fe02'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [zigpy.zcl] [0xf80d:1:0x0008] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=87 command_id=Command.Read_Attributes_rsp>#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'000df80401080001014001000064580000'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [homeassistant.components.zha.core.channels.base] [0xF80D:1:0x0008]: received attribute: 0 update with value: 254#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000401060001014001000003fec229d0ffff08184d010000001001'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [zigpy.zcl] [0xd029:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=True> manufacturer=None tsn=77 command_id=Command.Read_Attributes_rsp>#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'0029d0040106000101400000005f4e0000'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'29d00e82a6feff570b00ffc300'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xD029, EmberApsFrame(profileId=260, clusterId=8, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=89), 90, b'\x00Y\x00\x00\x00')#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'0011e0040106000101400000005e4c0000'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'29d00e82a6feff570b00ffc300'#033[0m
Nov 23 17:47:47 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:47 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 89 (incomingRouteRecordHandler) received: b'958b554398feff570b00ffc300'#033[0m
[... cut ...]
Nov 23 17:47:48 Flycore kernel: [64431.842980] usb 1-4.2: Manufacturer: Silicon Labs
Nov 23 17:47:48 Flycore kernel: [64431.842988] usb 1-4.2: SerialNumber: 04000B47
Nov 23 17:47:48 Flycore kernel: [64431.846143] cp210x 1-4.2:1.0: cp210x converter detected
Nov 23 17:47:48 Flycore kernel: [64431.847663] usb 1-4.2: cp210x converter now attached to ttyUSB0
Nov 23 17:47:49 Flycore c57949a93393[567]: #033[32m2020-11-23 17:47:49 INFO (MainThread) 
Nov 23 17:47:53 Flycore c57949a93393[567]: #033[36m2020-11-23 17:47:53 DEBUG (MainThread) [bellows.ezsp] Resetting EZSP#033[0m

```

Adminiuga, 2020-11-23:
i guess i should refresh the page more often.

> Ok, with the help of your link, I've managed to update EmberZNet from version: 5.0.0.0 build 95 to version: 6.4.1.0 build 408

Cool. After an "NCP Reset" can you please post the output similar to:
```
2020-11-23 11:48:48 INFO (MainThread) [bellows.zigbee.application] EZSP Radio manufacturer: HubZ ZigBee
2020-11-23 11:48:48 INFO (MainThread) [bellows.zigbee.application] EZSP Radio board name: HUSBZB-1
2020-11-23 11:48:48 INFO (MainThread) [bellows.zigbee.application] EmberZNet version: 5.7.2.0 build 444
```
I'd like to update the documentation with confirmation of a successful update, but like to confirm exact board name.

Regarding the disconnects, per 
```
Nov 23 17:47:47 Flycore kernel: [64431.121857] usb 1-4.2: USB disconnect, device number 54
Nov 23 17:47:47 Flycore kernel: [64431.122123] cp210x ttyUSB0: failed set request 0x7 status: -19
Nov 23 17:47:47 Flycore kernel: [64431.124715] cp210x ttyUSB0: failed set request 0x12 status: -19
Nov 23 17:47:47 Flycore kernel: [64431.127764] cp210x ttyUSB0: failed set request 0x0 status: -19
Nov 23 17:47:47 Flycore kernel: [64431.130796] cp210x ttyUSB0: cp210x converter now disconnected from ttyUSB0
```
It does look like full hard disconnect. Either the problem with
1. USB driver problem???
2. Bad physical connection???
3. Power problem or over-current situations?

What else do you have connected to that USB? I'd try with USB powered hub next, to exclude bad power and or over current on USB port.

macfly92, 2020-11-23:
Sure here it is : 
```
Nov 23 18:29:21 Flycore c57949a93393[567]: #033[32m2020-11-23 18:29:21 INFO (MainThread) [bellows.zigbee.application] EZSP Radio manufacturer: Telegesis#033[0m
Nov 23 18:29:21 Flycore c57949a93393[567]: #033[32m2020-11-23 18:29:21 INFO (MainThread) [bellows.zigbee.application] EZSP Radio board name: ETRX357-LRS#033[0m
Nov 23 18:29:21 Flycore c57949a93393[567]: #033[32m2020-11-23 18:29:21 INFO (MainThread) [bellows.zigbee.application] EmberZNet version: 6.4.1.0 build 408#033[0m
```

For what it worth, I will post what I've done to upgrade in the issue : https://github.com/zigpy/bellows/issues/347

I've got one USB Z-wave dongle and my embeded ethernet adapter seem to be USB related too.

I will try with a USB powered Hub to see if it help and report back here as soon as I catch one.

Anyway, thanks you for your work, your time and your help !

macfly92, 2020-11-23:
Ok, so I've tested with 2 differents USB self powered Hub with 2 different 2A power supply. It seem to be worse. Also I've removed my Z-Wave adapter and try to run the Zigbee dongle alone, all without any success. The Z-Wave dongle works well on the two different USB port. Maybe I must just admit that I have a faulty dongle...

Thanks again for all.

dmulcahey, 2020-11-23:
Have you checked to make sure modemmanager or whatever thats is in the OS is disabled?

dmulcahey, 2020-11-23:
https://howtoinstall.co/en/debian/stretch/modemmanager-dev?action=remove

Try this

macfly92, 2020-11-23:
Hi, thanks for the idea, I have check that more than twice ...
I've also try a "lsof | grep ttyUSB0" to seek a serial session opened and I see nothing when HA is not running, and lot of Python3 only when it run.

macfly92, 2020-11-23:
Oh wait ! 
Looking at your link I've realized that I've done a remove and purge of modemmanager but not a apt autoremove.
Trying that shown me some suspicious packages remain : usb-modeswitch and usb-modeswitch-data.

I will testing further, but remove them seem to simply remove also the problem ...

Thanks you both for your precious help, and sorry for inconvenience

closed 2020-11-24 by macfly92

dmulcahey, 2020-11-24:
@macfly92 noticed you closed this... are you all good now?

macfly92, 2020-11-24:
Yes, it is working perfectly now.
I am a little ashamed, I read several times to remove the modem manager and I just forget about the Autoremove ...
Anyway my dongle is working, I updated the Ember firmware and maybe this story will help someone someday.

Thanks again !

Best regards

dmulcahey, 2020-11-24:
Don’t be! This stuff gets complicated. Glad it’s working.