zigpy/bellows #377: ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart

georaspi, 2020-12-07:
Hello all,

I have been getting those errors the HA log, in random intervals. I am using the Sonoff Zigbee bridge with the custom ncp-uart-sw_6.7.6_115200.ota firmware. My setup consists of various Sonoff PIR and TH sensors. I am getting those Errors and then out of the blue the sensors become unstable and stop working. Usually a HA restart or a ZHA integration reload fixes the issue.
My system info and the Error details are shown below. 

Would anyone have any idea what could be wrong? Many thanks in advance.

==========
Error details
==========
Logger: bellows.ezsp
Source: /usr/local/lib/python3.8/site-packages/bellows/ezsp/__init__.py:194
First occurred: 11:12:34 AM (4 occurrences)
Last logged: 6:28:43 PM

NCP entered failed state. Requesting APP controller restart

===========
System setup
===========
System Health

Home Assistant Core Integration

version: 0.118.5
installation_type: Home Assistant Supervised
dev: false
hassio: true
docker: true
virtualenv: false
python_version: 3.8.6
os_name: Linux
os_version: 5.4.79-v7+
arch: armv7l
timezone: Europe/London

Home Assistant Cloud

logged_in: false
can_reach_cert_server: ok
can_reach_cloud_auth: ok
can_reach_cloud: pending

Hass.io

host_os: Raspbian GNU/Linux 10 (buster)
update_channel: stable
supervisor_version: 2020.12.6
docker_version: 19.03.14
disk_total: 109.5 GB
disk_used: 15.8 GB
healthy: true
supported: failed to load: Unsupported
supervisor_api: ok
version_api: ok
installed_addons: Duck DNS (1.12.4), NGINX Home Assistant SSL proxy (3.0.1), File editor (5.2.0), Mosquitto broker (5.1), AdGuard Home (2.6.1), Glances (0.9.1), TasmoAdmin (0.13.1), WireGuard (0.4.0), Plex Media Server (2.4.1), Alsa & PulseAudio Fix (3.4), ESPHome (1.15.3)

Lovelace

dashboards: 1
mode: yaml
views: 11
resources: 11

MattWestb, 2020-12-10:
Is you using the SonOff Zigbee Bridge as one serial 2 net for the NCP ?
I have doing that and if my WiFi is restarting or being jammed (of bad behaving ESP8266) the ser2net connection is going down and ZHA is complaining.

Can you catching some deeper debug logs from HA then its happening ?

georaspi, 2020-12-10:
> Is you using the SonOff Zigbee Bridge as one serial 2 net for the NCP ?
> I have doing that and if my WiFi is restarting or being jammed (of bad behaving ESP8266) the ser2net connection is going down and ZHA is complaining.
> 
> Can you catching some deeper debug logs from HA then its happening ?

That makes sense to me and I have noticed that happening to me, Wifi restarts etc. Will try to pull some more detailed logs if it happens again

MattWestb, 2020-12-10:
Great !!
I was putting in one new Netgear Nighthawk router for some weeks ago and moved it away from my Zigbee coordinators for getting better signal to noise for the Zigbee networks but then 4 ESP was getting lower signals and start restarting and jamming / blocking the WiFi network.
Was moving the router 1/2 between the internet router / modem and the Zigbee coordinators and all systems is more (not 110%) happier :-))

Between I have HA running on one Pi3B+ and is using with WiFi only (NOT RECOMMENDED) but its working OK if the WiFi is OK but if getting problems its being one large mess in HA :-((

Hedda, 2020-12-10:
@georaspi Just to clarify; you use Sonoff ZBBridge WiFi-to-Zigbee bridge from ITead flashed with Tasmota ZBBridge firmware?

That is, you have basically followed this -> https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html or?

Sounds as if would need some kind of extra cache or other solution on one or both sides to handle drops in communication?

georaspi, 2020-12-10:
> @georaspi Just to clarify; you use Sonoff ZBBridge WiFi-to-Zigbee bridge from ITead flashed with Tasmota ZBBridge firmware?
>
> That is, you have basically followed this -> [...]

This is exactly what I am using and what I have followed, yes.

I can see your point yes. You might be right. I am not that advanced though to go that deep ...

Adminiuga, 2020-12-10:
That's normal part of operation and is a side effect of connecting to the serial port over the WiFi. In EZSP protocol, if there's packet (serial packet) loss or loss of sync, then both Host and NCP server enter into the failed state and require a full connection/stack reset to continue further operation.

closed 2020-12-10 by Adminiuga

Hedda, 2020-12-11:
@Adminiuga So is your recommendation to avoid this then to simply never ever use a serial TCP server over WiFi with bellows?

Are there any ways to make bellows more resilient to packet (serial packet) loss or loss of sync when using serial TCP connection?

If not then maybe Sonoff ZBBridge flashed with Tasmota ZBBridge firmware should be listed as experimental with a warning?

- https://github.com/zigpy/bellows/blob/dev/README.md
- https://github.com/zigpy/zigpy/blob/dev/README.md
- https://www.home-assistant.io/integrations/zha/#compatible-hardware

By the way, does protocols that ZiGate and Texas Instruments use to handle packet (serial packet) loss or loss of sync better?

ZiGate Pack WiFi is a ESP8266 with a serial TCP server (so basically same as Sonoff ZBBridge with Tasmota serial TCP server):

https://zigate.fr/produit/zigate-pack-wifi-v1-3/

Zigbee2MQTT recommend a similar serial over WiFi solution using ser2net serial TCP server on a Raspberry Pi Zero:

https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html

Interestingly they mention on that Zigbee2MQTT page that deCONZ Conbee II is not supported over such a TCP connection.

Adminiuga, 2020-12-11:
Yes.
No.
Read the ash protocol specs.

MattWestb, 2020-12-11:
I think somthing like tuya Zigbee Gateway [ TYGWZ01](https://fccid.io/2ANDL-TYGWZ01/Internal-Photos/Int-Photos-4202979.iframe) with ethernet is more or less ideal but i have not finding one firmware that is working in the RTL chip.

If connecting the Zigbee module to the RPIs GPO one problem is away and dont have connecting problems with the Zigbee but then not using the ethernet HA is having problem then WiFi is dropping / getting interference.

[image]

PI3b+ with ESP32 cam and IKEA ICC-A-1 module.

Hedda, 2020-12-15:
Started a separate discussion about low-level bridge solutions here -> https://github.com/zigpy/zigpy/discussions/580