zigpy/bellows #385: Network unstable

Ricciolo, 2021-01-06:
Hi folks,
I'm using Home Assistant with ZHA and a flashed Sonof with Tasmota firmware (9.2.0.2 zbbridge).
All devices work but often my switches don't work with the first try, especially with automations and I have to turn the switch off manually.
I've activated the log but it's hard for me to understand what is the issue. Look likes the coordinator is unreachable or some commands go in timeout.
The devices dignature are [TS0003.txt](https://github.com/zigpy/bellows/files/5775326/TS0003.txt) and [TS0012.txt](https://github.com/zigpy/bellows/files/5775329/TS0012.txt)
Can you help me to diagnose the issue?

Thanks
```
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'00a1'
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xA099, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=42), 43, b'\x01*\x01')
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 52 (sendUnicast) received: b'00a2'
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xE305, EmberApsFrame(profileId=260, clusterId=1024, sourceEndpoint=2, destinationEndpoint=2, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=179), 44, b'\x18\xb3\x0b\n\x00')
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010600010140010000949cc399a0ffff05082a0b0100'
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=148), 156, -61, 0xa099, 255, 255, b'\x08*\x0b\x01\x00']
2021-01-06 11:43:59 DEBUG (MainThread) [zigpy.zcl] [0xa099:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=True disable_default_response=False> manufacturer=None tsn=42 command_id=Command.Default_Response>
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'0099a004010600010140010000a22b0000'
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 41113, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=162), 43, <EmberStatus.SUCCESS: 0>, b'']
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'000401060001010001000095a0c499a0ffff0718670a00001001'
2021-01-06 11:43:59 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=149), 160, -60, 0xa099, 255, 255, b'\x18g\n\x00\x00\x10\x01']
2021-01-06 11:43:59 DEBUG (MainThread) [zigpy.zcl] [0xa099:1:0x0006] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=103 command_id=Command.Report_Attributes>
2021-01-06 11:43:59 DEBUG (MainThread) [zigpy.zcl] [0xa099:1:0x0006] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=Bool, value=Bool.true>)]]
2021-01-06 11:43:59 DEBUG (MainThread) [zigpy.zcl] [0xa099:1:0x0006] Attribute report received: on_off=1
2021-01-06 11:44:01 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 63 (messageSentHandler) received: b'0005e304010604020240010000a1290000'
2021-01-06 11:44:01 DEBUG (MainThread) [bellows.zigbee.application] Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 58117, EmberApsFrame(profileId=260, clusterId=1030, sourceEndpoint=2, destinationEndpoint=2, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=161), 41, <EmberStatus.SUCCESS: 0>, b'']
2021-01-06 11:44:01 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'040401000402ff00000000a79cc305e3ffff0818b40a0000214503'
2021-01-06 11:44:01 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=260, clusterId=1024, sourceEndpoint=2, destinationEndpoint=255, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=167), 156, -61, 0xe305, 255, 255, b'\x18\xb4\n\x00\x00!E\x03']
2021-01-06 11:44:01 DEBUG (MainThread) [zigpy.zcl] [0xe305:2:0x0400] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=180 command_id=Command.Report_Attributes>
2021-01-06 11:44:01 DEBUG (MainThread) [zigpy.zcl] [0xe305:2:0x0400] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=uint16_t, value=837>)]]
2021-01-06 11:44:01 DEBUG (MainThread) [zigpy.zcl] [0xe305:2:0x0400] Attribute report received: measured_value=837
2021-01-06 11:44:04 WARNING (MainThread) [bellows.zigbee.application] Watchdog heartbeat timeout: 
2021-01-06 11:44:09 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0004010604020100010000a89cc305e3ffff0708b50a00001800'
2021-01-06 11:44:09 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1030, sourceEndpoint=2, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY: 256>, groupId=0, sequence=168), 156, -61, 0xe305, 255, 255, b'\x08\xb5\n\x00\x00\x18\x00']
2021-01-06 11:44:09 DEBUG (MainThread) [zigpy.zcl] [0xe305:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=False> manufacturer=None tsn=181 command_id=Command.Report_Attributes>
2021-01-06 11:44:09 DEBUG (MainThread) [zigpy.zcl] [0xe305:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
2021-01-06 11:44:09 DEBUG (MainThread) [zigpy.zcl] [0xe305:2:0x0406] Attribute report received: occupancy=bitmap8.0
2021-01-06 11:44:09 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xE305, EmberApsFrame(profileId=260, clusterId=1030, sourceEndpoint=2, destinationEndpoint=2, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=181), 45, b'\x18\xb5\x0b\n\x00')
2021-01-06 11:44:11 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'040401060402ff00000000a99cc305e3ffff0718b60a00001800'
2021-01-06 11:44:11 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=260, clusterId=1030, sourceEndpoint=2, destinationEndpoint=255, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=169), 156, -61, 0xe305, 255, 255, b'\x18\xb6\n\x00\x00\x18\x00']
2021-01-06 11:44:11 DEBUG (MainThread) [zigpy.zcl] [0xe305:2:0x0406] ZCL deserialize: <ZCLHeader frame_control=<FrameControl frame_type=GLOBAL_COMMAND manufacturer_specific=False is_reply=False disable_default_response=True> manufacturer=None tsn=182 command_id=Command.Report_Attributes>
2021-01-06 11:44:11 DEBUG (MainThread) [zigpy.zcl] [0xe305:2:0x0406] ZCL request 0x000a: [[Attribute(attrid=0, value=<TypeValue type=bitmap8, value=bitmap8.0>)]]
[... cut ...]
2021-01-06 11:45:17 DEBUG (MainThread) [bellows.ezsp.protocol] Send command getMulticastTableEntry: (14,)
2021-01-06 11:45:18 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 99 (getMulticastTableEntry) received: b'00000000'
2021-01-06 11:45:18 DEBUG (MainThread) [bellows.ezsp.protocol] Send command getMulticastTableEntry: (15,)
2021-01-06 11:45:18 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 99 (getMulticastTableEntry) received: b'00000000'
2021-01-06 11:45:18 DEBUG (MainThread) [bellows.zigbee.application] ControllerApplication successfully reset
```

Ricciolo, 2021-01-06:
I got another situation where I see _EZSP is not running_ for few seconds, but Tasmota is running (all pings work)
```
2021-01-06 12:19:04 INFO (MainThread) [zigpy.application] Device 0xe305 (00:17:88:01:08:67:9d:6c) joined the network
2021-01-06 12:19:04 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 00:17:88:01:08:67:9d:6c
2021-01-06 12:19:04 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 196 (changeSourceRouteHandler) received: b'05e399a000'
2021-01-06 12:19:04 DEBUG (MainThread) [bellows.zigbee.application] Received changeSourceRouteHandler frame with [0xe305, 0xa099, <Bool.false: 0>]
2021-01-06 12:19:04 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 69 (incomingMessageHandler) received: b'0400001300000000000000d598c205e3ffff0c5505e36c9d6708018817008004'
2021-01-06 12:19:04 DEBUG (MainThread) [bellows.zigbee.application] Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_BROADCAST: 4>, EmberApsFrame(profileId=0, clusterId=19, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=213), 152, -62, 0xe305, 255, 255, b'U\x05\xe3l\x9dg\x08\x01\x88\x17\x00\x80']
2021-01-06 12:19:04 INFO (MainThread) [bellows.zigbee.application] ZDO Device announce: 0xe305, 00:17:88:01:08:67:9d:6c
2021-01-06 12:19:04 INFO (MainThread) [zigpy.application] Device 0xe305 (00:17:88:01:08:67:9d:6c) joined the network
2021-01-06 12:19:04 DEBUG (MainThread) [zigpy.application] Skip initialization for existing device 00:17:88:01:08:67:9d:6c
2021-01-06 12:19:04 DEBUG (MainThread) [zigpy.zdo] [0xe305:zdo] ZDO request ZDOCmd.Device_annce: [0xE305, 00:17:88:01:08:67:9d:6c, 128]
2021-01-06 12:19:07 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2021-01-06 12:19:12 WARNING (MainThread) [bellows.zigbee.application] Watchdog heartbeat timeout: 
2021-01-06 12:19:22 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2021-01-06 12:19:27 WARNING (MainThread) [bellows.zigbee.application] Watchdog heartbeat timeout: 
2021-01-06 12:19:33 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xA099, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=129), 130, b'\x01\x81\x00')
2021-01-06 12:19:37 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2021-01-06 12:19:39 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xA099, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=131), 132, b'\x01\x83\x01')
2021-01-06 12:19:42 WARNING (MainThread) [bellows.zigbee.application] Watchdog heartbeat timeout: 
2021-01-06 12:19:42 DEBUG (MainThread) [bellows.zigbee.application] Resetting ControllerApplication. Cause: 'Watchdog timeout. Heartbeat timeouts: 5'
2021-01-06 12:19:43 DEBUG (MainThread) [bellows.ezsp] Resetting EZSP
2021-01-06 12:19:44 DEBUG (MainThread) [bellows.ezsp] Couldn't send command sendUnicast((<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xA099, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=133), 134, b'\x01\x85\x00')). EZSP is not running
2021-01-06 12:19:44 DEBUG (MainThread) [bellows.ezsp] Couldn't send command sendUnicast((<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xA099, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=135), 136, b'\x01\x87\x00')). EZSP is not running
2021-01-06 12:19:44 DEBUG (MainThread) [bellows.ezsp.protocol] Send command version: (4,)
2021-01-06 12:19:44 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 0 (version) received: b'07025065'
2021-01-06 12:19:44 DEBUG (MainThread) [bellows.ezsp] Switching to EZSP protocol version 7
2021-01-06 12:19:44 DEBUG (MainThread) [bellows.ezsp.protocol] Send command version: (7,)
2021-01-06 12:19:45 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 0 (version) received: b'07025065'
2021-01-06 12:19:45 DEBUG (MainThread) [bellows.ezsp] EZSP Stack Type: 2, Stack Version: 6550, Protocol version: 7
2021-01-06 12:19:45 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setConfigurationValue: (<EzspConfigId.CONFIG_MAX_END_DEVICE_CHILDREN: 17>, 32)
2021-01-06 12:19:45 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 83 (setConfigurationValue) received: b'00'
2021-01-06 12:19:45 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setConfigurationValue: (<EzspConfigId.CONFIG_APPLICATION_ZDO_FLAGS: 42>, <EmberZdoConfigurationFlags.APP_HANDLES_UNSUPPORTED_ZDO_REQUESTS|APP_RECEIVES_SUPPORTED_ZDO_REQUESTS: 3>)
2021-01-06 12:19:45 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 83 (setConfigurationValue) received: b'00'
2021-01-06 12:19:45 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setConfigurationValue: (<EzspConfigId.CONFIG_ADDRESS_TABLE_SIZE: 5>, 16)
[... cut ...]
2021-01-06 12:19:46 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 83 (setConfigurationValue) received: b'00'
2021-01-06 12:19:46 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
2021-01-06 12:19:46 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame 83 (setConfigurationValue) received: b'00'
2021-01-06 12:19:46 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setConfigurationValue: (<EzspConfigId.CONFIG_MULTICAST_TABLE_SIZE: 6>, 16)

```

Adminiuga, 2021-01-06:
If you get no replies to time command sent to serial interface, you get a reset. This is working as designed.
```
2021-01-06 12:19:07 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2021-01-06 12:19:12 WARNING (MainThread) [bellows.zigbee.application] Watchdog heartbeat timeout: 
2021-01-06 12:19:22 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2021-01-06 12:19:27 WARNING (MainThread) [bellows.zigbee.application] Watchdog heartbeat timeout: 
2021-01-06 12:19:33 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xA099, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=129), 130, b'\x01\x81\x00')
2021-01-06 12:19:37 DEBUG (MainThread) [bellows.ezsp.protocol] Send command readCounters: ()
2021-01-06 12:19:39 DEBUG (MainThread) [bellows.ezsp.protocol] Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0xA099, EmberApsFrame(profileId=260, clusterId=6, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=131), 132, b'\x01\x83\x01')
2021-01-06 12:19:42 WARNING (MainThread) [bellows.zigbee.application] Watchdog heartbeat timeout: 
2021-01-06 12:19:42 DEBUG (MainThread) [bellows.zigbee.application] Resetting ControllerApplication. Cause: 'Watchdog timeout. Heartbeat timeouts: 5'
```

Ricciolo, 2021-01-06:
So the problem is just that sometimes the module doesn't reply? They are pretty close, just 3 meters, and RSSI is -62, LQI 152

Adminiuga, 2021-01-06:
not the module. The sonoff bridge does not reply to data sent over wifi connection. Most likely unstable WiFi.

Ricciolo, 2021-01-06:
Sonoff is behind the access point:  RSSI is100%, -23 dBm :-(

Hedda, 2021-01-07:
FYI, README file has now been updated with a warning about Zigbee to WiFi bridges:

https://github.com/zigpy/bellows/pull/386

"ZHA requires a robust connection between the zigpy radio library and the serial port of the Zigbee radio. With solutions such as _ITEAD Sonoff ZBBridge_ or a Ser2Net serial proxy connection over a WiFi network it is expected to see `NCP entered failed state. Requesting APP controller restart` in the logs. This is a normal part of the operation and indicates there was a drop in communication which caused packet loss to occurred between the zigpy radio library and the Zigbee radio. The use of serial network proxies/bridges/servers over WiFi is therefore not recommended when wanting a stable Zigbee environment in ZHA."

closed 2021-01-11 by Ricciolo