zigpy/bellows #392: Nortek HUSBZB-1 shows sometimes as offline

JMan7777, 2021-01-27:
Hello,

my ZigBee stick Nortek HUSBZB-1 shows sometimes as offline even it is my Coordinator.
How can it be that the Coordinator is shown as offline even the communication with all my other devices works fine?

[image]

**Device info**

EZSP = Silicon Labs EmberZNet protocol: Elelabs, HUSBZB-1, Telegesis
by ZHA 

FYI: I upgraded the Nortek HUSBZB-1 firmware some time ago, so it should have a recent firmware (I think I put v6.3 on it).

**This is my setup:**

Version | 2021.1.5
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
Virtual Environment | false
Python Version | 3.8.7
Operating System Family | Linux
Operating System Version | 5.4.79-v8
CPU Architecture | aarch64

Host Operating System | Home Assistant OS 5.10
Update Channel | stable
Supervisor Version | 2021.01.7
Docker Version | 19.03.13
Disk Total | 219.4 GB
Disk Used | 19.3 GB
Healthy | true
Supported | true
Board | rpi4-64
Supervisor API | ok
Version API | ok
Installed Add-ons | Samba share (9.3.0), SSH & Web Terminal (7.8.0),  Let's Encrypt (4.11.0), File editor (5.2.0), Glances (0.10.0), InfluxDB  (4.0.1), Grafana (6.0.0), APC UPS Daemon (1.9), Home Assistant Google  Drive Backup (0.103.1)

MattWestb, 2021-01-27:
I seeing the same phenomene with my "Billy EZSP" have RSSI N/A and LQI 255 all the time and being lazy to updating last  seen in device card and ZHA-Network-Card.
Its always being flagged online after the last seen is being updated.
It was 05:13 and being updated 09:15 so its around 4 hours.
Then the coordinator is one "normal" router with trustcenter i think the last seen is coming from then ZHA is doing the magic network scan (reading the neighbor table in all routers) tha shall happening in intervall 3-4 hours.
Was doing one "ZHA_MAP.scan_now" from ZHA services at 09:26 and the last seen and the network card was updated to last  seen to 09:26.

I seeing its some "personality" of the EZSP coordinators but its looking little not so nice on the device card. network-map, ZigZag and ZHA-Network Visualization.

Adminiuga, 2021-01-27:
you can disregard it being shown as offline. The actual device for coordinator is a fake device and sometimes ZHA is missinterpreting it's status, because it behaves very differently from every other device.

JMan7777, 2021-01-28:
@Adminiuga 
Ok, I can understand that the coordinator is just a fake device. 
But even so, shouldn't the coordinator only show as offline when the related hardware device (in my case Nortek HUSBZB-1) is not available or not responding.
In case the coordinator is functional it should be always show as online.

The main reason I created this issue is, that I want to react when one of my ZigBee devices goes offline.
So the offline state will be a trigger for me. 
I have some smoke sensors which from time to time drop off the ZigBee network and I need to pair them manually again.
Because of that I created an automation which informs me when a specific sensor drops out the network.

So even the coordinator is just a fake device it still should have the right status all the time.

Adminiuga, 2021-01-28:
When device is marked offline, zha sends an event. Listen to that event

JMan7777, 2021-01-28:
> When device is marked offline, zha sends an event. Listen to that event

Understand but nevertheless the coordinator should never go offline unless some problem with it.

Adminiuga, 2021-01-28:
Yeah, it's a bug.