zigpy/bellows #393: CLI async uart timeout errors

jeffleary00, 2021-02-06:
Hi,

Currently using the EleLabs Zigbee Raspberry Pi shield (RPi 4) for a custom project (Not using HomeAssistant or OpenHab)

The hardware device responds to firmware updates, and other basic interface commands. But bellows CLI is giving asyncio uart timeout errors on every command I try.
Advice on what may be wrong? Version conflicts?

**Python Version**
```
pi@raspberrypi:~ $ python3 --version
Python 3.7.3
```

**Raspberry Pi OS Details**
```
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

**Installed versions of bellows and zigpy**
```
pi@raspberrypi:~ $ pip3 show bellows zigpy
Name: bellows
Version: 0.21.0
Summary: Library implementing EZSP
Home-page: http://github.com/zigpy/bellows
Author: Russell Cloran
Author-email: rcloran@gmail.com
License: GPL-3.0
Location: /home/pi/ve/lib/python3.7/site-packages
Requires: voluptuous, zigpy, click-log, pure-pcapy3, pyserial-asyncio, Click
Required-by: 
---
Name: zigpy
Version: 0.32.0
Summary: Library implementing a ZigBee stack
Home-page: https://github.com/zigpy/zigpy
Author: Russell Cloran
Author-email: rcloran@gmail.com
License: GPL-3.0
Location: /home/pi/ve/lib/python3.7/site-packages
Requires: voluptuous, pycryptodome, crccheck, aiohttp, aiosqlite
Required-by: bellows
```

**Hardware device info**
```
pi@raspberrypi:~ $ python3 ./elelabs-zigbee-ezsp-utility/Elelabs_EzspFwUtility.py probe --port=/dev/ttyAMA0
2021/02/06 16:12:56 Elelabs_EzspFwUtility:   Elelabs adapter detected:
2021/02/06 16:12:56 Elelabs_EzspFwUtility:   Adapter: ELR023
2021/02/06 16:12:56 Elelabs_EzspFwUtility:   Firmware: 6.0.3-64
2021/02/06 16:12:56 Elelabs_EzspFwUtility:   EZSP v6
```

**Bellows command errors**
```
pi@raspberrypi:~ $ bellows --device=/dev/ttyAMA0 scan

Traceback (most recent call last):
  File "/home/pi/ve/bin/bellows", line 10, in <module>
    sys.exit(main())
[... 182 lines cut ...]
    return callback(*args, **kwargs)
  File "/home/pi/ve/lib/python3.7/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/ve/lib/python3.7/site-packages/bellows/cli/application.py", line 45, in form
    return util.app(inner, app_startup=False, extra_config=extra_config)(ctx)
  File "/home/pi/ve/lib/python3.7/site-packages/bellows/cli/util.py", line 76, in inner
    loop.run_until_complete(async_inner(*args, **kwargs))
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/home/pi/ve/lib/python3.7/site-packages/bellows/cli/util.py", line 62, in async_inner
    await f(ctx, *args, **kwargs)
  File "/home/pi/ve/lib/python3.7/site-packages/bellows/cli/application.py", line 43, in inner
    await app.startup(auto_form=True)
  File "/home/pi/ve/lib/python3.7/site-packages/bellows/zigbee/application.py", line 108, in startup
    self._ezsp = await bellows.ezsp.EZSP.initialize(self.config)
  File "/home/pi/ve/lib/python3.7/site-packages/bellows/ezsp/__init__.py", line 79, in initialize
    await ezsp.reset()
  File "/home/pi/ve/lib/python3.7/site-packages/bellows/ezsp/__init__.py", line 94, in reset
    await self._gw.reset()
  File "/home/pi/ve/lib/python3.7/site-packages/bellows/uart.py", line 218, in reset
    return await asyncio.wait_for(self._reset_future, timeout=RESET_TIMEOUT)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
```

Adminiuga, 2021-02-07:
elelabs is using 115200 baud rate by default, vs assumed 57600 used in bellows. Specify the baudrate in CLI

jeffleary00, 2021-02-11:
Ah! thanks for the baud rate suggestion. 
That was it. Set it to 115200 and it works perfectly now.
Thank You!!

closed 2021-02-11 by jeffleary00