zigpy/bellows #396: [REQUEST] Probe method feature with automatically probing of first 115200 baud rate and then 57600 baud rate?

Hedda, 2021-02-19:
Follow-up from discussion in https://github.com/walthowd/husbzb-firmware/issues/12

Feature request to extend bellows probe method feature with automatic probing function that first try probe with 115200 baud rate speed and then also try to probe 57600 baud rate speed but only if 115200 baud rate fail?

Reason for this request is that would make the initial installation config flow in ZHA even more user-friendly.

Many updated/newer EmberZNet NCP firmware support 57600 baud rate speeds, and that will reportedly soon also include the v6.7.8 firmware that will ship the upcoming inexpensive EFR32MG21 based "Zigbee 3.0 dongle" that ITead should soon start selling:

https://www.facebook.com/watch/?v=262086502015726

PS: By the way, bellows test code looks to use 115200 by default -> https://github.com/zigpy/bellows/blob/dev/tests/test_ezsp.py

Hedda, 2021-02-22:
Perhaps this request/suggestion should maybe be moved to zigpy issues at https://github.com/zigpy/zigpy/issues instead?

Adminiuga, 2021-02-23:
Submit PR.

pipiche38, 2022-08-08:
https://github.com/zigpy/bellows/pull/480

closed 2023-01-06 by puddly