zigpy/bellows #399: The minimum version of EZSP that can being used with bellows.

MattWestb, 2021-03-03:
Wot is the absolute lowest version of EZSP that is working (but likely not very god) with bellow ?

I have reading that protocol version 4 was implanted in  EmberZNet 4.7.2 SDK so is that the absolute minimum ??

E357X devices with "AT firmware" also need being upgraded to one supported firmware and most have getting EZSP 6.7.8.0 or is having one stable 6.5.0.0 that is working OK so shall not being any problems.

Perhaps is it one @walthowd question then hi have very much knowledge.

Thanks in advance.

Adminiuga, 2021-03-03:
Yes

MattWestb, 2021-03-03:
Thanks i updating the Zigpy Wiki with the absolute minimal and also the "AT firmware" and Sonoff need updating for working with bellows (but not recommended firmware version).

And giving tube013 one extra 🥇 (EZSP PRO) for his EZSP 6.9.1.0 firmware ,-)

closed 2021-05-01 by MattWestb