zigpy/bellows #401: TypeError: __init__() missing 1 required positional argument: 'message'

halkeye, 2021-03-09:
I just updated my HUSBZB-1 with https://github.com/walthowd/husbzb-firmware
```
root@6b85518ede3f:/tmp/silabs# ./ncp.py flash -p /dev/ttyUSB1 -f ncp-uart-sw-6.7.8.ebl
Restarting NCP into Bootloader mode...
CEL stick
EM3581 Serial Bootloader v5.4.1.0 b962

Successfully restarted into bootloader mode! Starting upload of NCP image...
Finished!
Rebooting NCP...
```

then trying to add a device gives me an error

```
sudo /home/halkeye/tmp/bellows/.direnv/python-3.8.0/bin/bellows -d /dev/ttyUSB1 scan
Scanning channels 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26
[EmberZigbeeNetwork(channel=11, panId=0xf918, extendedPanId=b9:70:0e:d8:92:74:72:9c, allowingJoin=<Bool.false: 0>, stackProfile=2, nwkUpdateId=0), 0, -86]
[EmberZigbeeNetwork(channel=11, panId=0xf918, extendedPanId=b9:70:0e:d8:92:74:72:9c, allowingJoin=<Bool.false: 0>, stackProfile=2, nwkUpdateId=0), 242, -84]
[EmberZigbeeNetwork(channel=11, panId=0x6f07, extendedPanId=ac:26:1e:bf:b9:e8:2d:d1, allowingJoin=<Bool.false: 0>, stackProfile=2, nwkUpdateId=0), 247, -84]
[EmberZigbeeNetwork(channel=25, panId=0xb7cf, extendedPanId=48:e5:88:43:93:a8:7f:7b, allowingJoin=<Bool.true: 1>, stackProfile=2, nwkUpdateId=0), 255, -72]
```

```
 sudo /home/halkeye/tmp/bellows/.direnv/python-3.8.0/bin/bellows -v DEBUG -d /dev/ttyUSB1 joi
n -E 48:e5:88:43:93:a8:7f:7b
debug: Using selector: EpollSelector
debug: Using selector: EpollSelector
debug: Connected. Resetting.
debug: Resetting EZSP
debug: Resetting ASH
debug: Sending: b'1ac038bc7e'
debug: RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
debug: Send command version: (4,)
debug: Sending: b'004221a850ed2c7e'
debug: Data frame: b'0142a1a85c2895d52e8b7e'
debug: Sending: b'8160597e'
debug: Application frame 0 (version) received: b'08028067'
debug: Switching to EZSP protocol version 8
debug: Send command version: (8,)
debug: Sending: b'7d314221a9542a1d8cdf7e'
debug: Data frame: b'1242a1a9542a1db0d9f389507e'
debug: Sending: b'82503a7e'
debug: Application frame 0 (version) received: b'08028067'
debug: EZSP Stack Type: 2, Stack Version: 6780, Protocol version: 8
debug: Configuring...
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_STACK_PROFILE: 12>, 2)
debug: Sending: b'224321a9072a19b05908037e'
debug: Data frame: b'2343a1a9072a1554577e'
debug: Sending: b'83401b7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
debug: Sending: b'334021a9072a7d38b759d60c7e'
debug: Data frame: b'3440a1a9072a15b4887e'
debug: Sending: b'8430fc7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
debug: Sending: b'444121a9072a38b359c1ad7e'
debug: Data frame: b'4541a1a9072a15cd287e'
debug: Sending: b'8520dd7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
debug: Send command setConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>, 64)
debug: Sending: b'554621a9072a14f25973327e'
debug: Data frame: b'5646a1a9072a15ea907e'
debug: Sending: b'8610be7e'
debug: Application frame 83 (setConfigurationValue) received: b'00'
Traceback (most recent call last):
  File "/home/halkeye/tmp/bellows/.direnv/python-3.8.0/bin/bellows", line 8, in <module>
    sys.exit(main())
  File "/home/halkeye/tmp/bellows/.direnv/python-3.8.0/lib/python3.8/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/halkeye/tmp/bellows/.direnv/python-3.8.0/lib/python3.8/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/halkeye/tmp/bellows/.direnv/python-3.8.0/lib/python3.8/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/halkeye/tmp/bellows/.direnv/python-3.8.0/lib/python3.8/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/halkeye/tmp/bellows/.direnv/python-3.8.0/lib/python3.8/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/halkeye/tmp/bellows/.direnv/python-3.8.0/lib/python3.8/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/halkeye/tmp/bellows/.direnv/python-3.8.0/lib/python3.8/site-packages/bellows/cli/util.py", line 39, in inner
    loop.run_until_complete(f(*args, **kwargs))
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/halkeye/tmp/bellows/.direnv/python-3.8.0/lib/python3.8/site-packages/bellows/cli/network.py", line 36, in join
    raise click.BadOptionUsage(
TypeError: __init__() missing 1 required positional argument: 'message'
```

I used this stick a long time ago so if there's any state maybe I need to reset it or something?

Adminiuga, 2021-03-10:
Yeah, that part is broken. It needs a channel on CLI and then you run into another bug.
What are you trying to do? Once HUSBZB joins an existing network as an End Device, bellows probably would be unusable, as it was not design to operate as a child device

Adminiuga, 2021-03-10:
make the following change to https://github.com/zigpy/bellows/blob/46fed049a3d85a90fe823f7c3f6cfabf3f9ceafa/bellows/cli/network.py#L86 line
```
    from zigpy.config import SCHEMA_NETWORK
    initial_security_state = zutil.zha_security(SCHEMA_NETWORK({}))
```

then run your CLI command but also specify the channel with `-C 25` parameter

halkeye, 2021-03-10:
> Yeah, that part is broken. It needs a channel on CLI and then you run into another bug.

Okay cool

> What are you trying to do? Once HUSBZB joins an existing network as an End Device, bellows probably would be unusable, as it was not design to operate as a child device

honestly? i was just poking around at things. My blinds have a zigbee controller but I hadn't yet (just now) got pairing instructions, so was just trying out bellows.

Adminiuga, 2021-03-10:
Bellows is primarily used as a "supporting" radio lib for the [Home Assistant](https://github.com/home-assistant/core)  [ZHA integration](https://www.home-assistant.io/integrations/zha/)

Depending on the blinds, you could form a new zigbee network on HUSBZB-1 and join the blinds to this new network.  I think some people are using `bellows` or other radio libs, running it in a standalone application. Unfortunately no good documentation other than the code itself.

halkeye, 2021-03-10:
all good, i'll probably move the hub directly to HA, just wanted to test things out first. Still trying to flash a sonoff zigbee bridge as well to try that out.

learning++

I just wanted to report the bug when i encountered it

PR #404 (2021-03-13): CLI fix for joining as an end device
FIX CLI joining EZSP as an end device.
Fixes #401
closed 2021-03-13 by Adminiuga