zigpy/bellows #414: Support for multiple restore from backup could be prevented in adapter firmware?

Hedda, 2021-04-30:
Is it true that Elelabs adapters only allow a single restore from backup and if so should a warning be added to readme? 

https://github.com/zigpy/bellows#nvram-backup-and-restore

See this Jeedom blog post (translated from French to English with Google):

https://blog.jeedom.com/5183-tout-ce-quil-faut-savoir-sur-le-plugin-officiel-zigbee/

"_This backup can be restored on another controller of the same type. **Be careful, however, some types of controllers such as EZSP Elelabs only accept a single restoration procedure during their entire life .**_"

MattWestb, 2021-04-30:
It the normal thing with all EFR32 and EM35X devices.
You can writing in token on time then you need doing on chip erase (or the specific flash page) and you can writing on new token.

If you is having on SWD flasher is no problems but for normal user is on one time operation.

Adminiuga, 2021-04-30:
https://media.giphy.com/media/SVgKToBLI6S6DUye1Y/giphy.gif

You can restore backup as many times as you want. You can override EUI64 only once, unless you have a jlink

closed 2021-04-30 by Adminiuga