zigpy/bellows #425: EZSP 6.10.0.0 problem with bellows dump

MattWestb, 2021-06-30:
I have one LIDL / tuya LED stripe controller (with one nice MG21 module) up and running with EZSP 6.10.0.0 and its looks working OK but its my "second test network" so for the moment its only having one IKEA E27 CWS3 and is not good tested but looks working OK.
But then i is using bellow dump i is getting `Bad FCS` in the 15.4 layer on every package:
```
Frame 1: 59 bytes on wire (472 bits), 59 bytes captured (472 bits)
IEEE 802.15.4 Data, Dst: Broadcast, Src: 0xae2b, Bad FCS
    Frame Control Field: 0x8841, Frame Type: Data, PAN ID Compression, Destination Addressing Mode: Short/16-bit, Frame Version: IEEE Std 802.15.4-2003, Source Addressing Mode: Short/16-bit
    Sequence Number: 212
    Destination PAN: 0xd337
    Destination: 0xffff
    Source: 0xae2b
    FCS: 0x000f (Incorrect, expected FCS=0x1bdf)
    [Expert Info (Warning/Checksum): Bad FCS]
Data (48 bytes)
```
I have testing with one billy but that NCP is not tested then its maxed with 127 TC Link keys and GP support so its on the limit for the old bookshelf and the same maxed for tuya ZBGW (with MG1B chip) and looks working but nearly not tested but is giving error same then dumping with bellows.

I attaching one very small sniffing (network key not needed then its in 15.4 layer = not secured).
[dmp11.zip](https://github.com/zigpy/bellows/files/6740998/dmp11.zip)

I is doing more testing but its looks like working OK with com.zsmartsystems.zigbee.sniffer but i is doing little more testing so taking it as one "rep alert" and if i getting it working OK i closing or deleting this issue.

Thanks in advance !!

Adminiuga, 2021-06-30:
https://www.silabs.com/community/wireless/zigbee-and-thread/forum.topic.html/no_static_fcs_inmfglibrxframes-YwGU

MattWestb, 2021-06-30:
Thanks for info !!

I was working with wireshark with all versions also 6.9.2.0 but i have updating wireshark for 3 weeks ago and its have starting doing the checks and flagging the packages bad.
I remember that 6.9.x is not sending 15.4 acks to the sniffer that is being blocked in the EZSP stack but is working in 6.7.x.

So until / if  SIlab is fixing it is 6.7.x with wireshark or using PTI on the WSTK.

Closing the issue then its not one bellows issue and is good for info for other users.

closed 2021-06-30 by MattWestb

MattWestb, 2021-07-15:
For information is sniffing with one EZSP six ten working OK with `com.zsmartsystems.zigbee.sniffer` with local comport and wireshark:

```
PS C:\msys32\home\Mattias\com.zsmartsystems.zigbee.sniffer> java -jar ZigBeeSniffer.jar -port COM16  -baud 115200 -flow software -c 15
Z-Smart Systems Ember Packet Sniffer
NCP initialisation starting...
Opened serial port COM16 at 115200
DEBUG  17:01:42.103  Opening port COM16 at 115200 baud with FLOWCONTROL_OUT_XONOFF.
DEBUG  17:01:42.395  ASH: Connect
DEBUG  17:01:42.395  ASH: Reconnect
DEBUG  17:01:42.395  AshFrameHandler thread started
DEBUG  17:01:42.395  --> TX ASH frame: AshFrameRst []
DEBUG  17:01:42.623  TX EZSP: EzspVersionRequest [networkId=0, desiredProtocolVersion=4]
DEBUG  17:01:42.623  ASH: TX EZSP queue size: 1
DEBUG  17:01:43.657  <-- RX ASH frame: AshFrameRstAck [version=2, resetCode=11, Reset: Software]
DEBUG  17:01:44.915  ASH: Connected
DEBUG  17:01:44.915  Ember: Link State change to true, initialised=false, networkStateUp=false
DEBUG  17:01:44.915  Ember: Link State change to true ignored.
DEBUG  17:01:44.915  --> TX ASH frame: AshFrameData [frmNum=0, ackNum=0, reTx=false, data=00 00 00 04]
DEBUG  17:01:44.915  <-- RX ASH frame: AshFrameData [frmNum=0, ackNum=1, reTx=false, data=00 80 00 08 02 00 6A]
DEBUG  17:01:44.915  ASH: Frame acked and removed AshFrameData [frmNum=0, ackNum=0, reTx=false, data=00 00 00 04]
DEBUG  17:01:44.932  EzspVersionResponse [networkId=0, protocolVersion=8, stackType=2, stackVersion=6A00]
DEBUG  17:01:44.932  RX EZSP: EzspVersionResponse [networkId=0, protocolVersion=8, stackType=2, stackVersion=6A00]
DEBUG  17:01:44.933  TX EZSP: EzspVersionRequest [networkId=0, desiredProtocolVersion=8]
DEBUG  17:01:44.933  --> TX ASH frame: AshFrameAck [ackNum=1, notRdy=false]
DEBUG  17:01:44.933  ASH: TX EZSP queue size: 1
DEBUG  17:01:44.933  --> TX ASH frame: AshFrameData [frmNum=1, ackNum=1, reTx=false, data=01 00 01 00 00 08]
DEBUG  17:01:44.933  <-- RX ASH frame: AshFrameData [frmNum=1, ackNum=2, reTx=false, data=01 80 01 00 00 08 02 00 6A]
DEBUG  17:01:44.933  ASH: Frame acked and removed AshFrameData [frmNum=1, ackNum=1, reTx=false, data=01 00 01 00 00 08]
DEBUG  17:01:44.933  RX EZSP: EzspVersionResponse [networkId=0, protocolVersion=8, stackType=2, stackVersion=6A00]
DEBUG  17:01:44.933  --> TX ASH frame: AshFrameAck [ackNum=2, notRdy=false]
DEBUG  17:01:44.933  EzspVersionResponse [networkId=0, protocolVersion=8, stackType=2, stackVersion=6A00]
DEBUG  17:01:44.933  EzspVersionResponse [networkId=0, protocolVersion=8, stackType=2, stackVersion=6A00]
DEBUG  17:01:44.933  TX EZSP: EzspGetStandaloneBootloaderVersionPlatMicroPhyRequest []
DEBUG  17:01:44.933  ASH: TX EZSP queue size: 1
DEBUG  17:01:44.933  --> TX ASH frame: AshFrameData [frmNum=2, ackNum=2, reTx=false, data=02 00 01 91 00]
DEBUG  17:01:44.947  <-- RX ASH frame: AshFrameData [frmNum=2, ackNum=3, reTx=false, data=02 80 01 91 00 00 1C 04 18 0F]
DEBUG  17:01:44.947  ASH: Frame acked and removed AshFrameData [frmNum=2, ackNum=2, reTx=false, data=02 00 01 91 00]
DEBUG  17:01:44.947  RX EZSP: EzspGetStandaloneBootloaderVersionPlatMicroPhyResponse [networkId=0, bootloaderVersion=1C00, nodePlat=4, nodeMicro=24, nodePhy=15]
DEBUG  17:01:44.947  --> TX ASH frame: AshFrameAck [ackNum=3, notRdy=false]
Ember NCP version     : 6.10.0.0
DEBUG  17:01:44.956  TX EZSP: EzspGetEui64Request []
DEBUG  17:01:44.956  ASH: TX EZSP queue size: 1
DEBUG  17:01:44.956  --> TX ASH frame: AshFrameData [frmNum=3, ackNum=3, reTx=false, data=03 00 01 26 00]
DEBUG  17:01:44.956  <-- RX ASH frame: AshFrameData [frmNum=3, ackNum=4, reTx=false, data=03 80 01 26 00 A0 94 80 FE FF 57 B4 14]
DEBUG  17:01:44.956  ASH: Frame acked and removed AshFrameData [frmNum=3, ackNum=3, reTx=false, data=03 00 01 26 00]
Ember NCP EUI         : 14B457FFFE8094A0
DEBUG  17:01:44.956  RX EZSP: EzspGetEui64Response [networkId=0, eui64=14B457FFFE8094A0]
DEBUG  17:01:44.956  --> TX ASH frame: AshFrameAck [ackNum=4, notRdy=false]
DEBUG  17:01:44.956  TX EZSP: EzspMfglibStartRequest [networkId=0, rxCallback=true]
DEBUG  17:01:44.972  ASH: TX EZSP queue size: 1
```
The sniff in the IKEA Styrbar (E2001/E2002 remote) https://github.com/zigpy/zha-device-handlers/pull/867#issuecomment-880766467 is made with it so bellows is making somthing different (i dont saying wrong!! ) but one tuya ZBGW with the same version of EZSP is not working with bellows.