zigpy/bellows #431: How to run Zigbee co-ordinator application based on ZM35X in Linux RPI4 platform

RoyAnupam, 2021-07-27:
Hi,
I am trying to build a Zigbee Co-Ordinator application in my RPI4 running Linux platform.
I am planning to buy a ZM35XX Zigbee USB dongle for this purpose.
[https://www.allicdata.com/products/zm3588s-usb/2914948.html]
 
But in my understanding, I would also need a Host software stack running on my RPI, in order to talk to the Zigbee USB module.
I figured out that, there is EmberZNet Stack from SILabs which can be used for my purpose, but I don't have access to it currently.
Then, I happened to bump into  zigpy/bellows python based stack. Is it possible to run this bellows stack on RPI4 and run Zigbee Co-ordinator functionality in my ZM35XX USB module? Kindly share some insights.

Thank You very much in advance~

BR,

MattWestb, 2021-07-27:
Zigpy / bellows runs on PI if you is having python 3.9 installed or in one docker container like HA is doing.

The question is if your USB dongle is running Silabs EZSP zigbee stack coordinator firmware (EZSP NCP) (one must for Bellows) or if its having the proprietary MeshConnect firmware.

Also with EZSP NCP firmware you can using it in all open systems that is supporting SIlcabs Zigbee chips like ZHA, Tasmota and Z2M.

We have getting firmware for EM357 and EM3581 but not for the EM3588 chip devices and its needs much knowledge plus licence for the zigbee stack (Silabs EZSP 100 $ some devs is having it) and IAR compiler that is costing over 3500 $ and its tricky compiling then support for the chip is dropped.

I think its better putting the money on one coordinator that is not flagged end off life for doing more serous things but i cant recommending Sonoff coordinators then they is having so much hardware problems (not calibrated radios from the factory that is one must and so on).

One alternative is one coordinator that is network connected and is using serial to net for communication. Its little more expensive but is working great (in ZHA).

USB dongles and PI-HAT with EFR32 chips is not so expensive and is working OK.

Adminiuga, 2021-07-27:
Yes, you could, but you would need to add some code on top of bellows. Bellows per se is a low level library for EZSP protocol. This code exist as part of ZHA integration in Home Assistant Core. 
Take a look at zigpy/zigpy-cli as well

RoyAnupam, 2021-07-30:
@MattWestb
> The question is if your USB dongle is running Silabs EZSP zigbee stack coordinator firmware (EZSP NCP) (one must for Bellows)or if its having the proprietary MeshConnect firmware.

Okay, It means, the dongle must be pre-flashed with EZSP NCP firmware for bellows to be able to use it?
I noticed link to update firmware in dongles in project Wiki page [https://github.com/walthowd/husbzb-firmware](url)
So, it means, we have access to the EZSP NCP firmware's for the compatible radios, but, I was wondering whether any additional hardware is required to flash the firmware into the radio? Any idea?

> Also with EZSP NCP firmware you can using it in all open systems that is supporting SIlcabs Zigbee chips like ZHA, Tasmota and Z2M.

Okay, got it. Actually, my intention was to use bellows as a standalone library instead of using HomeAssistant OS with ZHA integration. 

>USB dongles and PI-HAT with EFR32 chips is not so expensive and is working OK.

Okay I see. Does it work out of the box  with bellows? Could you kindly share some Wiki info for above hardware?

Thank You very much

RoyAnupam, 2021-07-30:
@Adminiuga 
> Yes, you could, but you would need to add some code on top of bellows. Bellows per se is a low level library for EZSP protocol. This code exist as part of ZHA integration in Home Assistant Core.

Thank you for your reply. Okay I got it. Actually, my intention was to use bellows as a standalone library instead of using HomeAssistant OS with ZHA integration. So If I understand  correctly, I would need to develop some  use case scenarios on top of bellows library (like Create Network, Permit Join, Add Device, Browse Endpoints, Control operations etc.) ? Please share your opinion. Thanks

BR,

Adminiuga, 2021-07-30:
Yes, it is possible to be standalone, but you do have to add this layer on top and essentially implement what zha integration does. 
For example start the radio library, initialize radio, open network for joining, wait till a device joins and for example the if the device has an on_off cluster then present the user with a light bulb. And if the user clicks on the bulb, then send the toggle command to on_off cluster

MattWestb, 2021-07-30:
In the Zigpy wiki you can also finding all adapters /  firmware we have funding working with Bellows in the same page but under `Recommended firmware versions for EZSP coordinators`.
EByte have proprietary firmware and need SWD probe for being flashed one time (new bootloader) all other is having one working bootlooader (if not written somthing else in the wiki) and can being updated  with one terminal with X-Mode.

Small USB and PI-HAT i think only Elelabs have making good working ones.
One link: https://elelabs.com/shop/ (can also being found on other market pleases like Amazon).

And updated firmware is made of the community and also updating program for users (linked in the wiki).

With PI-HAT its needed doing some extra setting in the PIs OS for getting  hardware UART working OK but its easy made.

RoyAnupam, 2021-08-04:
> Yes, it is possible to be standalone, but you do have to add this layer on top and essentially implement what zha integration does.
> For example start the radio library, initialize radio, open [...]

Thank you for sharing your suggestions!

RoyAnupam, 2021-08-04:
> In the Zigpy wiki you can also finding all adapters / firmware we have funding working with Bellows in the same page but under `Recommended firmware versions for EZSP coordinators`.
> EByte have [...]

Okay great. Thank you for sharing your suggestions. Can the following Zigbee USB dongle work with Bellows?
https://elelabs.com/products/elelabs-usb-adapter.html

> 
> And updated firmware is made of the community and also updating program for users (linked in the wiki).
> 
> With PI-HAT its needed doing some extra setting in the PIs OS for getting hardware UART working OK but its easy made.
> 
> Edit: PI4 its important getting USB devices away from the USB-Zigbee dongle then the USB3 is making strong interference with Zigbee radio so normal one good extension cable is needed for getting the Zigbee dongle working OK.

MattWestb, 2021-08-04:
Some devs is / have using it and Elelabs have making one  update utility for updating the firmware [Elelabs EZSP Firmware Update Utility](https://github.com/Elelabs/elelabs-zigbee-ezsp-utility#flash--perform-firmware-update-on-the-generic-ezsp-product) and also released updated firmware in the same git.

And the community have cooking more updated firmware for the device that is  also working OK [Compatible with Elelabs-ELU013 and ELR023](https://github.com/zha-ng/EZSP-Firmware/tree/master/Elelabs-ELU013).

I have seen one clone of this device on Amazon but i dont remember the name they was using.

RoyAnupam, 2021-08-04:
> Some devs is / have using it and Elelabs have making one update utility for updating the firmware [Elelabs EZSP Firmware Update [...]

Okay sure. I got it. Thanks.
Do the https://github.com/zigpy/bellows github repo contain any command-line utility tool to do basic Zigbee Co-ordinator operations based on bellows library? I want to take reference for developing my own application

MattWestb, 2021-08-04:
https://github.com/zigpy/bellows/tree/dev/bellows/cli is the CLI part and you can getting online help like `belows info --help`.

RoyAnupam, 2021-08-04:
> https://github.com/zigpy/bellows/tree/dev/bellows/cli is the CLI part and you can getting online help like `belows info --help`.
> 
> I think most NCP ops is defined (but not all) in the `application.py`.

Thank you very much!

Hedda, 2021-08-09:
> Take a look at zigpy/zigpy-cli as well

As mentioned, also check out the new zigpy-cli at https://github.com/zigpy/zigpy-cli as maybe could extend that to fit your needs.

> Actually, my intention was to use bellows as a standalone library instead of using HomeAssistant OS with ZHA integration.

A tip could be to also see the code from Jeedom Official Zigbee Plugin based on zigpy (access requires purchasing their plugin):

https://github.com/zigpy/zigpy/discussions/725

Again, also as already mentioned, another tip is to not use a Zigbee coordinator with SoC that is end-of-life so buy newer hardware.

> I have seen one clone of this device on Amazon but i dont remember the name they was using.

POPP ZB-Stick and POPP ZB-Shield are OEM versions of Elelabs ELU013 and ELR023 sold in Europe  (exact same hardware):

https://www.popp.eu/products/

https://elelabs.com/shop/

Alternative Silabs Zigbee coordinator hardware is Tube's Zigbee Gateways (Silicon Labs EFR32 variant) though stock is low:

https://github.com/tube0013/tube_gateways

https://www.tubeszb.com/shop/2

RoyAnupam, 2021-08-12:
> > Take a look at zigpy/zigpy-cli as well
> 
> As mentioned, also check out the new zigpy-cli at https://github.com/zigpy/zigpy-cli as maybe could extend that to fit your needs.
> 
> > Actually, my intention was to use bellows as a standalone library instead of using HomeAssistant OS with ZHA integration.
> 
> A tip could be to also see the code from Jeedom Official Zigbee Plugin based on zigpy (access requires purchasing their plugin):
> 
> [zigpy/zigpy#725](https://github.com/zigpy/zigpy/discussions/725)
> 
> > I am planning to buy a ZM35XX Zigbee USB dongle for this purpose.
> 
> Again, also as already mentioned, another tip is to not use a Zigbee coordinator with SoC that is end-of-life so buy newer hardware.
> 
> Silabs list ZM35XX as "Not Recommended for New Designs (NRND)" so try to find a newer dongle with EFR32MG12 or EFR32MG21.
> 
> > I have seen one clone of this device on Amazon but i dont remember the name they was using.
> 
> POPP ZB-Stick and POPP ZB-Shield are OEM versions of Elelabs ELU013 and ELR023 sold in Europe (they are exact same hardware):
> 
> https://www.popp.eu/products/
> 
> https://elelabs.com/shop/
> 
> Alternative Silabs Zigbee coordinator hardware is Tube's Zigbee Gateways (Silicon Labs EFR32 variant) though stock is low:
> 
> https://github.com/tube0013/tube_gateways
> 
> https://www.tubeszb.com/shop/2

Thank You very much for your kind reply & suggestions! I will check these in detail~

RoyAnupam, 2021-08-25:
Hi, I have a basic question about Zigbee onboarding in Home Assistant setup. Actually, I never used HA, but, just being curios to know how it works! Does Zigbee 3.0 device onboarding mandate QR code for transferring the install code from *Thing device* to *Trust Center* device? I am not sure, if this is right forum to ask this question. But, if you have some insight about this, it will be highly appreciated.

MattWestb, 2021-08-25:
ZHA is normally using the well known HA key for joining devices (Zigbee 1.x and 3) but its possible adding install code with services in HA for joining devices with ( have not trying it but some deva and user have using it).
There is no GUI for scanning QR-codes but you can do it and pasting the code in the service and it shall working.

Hedda, 2021-08-25:
> There is no GUI for scanning QR-codes but you can do it and pasting the code in the service and it shall working.

Exactly, underlying support is there but there is no GUI or app for scanning QR codes to get install codes into Home Assistant, see:

https://community.home-assistant.io/t/request-qr-code-tags-scanning-for-joining-new-zigbee-3-0-device-to-home-assistant-zha-integration-component-via-ui-and-companion-apps/231185

RoyAnupam, 2021-08-25:
Thank You @Hedda  & @MattWestb for quick response...

Hedda, 2021-12-01:
FYI, the Z-Wave JS integration UI will soon implement a QR scanner in Home Assistant's frontend code that will be used for Z-Wave SmartStart provisioning (which is similar to Zigbee 3.0 device inclusion via QR code), and sounds as if that QR scanner component will be implemented as a separate part that could potentially be a reusable element that the ZHA UI could use for QR code scanning? More information is posted here -> https://github.com/zigpy/zigpy/discussions/857

Pmctst, 2022-11-04:
@RoyAnupam 

Hi I am starting with a development I think similar to what you tried to achieve. Did you finally created an application on top of Bellows library? Could you give me some tips about how to interact for creating a coordinator or a end device and managing messages.

I think I have understood how EZSP protocol works but I have not a clear view of how to develop an application on top of that. Should it be using callbacks?

Hedda, 2023-05-19:
@Pmctst may be a little too late but check out the related question/request and answers/discussion here -> https://github.com/zigpy/zigpy/issues/1087