zigpy/bellows #444: Add Child, Remove Child, Remove Neighbor via direct manipulation of child/neighbor table by the application?

Hedda, 2021-12-07:
Direct manipulation of child/neighbor table by the application to "Add Child", "Remove Child", "Remove Neighbor" might be nice to have for a few reasons but I think that it would be especially nice to be able to configure some devices to not use some device as a router(?).

All new API feature added Silabs Zigbee EmberZNet 6.10.0.0 which might be nice to consider for zigpy-cli/bellows for ZHA integration?

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-6.10.3.0.pdf

### Added in release 6.10.0.0

_**emberAddChild()** - SoC-only API for adding a child to the child/neighbor table, allowing direct manipulation of these tables by the application._

_**emberRemoveChild()** - SoC-only API for removing a node from child/neighbor table, allowing direct manipulation of these tables by the application._

_**emberRemoveNeighbor()** - SoC-only API for removing a node from neighbor table, allowing direct manipulation of these tables by the application._

MattWestb, 2021-12-07:
I think you need understanding what "SoC-only API" is then its not EZSP commands API.

Adminiuga, 2021-12-07:
Closing as it has nothing to do with EZSP

closed 2021-12-08 by Hedda