zigpy/bellows #446: Low Transmission Power

alelucc, 2022-01-23:
Hi,
I'm using an Eachen (eWelink) Zigbee Bridge Pro ZB-GW03 with syssi EspHome firmware [https://github.com/syssi/esphome-zb-gw03](url)  within Home Assistant using the native ZHA integration.
It's working great but looking into HA logs i noticed that the radio module was set with a very low Tx power (8).
`2022-01-23 15:49:36 INFO (MainThread) [bellows.zigbee.application] Node type: EmberNodeType.COORDINATOR, Network parameters: EmberNetworkParameters(extendedPanId=3c:08:6f:60:fe:2a:94:60, panId=0x3f21, radioTxPower=8, radioChannel=25, joinMethod=<EmberJoinMethod.USE_MAC_ASSOCIATION: 0>, nwkManagerId=0x0000, nwkUpdateId=0, channels=<Channels.ALL_CHANNELS: 134215680>)`.

In some HA discussion i saw that some users having the same parameter set to 20. `2021-01-25 09:04:00 INFO (MainThread) [bellows.zigbee.application] Node type: EmberNodeType.COORDINATOR, Network parameters: EmberNetworkParameters(extendedPanId=cc:cc:cc:cc:e3:ab:00:78, panId=0x3498, radioTxPower=20, radioChannel=11, joinMethod=<EmberJoinMethod.USE_MAC_ASSOCIATION: 0>, nwkManagerId=0x0000, nwkUpdateId=0, channels=<Channels.ALL_CHANNELS: 134215680>)`
[https://community.home-assistant.io/t/zha-zigbee-change-channel/70396/63](url)

Looking trough code i see that TxPower is hard coded with the 8 value. Is it the maximum value?
If not, it would be great if was possible to configure trough HA like the `CONF_NWK_CHANNEL`
`zha:
  zigpy_config:
    network:
      channel: 15             # What channel the radio should try to use.
      channels: [15, 20, 25]  # Channel mask`

Hedda, 2022-01-31:
Do not know for ezsp (bellows) but for znp (zigpy-znp) it uses firmware default unless set tx_power via ZHA YAML configuration:

https://github.com/zigpy/zigpy-znp/blob/dev/README.md#configuration

```
zha:
  zigpy_config:
    znp_config:
      # "auto" picks the largest value that keeps the device's transmit buffer from getting full
      max_concurrent_requests: auto

      # Only if your stick has a built-in power amplifier (i.e. CC1352P and CC2592)
      # If set, must be between:
      #  * CC1352/2652:  -22 and 19
      #  * CC253x:       -22 and 22
      tx_power:  
```

So if same command is available for ezsp in ZHA and in the bellow library then should be something like this:

https://www.home-assistant.io/integrations/zha#configuration---yaml

```
zha:
  zigpy_config:
    ezsp_config:
      tx_power: 20
```

Default in firmware will probably be set to 10 dBm or less than that FCC and CE certification regulation does not allow higher.

Silicon Labs EFR32MG21 hardware does, however, support up to 20 dBm TX power but user need to set it themself if not default.

syssi, 2022-01-31:
@Hedda At ezsp (bellows) the tx_power is hardcoded at the moment: https://github.com/zigpy/bellows/blob/dev/bellows/zigbee/application.py#L233

overas, 2023-11-28:
I think Home Assistant SkyConnect also uses firmware default for tx_power. I tried to replace my ZigStar Stick v4 (with TI CC2652P) coordinator with SkyConnect, but with the low tx_power the batterypowered devices was troublesome to pair with ZHA. I also had lots of "EmberStatus.DELIVERY_FAILED: 102" messages in the log and lights which did not shut off or on.

I use these setting for replacing TI CC2652P firmware default to 20 dB:
```
zha:
  zigpy_config:
    znp_config:
      tx_power: 20
```

MattWestb, 2023-11-28:
Look on the network map you must have one good connection thru all routers and if you is having one week spot you  need moving or adding more routers.
Also its one mesh network so one longer way can being better then shorter one but normal 3 good links from the coordinator and then its the "second level".