zigpy/bellows #450: Domoticz-Zigbee plugin integration

SylvainPer, 2022-03-03:
Hello,
I'm working on the bellows integration in the plugin.

I have some questions first about the config part that is quite different from the others libraries and not sure everything is fine.

- There is a mix in the code between bellows/zigpy, for the init, the function is awaiting a bellows.config.CONFIG_SCHEMA with path and baudrate for example. 
For the network config, CONF_NWK is not imported in bellows.config and all the references in code are using zigpy.config. I can't configure the channel nor the ext pan id with this structure (config can't be both bellows and zigpy type).

- My network is well initialized (answer 0x90 on the stackStatusHandler) but if I try to form a new one, I end up with an exception with the setInitialSecurityState call as the answer is 0x70. Maybe due to the point before.

This is the first steps, still a lot to do.

Adminiuga, 2022-03-03:
Before you can form a new network, you need to "leave" the current one. 
Regarding the config, radio libraries may extend the zigpy base config. IIRC bellows doesn't change the network config params, and only adds a few specific belloes options and hardware flow control for the device config.

To change panid, ext pan id etc, you need to form a new network. Channel could be set during formation as well, or could be changed later, but it is more convoluted. You could check zha-toolkit by mdeweerd, in ezsp.py there's ezsp_set_channel() function which changes the channel

pipiche38, 2022-03-03:
How do you leave the network ?

SylvainPer, 2022-03-03:
Many thanks for your answer @Adminiuga.

@pipiche38 there is an ezsp command : leaveNetwork(). I join you on the standardisation need, it's not required by other libs and should probably added in the form_network() function, @puddly is there something that could meet the new-radio-api work ?

@Adminiuga Bellows config doesn't extend the zigpy config for CONF_NWK. If I try to add it, I have an error. 
I need to both import bellows.config and zigpy.config and mix the two to build the config, it's easier with znp as it import it.
Here the znp config import part:
`from zigpy.config import (  # noqa: F401
    CONF_NWK,
    CONF_DEVICE,
    CONF_NWK_KEY,
    CONFIG_SCHEMA,
    SCHEMA_DEVICE,
    CONF_NWK_PAN_ID,
    CONF_DEVICE_PATH,
    CONF_NWK_CHANNEL,
    CONF_NWK_CHANNELS,
    CONF_NWK_UPDATE_ID,
    CONF_NWK_TC_ADDRESS,
    CONF_NWK_TC_LINK_KEY,
    CONF_NWK_EXTENDED_PAN_ID,
    cv_boolean,
)`

For the channel, we can effectively change it on live and we already do it with zigate for example but first I need to do it properly at the init phase.

Thanks for the support.

puddly, 2022-03-03:
> How do you leave the network ?

There's no standard way to leave the network.  With the new radio API, you can call `self.form_network()`, it is supposed to erase all settings and form a new one regardless of whether there is a network formed or not.

> is there something that could meet the new-radio-api work ?

It's already there, `write_network_settings` will always succeed: https://github.com/puddly/bellows/blob/a4bac09c27c63de8588eb0527fb75b339ef397fe/bellows/zigbee/application.py#L272-L283

> For the channel, we can effectively change it on live and we already do it with zigate for example but first I need to do it properly at the init phase.

Bellows will need to be updated to support the `app.get_device(nwk=0x0000).zdo.Mgmt_NWK_Update_req(...)` interface if you want to change the channel at runtime and notify other devices. It does not support loopback ZDO requests and instead requires these commands to be translated to EZSP commands.

pipiche38, 2022-03-03:
> Bellows will need to be updated to support the `app.get_device(nwk=0x0000).zdo.Mgmt_NWK_Update_req(...)` interface if you want to change the channel at runtime and notify other devices. It does not support loopback ZDO requests and instead requires these commands to be translated to EZSP commands.

@puddly, was it the only one we had on ZNP, I feel they were other as well, no  ?
Node Descriptor 
Management LQI Request

puddly, 2022-03-03:
All others are supported without any extra code.  Z-Stack has a bug where it "responds" to the `Mgmt_NWK_Update_req` for changing the channel but does not actually change the channel, so that specific `Mgmt_NWK_Update_req` needs to be intercepted and handled separately.

SylvainPer, 2022-03-04:
> > How do you leave the network ?
> 
> There's no standard way to leave the network. With the new radio API, you can call `self.form_network()` (really `write_network_info`), it is supposed to erase all settings and form a new one regardless of whether there is a network formed or not.
> 
> > is there something that could meet the new-radio-api work ?
> 
> It's already there, `write_network_info` will always succeed: https://github.com/puddly/bellows/blob/a4bac09c27c63de8588eb0527fb75b339ef397fe/bellows/zigbee/application.py#L272-L283

Thanks, I was looking for form_network() and as I didn't see it, I thought that it was not yet implemented but write_network_info() seems to cover it.

SylvainPer, 2022-03-07:
I'm working with a schneider wiser device and try to add a new endpoint (0x0b). The tests are quite conclusive but when I checked the log, I found this:

2022-03-07 17:34:09,360 INFO    :[0x0000:11] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=11, profile=260, device_type=48879, device_version=0, input_clusters=[], output_clusters=[])
2022-03-07 17:34:09,360 INFO    :[0x0000] Read model None and manufacturer None from <Endpoint id=1 in=[] out=[ias_zone:0x0500] status=<Status.ZDO_INIT: 1>>
2022-03-07 17:34:09,361 INFO    :[0x0000] Read model None and manufacturer None from <Endpoint id=11 in=[] out=[] status=<Status.ZDO_INIT: 1>>
2022-03-07 17:34:09,361 INFO    :[0x0000] Discovered basic device information for <Device model=None manuf=None nwk=0x0000 ieee=84:fd:27:ff:fe:82:2c:8a is_initialized=True>
2022-03-07 17:34:09,361 DEBUG   :Device is initialized <Device model=None manuf=None nwk=0x0000 ieee=84:fd:27:ff:fe:82:2c:8a is_initialized=True>
2022-03-07 17:34:09,362 DEBUG   :Checking quirks for None None (84:fd:27:ff:fe:82:2c:8a)
2022-03-07 17:34:09,362 DEBUG   :Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2022-03-07 17:34:09,362 DEBUG   :Fail because endpoint list mismatch: {1} {1, 11}
2022-03-07 17:34:09,362 DEBUG   :Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2022-03-07 17:34:09,362 DEBUG   :Fail because endpoint list mismatch: {1} {1, 11}
2022-03-07 17:34:09,362 DEBUG   :Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2022-03-07 17:34:09,363 DEBUG   :Fail because endpoint list mismatch: {1} {1, 11}
2022-03-07 17:34:09,363 DEBUG   :Considering <class 'bellows.zigbee.application.EZSPCoordinator'>
2022-03-07 17:34:09,363 DEBUG   :Fail because endpoint list mismatch: {1} {1, 11}

Do I have to do something more or it's normal ?

puddly, 2022-03-07:
Is it necessary for you to create the `0x0B` endpoint?  The new endpoint is causing the coordinator quirk to not match, which will prevent `EZSPEndpoint.add_to_group` from being created.  Regardless, I've pushed a fix for this problem to the new radio API branch: https://github.com/zigpy/bellows/pull/445/commits/eb0096745d9ce4a983c48a25d33394fe355b2eb2

SylvainPer, 2022-03-07:
Thanks for the update !

In fact, we create more endpoints for specific manufacturer:
0x0b: Schneider Wiser
0x03: New Schneider Wiser
0x0a: Orvibo
0x6e: Terncy
0x15: Konke
0x08: Livolo

I didn't try all of them yet with bellows but it's working with znp.

pipiche38, 2022-03-07:
Because those brands do not expect the Coordinator on EndPoint 01, but are hardcoded on others

pipiche38, 2022-03-07:
On Zigate, this is handle at the firmware level, on ZNP we are creating those EndPoint on the Coordinator side (zigpy-znp), and so for bellows we have to the same.

And here I'm talking about Zigbee command send from a device to the coordinator on an Ep != 0x01.  Client -> Server

pipiche38, 2022-03-07:
> Is it necessary for you to create the `0x0B` endpoint? The new endpoint is causing the coordinator quirk to not match, which will prevent `EZSPEndpoint.add_to_group` from being created. Regardless, I've pushed a fix for this problem to the new radio API branch: [eb00967](https://github.com/zigpy/bellows/commit/eb0096745d9ce4a983c48a25d33394fe355b2eb2)

Doing the same on ZNP without issues ;-)

puddly, 2022-03-07:
I would double check if you actually have to create the endpoints. EZSP handles them differently and you may not need them to be explicitly registered unless the joining device is itself requesting the coordinator's simple descriptor.

Adminiuga, 2022-03-08:
Probably no need to create those endpoints on ZNP as well, as long as the latest firmware is used.

pipiche38, 2022-03-08:
I think the way it is , is good. Adding the Ep when needed, whiteout impacting the lib and all users.

On zigate we suffered from the fact that everyone a new firmware was beefed, and it would be a shame that it becomes the same with zigpy lib

Hedda, 2022-03-08:
> To change panid, ext pan id etc, you need to form a new network. Channel could be set during formation as well, o**r could be changed later, but it is more convoluted. You could check zha-toolkit by mdeweerd, in ezsp.py there's ezsp_set_channel() function which changes the channel**

For reference there was/is a separate general discussion about changing Zigbee channel with zigpy/ZHA here -> https://github.com/zigpy/zigpy/discussions/908

> Bellows will need to be updated to support the `app.get_device(nwk=0x0000).zdo.Mgmt_NWK_Update_req(...)` interface if you want to change the channel at runtime and notify other devices. It does not support loopback ZDO requests and instead requires these commands to be translated to EZSP commands.

As mentioned, [zha-toolkit](https://github.com/mdeweerd/zha-toolkit) by @mdeweerd (based on https://github.com/Adminiuga/zha_custom) can change the Zigbee channel but as also mentioned it does currently not notify any connected Zigbee devices about the Zigbee channel change so currently, some connected devices might automatically switch over automatically after 24-hours while other might need to be manually reset / power-cycled and yet others might need to be re-paried again.

SylvainPer, 2022-03-08:
> I would double check if you actually have to create the endpoints. EZSP handles them differently and you may not need them to be explicitly registered unless the joining device is itself requesting the coordinator's simple descriptor.

Just checked with EZSP, add_endpoint is not needed, everything is working fine with the wiser endpoint (0x0b), even the binding between devices.

SylvainPer, 2022-03-11:
> @Adminiuga Bellows config doesn't extend the zigpy config for CONF_NWK. If I try to add it, I have an error. I need to both import bellows.config and zigpy.config and mix the two to build the config, it's easier with znp as it imports it. Here the znp config import part: `from zigpy.config import ( # noqa: F401 CONF_NWK, CONF_DEVICE, CONF_NWK_KEY, CONFIG_SCHEMA, SCHEMA_DEVICE, CONF_NWK_PAN_ID, CONF_DEVICE_PATH, CONF_NWK_CHANNEL, CONF_NWK_CHANNELS, CONF_NWK_UPDATE_ID, CONF_NWK_TC_ADDRESS, CONF_NWK_TC_LINK_KEY, CONF_NWK_EXTENDED_PAN_ID, cv_boolean, )`

Created https://github.com/zigpy/bellows/pull/451 for this point and I'll close this issue, thanks for the support, the lib integration in the plugin in now operational.

closed 2022-03-11 by SylvainPer