zigpy/bellows #458: Not able to run bellows cli

pipiche38, 2022-05-08:
I'm trying to run bellows cli in order to backup the current configuration

I did 

`pip3 install bellows`

then when running what ever option of bellows I'm get stuck with the Timeout Excetion
However the Key runs well on my environment. What did I do wrong ?

```
[domoticz@rasp ~]$ bellows -d /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 info
Traceback (most recent call last):
  File "/usr/lib/python3.9/asyncio/tasks.py", line 490, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/bin/bellows", line 8, in <module>
    sys.exit(main())
  File "/usr/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/lib/python3.9/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/bellows/cli/util.py", line 39, in inner
    loop.run_until_complete(f(*args, **kwargs))
  File "/usr/lib/python3.9/asyncio/base_events.py", line 647, in run_until_complete
    return future.result()
  File "/usr/local/lib/python3.9/site-packages/bellows/cli/ncp.py", line 69, in info
    s = await util.setup(ctx.obj["device"], ctx.obj["baudrate"])
  File "/usr/local/lib/python3.9/site-packages/bellows/cli/util.py", line 118, in setup
    await s.reset()
  File "/usr/local/lib/python3.9/site-packages/bellows/ezsp/__init__.py", line 98, in reset
    await self._gw.reset()
  File "/usr/local/lib/python3.9/site-packages/bellows/uart.py", line 223, in reset
    return await asyncio.wait_for(self._reset_future, timeout=RESET_TIMEOUT)
  File "/usr/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

pipiche38, 2022-05-08:
I also try the command bellows devices ,; but get the following errors ?

[domoticz@rasp ~]$ bellows devices
Usage: bellows devices [OPTIONS]
Try 'bellows devices --help' for help.

Error: Invalid value for '-D' / '--database': File '/usr/share/domoticz/.config/bellows/app.db' does not exist.

Does that mean thet bellows must be use only with zigpy based application which  use the zigpy database ?

NilsBohr, 2022-05-08:
@pipiche38 

You need to specify the baudrate most probably. `bellows` uses 57600 baudrate by default if I understand correctly.
Try using baudrate 115200.

You can specify the database file yourself if you "form" the network using `bellows`. Look into "form" command line options.

pipiche38, 2022-05-09:
Thanks @NilsBohr it works!

pipiche38, 2022-05-27:
I have done the following command 
`bellows -b 115200 -d /dev/ttyAMA0 backup`

and got only that result. There are almost 60 devices paired on that CIE .

> {"node_type": 1, "node_id": 0, "node_ieee": "60:xxxxxxxxx:ac", "panId": 64162, "extendedPanId": "79:xxxxxxxxxbf", "radioChannel": 15, "radioTxPower": 8, "nwkUpdateId": 0, "channels": 134215680, "tc_link_key": {"bitmask": 26, "type": 1, "key": [10, ........, 129], "outgoingFrameCounter": 135168, "incomingFrameCounter": 0, "sequenceNumber": 0, "partnerEUI64": "60:a4:23:ff:fe:aa:33:ac"}, "network_key": {"bitmask": 3, "type": 3, "key": [38, ........, 13], "outgoingFrameCounter": 1929217, "incomingFrameCounter": 0, "sequenceNumber": 0, "partnerEUI64": "00:00:00:00:00:00:00:00"}, "key_table": []}

MattWestb, 2022-05-27:
Bellows is using hashed link key so then Zigbee 3 device is requesting one update of the link key its getting one that is hashed if the device IEEE and is not stored in the NVR of the coordinator.
If you is not using APS link keys they shall not being any more keys in the coordinator key storage.

zigbeefordomoticz, 2022-05-27:
I’m sorry but it is hard to understand what you wrote.

Do you mean that nothing is stored in the nvram ?

MattWestb, 2022-05-27:
OK the read the https://www.silabs.com/documents/public/user-guides/ug100-ezsp-reference-guide.pdf and:
```
EMBER_TRUST_CENTER_USES_HASHED_LINK_KEY 0x0084 This denotes that the preconfiguredKey is not the actual
Link Key but a Secret Key known only to the Trust Center.
It is hashed with the IEEE Address of the destination
device in order to create the actual Link Key used in
encryption. This is bit is only used by the Trust Center. The
joining device need not set this.
```
and https://community.silabs.com/s/article/hashed-link-keys?language=en_US

MattWestb, 2022-05-27:
And one comment in the Silabs community posting:
`Note that because there is no permanent storage of these hashed keys on the Trust Center`.

pipiche38, 2022-05-27:
So in short, the backup as it is provided by bellows client backup what is needed in case of restore and all existing devices will be able to work. ?

MattWestb, 2022-05-27:
Puddly and Alexei is knowing that but the IEEE is needed being burned to the new coordinator chip for getting binding and reporting working and i think (= not knowing) that the hashed link key is not working if not doing it.

puddly, 2022-05-30:
My understanding is that you can restore only to the exact same EmberZNet adapter unless you pass the `--i-understand-i-can-update-eui64-only-once-and-i-still-want-to-do-it` flag to update the adapter EUI64 address (as the flag suggests, it can only be done once without using a SWD debugger).

closed 2022-07-24 by puddly