zigpy/bellows #461: Resetting ControllerApplication. Cause: 'NcpResetCode.RESET_WATCHDOG'

pipiche38, 2022-05-29:
while everything works well, we have the following error. This is occuri,ng several time in a day.
There are about 65 Devices. (28 routers).  ( https://github.com/zigbeefordomoticz/Domoticz-Zigbee/issues/1200 )

```
2022-05-29 18:27:42,263 DEBUG   :Sending: b'70dc21a9602a15fc2d904b23aa5493099d4e270ed9cb772ef6cc6383cb7e'
2022-05-29 18:27:44,231 DEBUG   :Error code: RESET_WATCHDOG, Version: 2, frame: b'c20203d20a7e'
2022-05-29 18:27:44,234 ERROR   :NCP entered failed state. Requesting APP controller restart
2022-05-29 18:27:44,235 DEBUG   :Received _reset_controller_application frame with (<NcpResetCode.RESET_WATCHDOG: 3>,)
2022-05-29 18:27:44,237 DEBUG   :Closed serial connection
```

And then nothing happen !!!!

```
2022-05-29 18:27:32,194 DEBUG   :Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=8, sourceEndpoint=3, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=74), 60, -85, 0xc501, 255, 255, b'\x18\x1d\n\x00\x00 \xfe']
2022-05-29 18:27:32,571 DEBUG   :Data frame: b'14d8b1a9112a15b658964d24ab1592499c079f61a6e59874efde6383fc7e1af3d1c8de6f8fffc3f179d669de957e'
2022-05-29 18:27:32,572 DEBUG   :Sending: b'82503a7e'
2022-05-29 18:27:32,573 DEBUG   :Application frame 69 (incomingMessageHandler) received: b'000401020701014000000049b8ca4b2bffff1218000a000025543a0500000000042aac0400'
2022-05-29 18:27:32,577 DEBUG   :Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1794, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=73), 184, -54, 0x2b4b, 255, 255, b'\x18\x00\n\x00\x00%T:\x05\x00\x00\x00\x00\x04*\xac\x04\x00']
2022-05-29 18:27:33,624 DEBUG   :Data frame: b'24d8b1a9112a15b658954824ab1593499c619362c8979874f5de2283fc7e1619e1c9d64e7e'
2022-05-29 18:27:33,627 DEBUG   :Sending: b'83401b7e'
2022-05-29 18:27:33,630 DEBUG   :Application frame 69 (incomingMessageHandler) received: b'00040101020101400100002fb4c92559ffff0818410a000029be0a04'
2022-05-29 18:27:33,635 DEBUG   :Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=513, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=47), 180, -55, 0x5925, 255, 255, b'\x18A\n\x00\x00)\xbe\n']
2022-05-29 18:27:34,188 DEBUG   :Data frame: b'34d8b1a9112a15b658964d24ab1592499c079f61a6e59874efde6383fc7e1af3d1c8de6f8fffc3f179d6695e787e'
2022-05-29 18:27:34,188 DEBUG   :Sending: b'8430fc7e'
2022-05-29 18:27:34,190 DEBUG   :Application frame 69 (incomingMessageHandler) received: b'000401020701014000000049b8ca4b2bffff1218000a000025543a0500000000042aac0400'
2022-05-29 18:27:34,194 DEBUG   :Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1794, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=73), 184, -54, 0x2b4b, 255, 255, b'\x18\x00\n\x00\x00%T:\x05\x00\x00\x00\x00\x04*\xac\x04\x00']
2022-05-29 18:27:34,855 DEBUG   :Data frame: b'44d8b1a90d2a9c5e134f10db55726fcd2c8627d2107e'
2022-05-29 18:27:34,855 DEBUG   :Sending: b'8520dd7e'
2022-05-29 18:27:34,857 DEBUG   :Application frame 89 (incomingRouteRecordHandler) received: b'89ec4adb5afeff27fd84b0c800'
2022-05-29 18:27:34,858 DEBUG   :Received incomingRouteRecordHandler frame with [0xec89, 84:fd:27:ff:fe:5a:db:4a, 176, -56, []]
2022-05-29 18:27:34,858 DEBUG   :Processing route record request: (0xec89, 84:fd:27:ff:fe:5a:db:4a, 176, -56, [])
2022-05-29 18:27:34,910 DEBUG   :Data frame: b'54d8b1a9112a15b658964d24ab1592499c3d976364229874efde6383fc7e1a71facdde6f8fffc3f1d5d2698e5e8c7e'
2022-05-29 18:27:34,910 DEBUG   :Sending: b'8610be7e'
2022-05-29 18:27:34,912 DEBUG   :Application frame 69 (incomingMessageHandler) received: b'000401020701014000000073b0c889ecffff1218000a000025d6110000000000042a00000002'
2022-05-29 18:27:34,914 DEBUG   :Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1794, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=115), 176, -56, 0xec89, 255, 255, b'\x18\x00\n\x00\x00%\xd6\x11\x00\x00\x00\x00\x00\x04*\x00\x00\x00']
2022-05-29 18:27:35,940 DEBUG   :Data frame: b'64d8b1a9112a15b65894b5caab5592499ccf8b6c82289874fb396289f87c3d41f87e'
2022-05-29 18:27:35,941 DEBUG   :Sending: b'87009f7e'
2022-05-29 18:27:35,942 DEBUG   :Application frame 69 (incomingMessageHandler) received: b'00040100ffef010000000081acc76fe6ffff06ff0100040202'
2022-05-29 18:27:35,944 DEBUG   :Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=65280, sourceEndpoint=239, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_NONE: 0>, groupId=0, sequence=129), 172, -57, 0xe66f, 255, 255, b'\xff\x01\x00\x04\x02\x02']
2022-05-29 18:27:36,163 DEBUG   :Data frame: b'74d8b1a90d2af09b174e10db55726fcd208526d85d25247e'
2022-05-29 18:27:36,163 DEBUG   :Sending: b'8070787e'
2022-05-29 18:27:36,165 DEBUG   :Application frame 89 (incomingRouteRecordHandler) received: b'e5294eda5afeff27fd84bccb0173b0'
2022-05-29 18:27:36,166 DEBUG   :Received incomingRouteRecordHandler frame with [0x29e5, 84:fd:27:ff:fe:5a:da:4e, 188, -53, [0xb073]]
2022-05-29 18:27:36,166 DEBUG   :Processing route record request: (0x29e5, 84:fd:27:ff:fe:5a:da:4e, 188, -53, [0xb073])
2022-05-29 18:27:36,173 DEBUG   :Data frame: b'04d8b1a9112a15b658964d24ab1592499ce09b6008e79874efde6383fc7e1af2e3cdde6f8fffc3f1d5d2698820707e'
2022-05-29 18:27:36,173 DEBUG   :Sending: b'8160597e'
2022-05-29 18:27:36,175 DEBUG   :Application frame 69 (incomingMessageHandler) received: b'0004010207010140000000aebccbe529ffff1218000a00002555080000000000042a00000004'
2022-05-29 18:27:36,178 DEBUG   :Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=260, clusterId=1794, sourceEndpoint=1, destinationEndpoint=1, options=<EmberApsOption.APS_OPTION_RETRY: 64>, groupId=0, sequence=174), 188, -53, 0x29e5, 255, 255, b'\x18\x00\n\x00\x00%U\x08\x00\x00\x00\x00\x00\x04*\x00\x00\x00']
2022-05-29 18:27:36,206 DEBUG   :Send command readCounters: ()
2022-05-29 18:27:36,208 DEBUG   :Sending: b'41d921a9a52abb6a7e'
2022-05-29 18:27:36,230 DEBUG   :Data frame: b'15d9a1a9a52aa0b6359479277455d649854e2eabe4ce7e8adec66c89fa7e30a7c9cdcc6f87ffc7dbd5d2698c4623a9ec763ba5ea75824198682713b1e070381c0e07bbe5ca6598459a4d9e4f9ff7c3d9d46a35a2519048246c217e'
2022-05-29 18:27:36,231 DEBUG   :Sending: b'82503a7e'
2022-05-29 18:27:36,233 DEBUG   :Application frame 241 (readCounters) received: b'b5046c003302de004400190009000900190123000f0006000f002200120008000000000000000000000000000000000000002401000000000000000000000000120000000000000000000000000000000000'
2022-05-29 18:27:36,237 DEBUG   :Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2022-05-29 18:27:36,238 DEBUG   :Sending: b'52de21a9fe2a1687c07e'
2022-05-29 18:27:36,245 DEBUG   :Data frame: b'26dea1a9fe2a15b3abd1017e'
2022-05-29 18:27:36,245 DEBUG   :Sending: b'83401b7e'
2022-05-29 18:27:36,256 DEBUG   :Application frame 170 (getValue) received: b'0001f2'
2022-05-29 18:27:36,258 DEBUG   :Free buffers status EzspStatus.SUCCESS, value: 242
2022-05-29 18:27:36,258 DEBUG   :ezsp_counters: [MAC_RX_BROADCAST = 1205, MAC_TX_BROADCAST = 108, MAC_RX_UNICAST = 563, MAC_TX_UNICAST_SUCCESS = 222, MAC_TX_UNICAST_RETRY = 68, MAC_TX_UNICAST_FAILED = 25, APS_DATA_RX_BROADCAST = 9, APS_DATA_TX_BROADCAST = 9, APS_DATA_RX_UNICAST = 281, APS_DATA_TX_UNICAST_SUCCESS = 35, APS_DATA_TX_UNICAST_RETRY = 15, APS_DATA_TX_UNICAST_FAILED = 6, ROUTE_DISCOVERY_INITIATED = 15, NEIGHBOR_ADDED = 34, NEIGHBOR_REMOVED = 18, NEIGHBOR_STALE = 8, JOIN_INDICATION = 0, CHILD_REMOVED = 0, ASH_OVERFLOW_ERROR = 0, ASH_FRAMING_ERROR = 0, ASH_OVERRUN_ERROR = 0, NWK_FRAME_COUNTER_FAILURE = 0, APS_FRAME_COUNTER_FAILURE = 0, UTILITY = 0, APS_LINK_KEY_NOT_AUTHORIZED = 0, NWK_DECRYPTION_FAILURE = 292, APS_DECRYPTION_FAILURE = 0, ALLOCATE_PACKET_BUFFER_FAILURE = 0, RELAYED_UNICAST = 0, PHY_TO_MAC_QUEUE_LIMIT_REACHED = 0, PACKET_VALIDATE_LIBRARY_DROPPED_COUNT = 0, TYPE_NWK_RETRY_OVERFLOW = 0, PHY_CCA_FAIL_COUNT = 18, BROADCAST_TABLE [... cut ...]
    await process_raw_command(self, data["datas"], AckIsDisable=data["ACKIsDisable"], Sqn=data["Sqn"])
  File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 407, in process_raw_command
    result, msg = await self.app.mrequest(destination, Profile, Cluster, sEp, sequence, payload)
  File "/home/pi/domoticz/plugins/Domoticz-Zigbee/bellows/zigbee/application.py", line 479, in mrequest
    raise ControllerError("ApplicationController is not running")
bellows.exception.ControllerError: ApplicationController is not running
```

pipiche38, 2022-05-30:
Is there anything expected to be done when receiving Exception  ?

bellows.exception.ControllerError: ApplicationController is not running

puddly, 2022-05-30:
There is a gap in the debug log of over an hour:

```
2022-05-29 18:27:44,237 DEBUG   :Resetting ControllerApplication. Cause: 'NcpResetCode.RESET_WATCHDOG'
...
2022-05-29 20:09:07,512 INFO    : [       MainThread] UpdateGroup  - (  Lampes sejour) 1:100
```

Please post the full log, bellows should be trying to reconnect.

pipiche38, 2022-05-30:
You have the full log.

After this message
`2022-05-29 18:27:47,267 ERROR   :Task exception was never retrieved`

pipiche38, 2022-05-30:
There is no more Log. the bellows just stop working and logging !
A restart of the plugin was tried and this doesn't help
A full power Off/On make it working again

As mentioned more than 60 devices. A lot of incoming traffic ( several metering equipments )

puddly, 2022-05-30:
What is in the debug log when you restart the plugin and it doesn't work?

pipiche38, 2022-05-30:
Here is what I have

[pp.zip](https://github.com/zigpy/bellows/files/8799401/pp.zip)

here is the log level set

# Debug Bellows/Ezsp
```
requests_logger = logging.getLogger("zigpy")
requests_logger.setLevel(logging.DEBUG)
requests_logger = logging.getLogger("bellows")
requests_logger.setLevel(logging.DEBUG)
requests_logger = logging.getLogger("AppEzsp")
requests_logger.setLevel(logging.DEBUG)
```

puddly, 2022-05-30:
I'm not sure. When EZSP's `_handle_reset_request` is received, it will log:

```Python
Resetting ControllerApplication. Cause: 'NcpResetCode.RESET_WATCHDOG'

pipiche38, 2022-07-11:
up+

pipiche38, 2022-07-11:
2022-07-09 15:34:00,013 DEBUG   :get_device raise KeyError ieee: None nwk: fcbd !!
2022-07-09 15:34:40,577 ERROR   :NCP entered failed state. Requesting APP controller restart
2022-07-09 15:34:40,579 ERROR   :NCP entered failed state. Requesting APP controller restart
2022-07-09 15:34:40,580 ERROR   :NCP entered failed state. Requesting APP controller restart
2022-07-09 15:34:40,580 ERROR   :NCP entered failed state. Requesting APP controller restart
2022-07-09 15:34:45,500 ERROR   :Task exception was never retrieved
future: <Task finished coro=<transport_request() done, defined at /home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py:565> exception=TimeoutError()>
Traceback (most recent call last):
  File "/home/pi/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 585, in transport_request
    result, msg = await self.app.request( destination, Profile, Cluster, sEp, dEp, sequence, payload, expect_reply, use_ieee )
  File "/home/pi/domoticz/plugins/Domoticz-Zigbee/bellows/zigbee/application.py", line 777, in request
    await self._ezsp.setExtendedTimeout(device.ieee, True)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 423, in wait_for
    raise futures.TimeoutError()
concurrent.futures._base.TimeoutError
2022-07-09 15:34:51,248 ERROR   :Task exception was never retrieved

closed 2022-11-05 by pipiche38