zigpy/bellows #465: WARNING :NWK conflict is reported for 0x6452

pipiche38, 2022-07-13:
This seems to be bellows specifics.
As we are not using the full ZHA stack and only the low zigpy layers, is there anything special we need to handle when getting such message ?

Is this message just a notification for informtion, and the coordinator is handling it , or is anything to be done by the application ?

pipiche38, 2022-08-08:
just to keep it open (before a bot close it)

dmulcahey, 2022-08-08:
It’s just a notification. You can read about this in the SI docs. the device(s) are sent leave and rejoin requests and things should continue to work normally for spec compliant devices. I think the info is in UG103.2 IIRC.

pipiche38, 2022-08-09:
I'm not used to the Sl stack, and I wonder if the coordinator can work without bellows up , because I feel the bellows library is doing a lot of thing. This is why I was ask I understand what the NWK conflict is , but I'm surprise that it is brought up to the bellows library level (and not handle at the firmware level)

MattWestb, 2022-08-09:
Silabs Zigbee stack is doing all things in the SOC and only need commands from the host system for joining device and the rest is in the SOC.

**NWK conflict** is working OK with bellows and also deCONZ is having the same functionality (reporting) but i dont knowing if IT is reporting it or only ignoring it.

The host  system need the information if one device is rejoining and changing its NWK for not getting problem but the host system cant do anything then all low level things is being made in the SOC and shall being made there and not in the host system.