zigpy/bellows #468: Not able to start zigpy/bellows since the latest update of master

pipiche38, 2022-07-18:
I have apply the latest zigpy and bellows version (from master) and not able to start. if I fall back to the previous version it works liek a charm

```
2022-07-18 14:16:51,717 INFO    :[0x0000] Already have endpoints: {0: <zigpy.zdo.ZDO object at 0xa97bae68>, 1: <EZSPEndpoint id=1 in=[] out=[] status=<Status.NEW: 0>>}
2022-07-18 14:16:51,723 INFO    :[0x0000] Initializing endpoints [<EZSPEndpoint id=1 in=[] out=[] status=<Status.NEW: 0>>]
2022-07-18 14:16:51,734 INFO    :[0x0000:1] Discovering endpoint information
2022-07-18 14:16:51,736 DEBUG   :Tries remaining: 3
2022-07-18 14:16:51,745 DEBUG   :Send command sendUnicast: (<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0x0000, EmberApsFrame(profileId=0, clusterId=4, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=3), 4, b'\x03\x00\x00\x01')
2022-07-18 14:16:51,761 DEBUG   :Sending: b'267b21a9602a15b259944a21aa5592099d4e27a8e9ca648bfdc7d3d57e'
2022-07-18 14:16:51,774 DEBUG   :Data frame: b'637ba1a9602a1549a0937e'
2022-07-18 14:16:51,776 DEBUG   :Sending: b'87009f7e'
2022-07-18 14:16:51,780 DEBUG   :Application frame 52 (sendUnicast) received: b'00fb'
2022-07-18 14:16:51,781 DEBUG   :Data frame: b'737bb1a9112a15b259904a25aa1593499cb5d8abedce9874f9c56389fd67907e'
2022-07-18 14:16:51,786 DEBUG   :Sending: b'8070787e'
2022-07-18 14:16:51,789 DEBUG   :Application frame 69 (incomingMessageHandler) received: b'0000000400000040010000fbff000000ffff0403000001'
2022-07-18 14:16:51,791 DEBUG   :Data frame: b'037bb1a9112a15b25990ca25aa1593499cb2d8abedce9874e2c56389fc643ea3eacdda6f8affc7ddd5d869954622ace8773b85ea7587439d34797e'
2022-07-18 14:16:51,801 DEBUG   :Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=4, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=251), 255, 0, 0x0000, 255, 255, b'\x03\x00\x00\x01']
2022-07-18 14:16:51,802 DEBUG   :Sending: b'8160597e'
2022-07-18 14:16:51,808 DEBUG   :Received frame on uninitialized device <EZSPCoordinator model='EZSP' manuf='Silicon Labs' nwk=0x0000 ieee=cc:cc:cc:ff:fe:a5:f3:07 is_initialized=False> from ep 0 to ep 0, cluster 4: b'\x03\x00\x00\x01'
2022-07-18 14:16:51,815 DEBUG   :Data frame: b'137bb1a96b2a15b259944a21aa5592099d4e2750e9ce67e0747e'
2022-07-18 14:16:51,819 DEBUG   :Sending: b'82503a7e'
2022-07-18 14:16:51,816 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.Simple_Desc_req: [0x0000, 1]
2022-07-18 14:16:51,824 DEBUG   :[0x0000:zdo] No handler for ZDO request:ZDOCmd.Simple_Desc_req([0x0000, 1])
2022-07-18 14:16:51,834 DEBUG   :Application frame 69 (incomingMessageHandler) received: b'0000000480000040010000fcff000000ffff1f030000001a01040100040005000006000a0019000105040100200000050205'
2022-07-18 14:16:51,842 DEBUG   :Received incomingMessageHandler frame with [<EmberIncomingMessageType.INCOMING_UNICAST: 0>, EmberApsFrame(profileId=0, clusterId=32772, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=252), 255, 0, 0x0000, 255, 255, b'\x03\x00\x00\x00\x1a\x01\x04\x01\x00\x04\x00\x05\x00\x00\x06\x00\n\x00\x19\x00\x01\x05\x04\x01\x00 \x00\x00\x05\x02\x05']
2022-07-18 14:16:51,852 DEBUG   :Received frame on uninitialized device <EZSPCoordinator model='EZSP' manuf='Silicon Labs' nwk=0x0000 ieee=cc:cc:cc:ff:fe:a5:f3:07 is_initialized=False> from ep 0 to ep 0, cluster 32772: b'\x03\x00\x00\x00\x1a\x01\x04\x01\x00\x04\x00\x05\x00\x00\x06\x00\n\x00\x19\x00\x01\x05\x04\x01\x00 \x00\x00\x05\x02\x05'
2022-07-18 14:16:51,865 DEBUG   :Application frame 63 (messageSentHandler) received: b'00000000000400000040010000fb040000'
2022-07-18 14:16:51,876 DEBUG   :Received messageSentHandler frame with [<EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 0, EmberApsFrame(profileId=0, clusterId=4, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_ENABLE_ROUTE_DISCOVERY|APS_OPTION_RETRY: 320>, groupId=0, sequence=251), 4, <EmberStatus.SUCCESS: 0>, b'']
2022-07-18 14:16:51,886 INFO    :[0x0000:1] Discovered endpoint information: SizePrefixedSimpleDescriptor(endpoint=1, profile=260, device_type=1024, device_version=0, input_clusters=[0, 6, 10, 25, 1281], output_clusters=[1, 32, 1280, 1282])
2022-07-18 14:16:51,893 INFO    :[0x0000] Already have model and manufacturer info
2022-07-18 14:16:51,899 INFO    :[0x0000] Discovered basic device information for <EZSPCoordinator model='EZSP' manuf='Silicon Labs' nwk=0x0000 ieee=cc:cc:cc:ff:fe:a5:f3:07 is_initialized=True>
2022-07-18 14:16:51,901 DEBUG   :Device is initialized <EZSPCoordinator model='EZSP' manuf='Silicon Labs' nwk=0x0000 ieee=cc:cc:cc:ff:fe:a5:f3:07 is_initialized=True>
2022-07-18 14:16:51,904 DEBUG   :Checking quirks for Silicon Labs EZSP (cc:cc:cc:ff:fe:a5:f3:07)
2022-07-18 14:16:51,906 ERROR   :Couldn't start application
2022-07-18 14:16:51,907 DEBUG   :Cancelling watchdog
2022-07-18 14:16:51,909 DEBUG   :Cancelling watchdog
2022-07-18 14:16:51,911 DEBUG   :Closed serial connection
```

pipiche38, 2022-07-18:
I did some debuging, and it looks like it crashes on the new added code

```
        if db_device is not None:
                ezsp_device.endpoints[1].member_of.update(db_device.endpoints[1].member_of)
```

puddly, 2022-07-18:
What is the traceback?

pipiche38, 2022-07-19:
```
2022-07-19 09:59:50,123 INFO    : [ ZigpyForwarder_7] ZigpyTransport: thread_processing_and_sending Thread start.
2022-07-19 09:59:50,126 INFO    : [       MainThread] Start Web Server connection
2022-07-19 09:59:50,129 INFO    : [       MainThread] Web backend for Web User Interface started on port: 9440
2022-07-19 09:59:50,132 INFO    : [       MainThread] Domoticz Widgets usage is at 13.7 %
2022-07-19 09:59:51,727 INFO    : [       ZigpyCom_7] Started radio ezsp port: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
2022-07-19 09:59:51,736 DEBUG   :Using selector: EpollSelector
2022-07-19 09:59:53,486 DEBUG   :Network info: NetworkInfo(extended_pan_id=3a:da:a7:1b:a8:eb:93:b7, pan_id=0xA943, nwk_update_id=0, nwk_manager_id=0x0000, channel=15, channel_mask=<Channels.ALL_CHANNELS: 134215680>, security_level=5, network_key=Key(key=7d:91:be:91:11:64:1c:74:b0:a3:5c:bf:8b:8e:5e:d3, tx_counter=2830337, rx_counter=0, seq=0, partner_ieee=ff:ff:ff:ff:ff:ff:ff:ff), tc_link_key=Key(key=5a:69:67:42:65:65:41:6c:6c:69:61:6e:63:65:30:39, tx_counter=65536, rx_counter=0, seq=0, partner_ieee=cc:cc:cc:ff:fe:a5:f3:07), key_table=[], children=[], nwk_addresses={}, stack_specific={'ezsp': {'hashed_tclk': 'a23bc18d7292917f2db9135ba9f56414'}}, metadata={'ezsp': {'manufacturer': 'Elelabs', 'board': 'ELR023', 'version': '6.10.3.0 build 297', 'stack_version': 8}}, source='bellows@0.31.1')
2022-07-19 09:59:53,489 DEBUG   :Node info: NodeInfo(nwk=0x0000, ieee=cc:cc:cc:ff:fe:a5:f3:07, logical_type=<LogicalType.Coordinator: 0>)
2022-07-19 09:59:53,954 ERROR   :Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppBellows.py", line 110, in _startup
    await self.start_network()
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/bellows/zigbee/application.py", line 208, in start_network
    ezsp_device.endpoints[1].member_of.update(db_device.endpoints[1].member_of)
KeyError: 1
```

pipiche38, 2022-07-19:
In order to avoid a potential issue in our _startup() which is just a copy of the zigpy one withoout the permit(0) here is without
```

2022-07-19 10:18:15,574 INFO    : [       ZigpyCom_7] Started radio ezsp port: /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
2022-07-19 10:18:15,582 DEBUG   :Using selector: EpollSelector
2022-07-19 10:18:17,833 ERROR   :Couldn't start application
2022-07-19 10:18:17,856 ERROR   :Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppBellows.py", line 65, in startup
    await super().startup( auto_form=True )
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/zigpy/application.py", line 80, in startup
    await self.start_network()
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/bellows/zigbee/application.py", line 208, in start_network
    ezsp_device.endpoints[1].member_of.update(db_device.endpoints[1].member_of)
KeyError: 1
```

pipiche38, 2022-07-20:
@puddly is that the stack trace you were looking for ? Should I do more tests ?

puddly, 2022-07-20:
This likely is caused by you not using the zigpy database, since the error says that your coordinator entry in the database has no endpoints.

pipiche38, 2022-07-21:
I don't understand your comment. The only thing that we do is that the zigpy database is not persistent (we do not define a filename to store it), but we didn't overwrite anything here.

If I understood correctly the logic

we do call super().startup()
    -> self.connect()
         -> register_endpoint()
                    -> super().register_endpoints()

So EndPoint should be registered, no ? In the log here after which give you the all debug info of startup, there is the endpoint_registration happening.

```
2022-07-21 15:22:56,935 DEBUG   :SQLite version for <module 'sqlite3' from '/usr/lib/python3.9/sqlite3/__init__.py'>: 3.34.1
2022-07-21 15:22:58,442 DEBUG   :Using selector: EpollSelector
2022-07-21 15:22:59,807 DEBUG   :Using selector: EpollSelector
2022-07-21 15:22:59,828 DEBUG   :Resetting EZSP
2022-07-21 15:22:59,834 DEBUG   :Resetting ASH
2022-07-21 15:22:59,836 DEBUG   :Sending: b'1ac038bc7e'
2022-07-21 15:23:00,915 DEBUG   :RSTACK Version: 2 Reason: RESET_SOFTWARE frame: b'c1020b0a527e'
2022-07-21 15:23:00,920 DEBUG   :Send command version: (4,)
2022-07-21 15:23:00,924 DEBUG   :Sending: b'004221a850ed2c7e'
2022-07-21 15:23:00,934 DEBUG   :Data frame: b'0142a1a85c2825d8e12b7e'
2022-07-21 15:23:00,936 DEBUG   :Sending: b'8160597e'
2022-07-21 15:23:00,939 DEBUG   :Application frame 0 (version) received: b'0802306a'
2022-07-21 15:23:00,942 DEBUG   :Switching to EZSP protocol version 8
2022-07-21 15:23:00,945 DEBUG   :Send command version: (8,)
2022-07-21 15:23:00,949 DEBUG   :Sending: b'7d314221a9542a1d8cdf7e'
2022-07-21 15:23:00,960 DEBUG   :Data frame: b'1242a1a9542a1db069fe46f07e'
2022-07-21 15:23:00,961 DEBUG   :Sending: b'82503a7e'
2022-07-21 15:23:00,965 DEBUG   :Application frame 0 (version) received: b'0802306a'
2022-07-21 15:23:00,967 DEBUG   :EZSP Stack Type: 2, Stack Version: 6a30, Protocol version: 8
2022-07-21 15:23:00,968 DEBUG   :Send command getValue: (<EzspValueId.VALUE_FREE_BUFFERS: 3>,)
2022-07-21 15:23:00,971 DEBUG   :Sending: b'224321a9fe2a1683467e'
2022-07-21 15:23:00,980 DEBUG   :Data frame: b'2343a1a9fe2a15b3a191d67e'
2022-07-21 15:23:00,982 DEBUG   :Sending: b'83401b7e'
2022-07-21 15:23:00,984 DEBUG   :Application frame 170 (getValue) received: b'0001f8'
2022-07-21 15:23:00,988 DEBUG   :Send command getConfigurationValue: (<EzspConfigId.CONFIG_PACKET_BUFFER_COUNT: 1>,)
2022-07-21 15:23:00,992 DEBUG   :Sending: b'334021a9062a14a80d7e'
2022-07-21 15:23:01,001 DEBUG   :Data frame: b'3440a1a9062a15485927a17e'
2022-07-21 15:23:01,003 DEBUG   :Sending: b'8430fc7e'
2022-07-21 15:23:01,006 DEBUG   :Application frame 82 (getConfigurationValue) received: b'00fa00'
2022-07-21 15:23:01,009 DEBUG   :Free/configured buffers before any configurations: 248/250
2022-07-21 15:23:01,011 DEBUG   :Send command setConfigurationValue: (<EzspConfigId.CONFIG_STACK_PROFILE: 12>, 2)
2022-07-21 15:23:01,015 DEBUG   :Sending: b'444121a9072a19b05925087e'
2022-07-21 15:23:01,024 DEBUG   :Data frame: b'4541a1a9072a15cd287e'
2022-07-21 15:23:01,025 DEBUG   :Sending: b'8520dd7e'
2022-07-21 15:23:01,028 DEBUG   :Application frame 83 (setConfigurationValue) received: b'00'
2022-07-21 15:23:01,031 DEBUG   :Send command setConfigurationValue: (<EzspConfigId.CONFIG_TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S: 56>, 90)
2022-07-21 15:23:01,035 DEBUG   :Sending: b'554621a9072a2de859c4be7e'
2022-07-21 15:23:01,044 DEBUG   :Data frame: b'5646a1a9072a15ea907e'
2022-07-21 15:23:01,046 DEBUG   :Sending: b'8610be7e'
2022-07-21 15:23:01,049 DEBUG   :Application frame 83 (setConfigurationValue) received: b'00'
2022-07-21 15:23:01,052 DEBUG   :Send command setConfigurationValue: (<EzspConfigId.CONFIG_SUPPORTED_NETWORKS: 45>, 1)
2022-07-21 15:23:01,056 DEBUG   :Sending: b'664721a9072a38b35920fd7e'
2022-07-21 15:23:01,068 DEBUG   :Data frame: b'6747a1a9072a154edc7e'
2022-07-21 15:23:01,069 DEBUG   :Sending: b'87009f7e'
2022-07-21 15:23:01,073 DEBUG   :Application frame 83 (setConfigurationValue) received: b'00'
2022-07-21 15:23:01,077 DEBUG   :Send command setConfigurationValue: (<EzspConfigId.CONFIG_ADDRESS_TABLE_SIZE: 5>, 16)
2022-07-21 15:23:01,080 DEBUG   :Sending: b'774421a9072a10a2594f707e'
2022-07-21 15:23:01,094 DEBUG   :Data frame: b'7044a1a9072a15ae037e'
2022-07-21 15:23:01,095 DEBUG   :Sending: b'8070787e'
2022-07-21 15:23:01,098 DEBUG   :Application frame 83 (setConfigurationValue) received: b'00'
2022-07-21 15:23:01,101 DEBUG   :Send command setConfigurationValue: (<EzspConfigId.CONFIG_SECURITY_LEVEL: 13>, 5)
2022-07-21 15:23:01,106 DEBUG   :Sending: b'004521a9072a7d38b75947f47e'
2022-07-21 15:23:01,117 DEBUG   :Data frame: b'0145a1a9072a15d7a37e'
2022-07-21 15:23:01,119 DEBUG   :Sending: b'8160597e'
2022-07-21 15:23:01,122 DEBUG   :Application frame 83 (setConfigurationValue) received: b'00'
2022-07-21 15:23:01,126 DEBUG   :Send command setConfigurationValue: (<EzspConfigId.CONFIG_PAN_ID_CONFLICT_REPORT_THRESHOLD: 34>, 2)
2022-07-21 15:23:01,130 DEBUG   :Sending: b'7d314a21a9072a37b059158b7e'
2022-07-21 15:23:01,139 DEBUG   :Data frame: b'124aa1a9072a15fd597e'
[... 300 lines cut ...]
2022-07-21 15:23:02,571 DEBUG   :Received frame on uninitialized device <EZSPCoordinator model='EZSP' manuf='Silicon Labs' nwk=0x0000 ieee=cc:cc:cc:ff:fe:a5:f3:07 is_initialized=False> from ep 0 to ep 0, cluster 4: b'\x03\x00\x00\x01'
2022-07-21 15:23:02,573 DEBUG   :Sending: b'8160597e'
2022-07-21 15:23:02,576 DEBUG   :[0x0000:zdo] ZDO request ZDOCmd.Simple_Desc_req: [0x0000, 1]
[... cut ...]
2022-07-21 15:23:02,656 DEBUG   :getting MFG_STRING token
2022-07-21 15:23:02,658 ERROR   : [       ZigpyCom_7] Issue in request , dumping stack: Traceback (most recent call last):
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/AppBellows.py", line 110, in _startup
    await self.start_network()
  File "/var/lib/domoticz/plugins/Domoticz-Zigbee/bellows/zigbee/application.py", line 208, in start_network
    ezsp_device.endpoints[1].member_of.update(db_device.endpoints[1].member_of)
KeyError: 1
```

PR #470 (2022-07-24): Check that endpoint 1 exists instead of assuming it does
Fixes #468
closed 2022-07-24 by puddly