zigpy/bellows #491: [HARDWARE] Would the EFR32xG21 Wireless Gecko Starter Kit work with this?

GilDev, 2022-08-30:
Hi,

I’m currently developing Zigbee products at work and I was wondering if one of the development boards included in the the [EFR32xG21 Wireless Gecko Starter Kit (SLWSTK6006A)](https://www.silabs.com/development-tools/wireless/efr32xg21-wireless-starter-kit) would work with bellows (using something like the NCP example firmwares, for instance “ncp-uart-hw” using USB connection)?

Just interested if that would be possible, maybe there are EZSP constraints behind all that, maybe the versions would be mismatched?

Thanks!

MattWestb, 2022-08-30:
With that WSTK you can doing all versions that you like of NCP for the board (very old is not supported but its not interesting using EZSP 6.5 on one MG21 device) and also multi rail NCP with GP supports.
Also RCP firmware with EZSP 7.x and Open Thread Boarder Router shall being possible if you like playing with Thread / Mather.

I have one MG22 WSTK and cooking NCPs for My "Billy" (IKEA old MG1P Zigbee module) and using then with external USB-TLL on Armbian boxes and PIs but MG22 is not so well supported as the MG21 chips.

Adminiuga, 2022-08-30:
You could also get the https://www.silabs.com/development-tools/wireless/efr32xg22-wireless-starter-kit (gives access to older emberznet) kit and if you really want an MG21 card, you could purchase that separately. $300 in savings

GilDev, 2022-08-30:
Thanks for the informations!

@Adminiuga Indeed that seems like a nice idea, it’s just that I have access to those boards through work, hence me wondering this (the MG21 kit has 3 boards, the MG22 kit has 1).

closed 2022-08-30 by GilDev

MattWestb, 2022-08-30:
The linked WSDK (is the one i having) is having 2 different MG22 board one with 32 pins and one with 40 pins / pads chip. 
The chip is only having +6 dBm but its OK for testing and most of the situations then the receiver sensitivity is the same as the standard Silabs radios.
Most MG21 board is having +20 dBm and more flash and ram if you need more "space" for you application code (512 kB flash and 32 kB RAM is enough for normal NCP firmware and is fully supported by Silabs for many year.

Adminiuga, 2022-08-30:
It should have two MG22 boards. MG21 has more resources, the Home Assistant Yellow is based on MG21 chip module

GilDev, 2022-08-30:
Ah sorry, there are indeed two boards with different pins count. We develop home automation products aimed at consumers with both MG21 and MG22 but only have the MG21 WSTK.