zigpy/bellows #493: [QUESTION] - routing question

pipiche38, 2022-09-07:
If this is not the appropriate place , please delete this issue and let me know where I should place it.

We are currently using the bellows and zigpy libraries in a plugin for Domoticz. In that context we are only using the low level zigpy libraries and for instance not using zcl/zdo implementation.

We have currently a system based on a Elelabs shield with about 70 devices and around 30 Routers.
Some times , 2 or 3 routers do not receive commands from the plugin/coordinator. failure when sending it, however the reports from the routers are coming and we can conclude that the device is still on the network and is healthy.

Every time we restart the plugin and so restart , we see immediately the possibility to reach the router via commands ( like On/Off).

We suspect a routing issue, which can be created by an other router on the network. However few questions : 
- why a simple software restart seems to free-up the issue (until a certain point of time ) ?
- Is bellows library handling the routing mechanism ? and so the software restart clean it up ?
- Is there any special attention we should have to particular ezsp parameters ?

puddly, 2022-09-07:
Please include a packet capture and debug log.

Bellows asks EmberZNet to send a packet and only ever sets routes if source routing is explicitly enabled:

https://github.com/zigpy/bellows/blob/3c3ee0296d35eb43d0493eb9b2160bc4484e892c/bellows/config/__init__.py#L41

If it is not enabled (the default), bellows performs no routing of any kind.

pipiche38, 2022-09-07:
thanks , I think I got my answer.

pipiche38, 2022-09-08:
I still wonder, so if source_routing is enabled, then it is the bellows which is setting the route to be used. So in my initial case, that would mean, that bellows was not able to find the right route as it never succeed.

The question, is more when source_routing should be enabled , and if such type of issue occured, what can we do ?

MattWestb, 2022-09-08:
Then the coordinator is not having source routing enabled its trying finding the rout to the device that is need talking with by broadcasting router discovery to find the rout to the device.
If its not finding the route its finito with uni cast until the device is sending somthing that can helping the coordinator finding it.

So likely you is having one very bad router in your system that is not working well as router like one OSRAM plug.

Source routing can helping the mesh network then dont need finding the route every time doing unicasts but it cant fixing very bad routers in the network only routing traffic around them if its possible.

pipiche38, 2022-09-09:
Is the route_sourcing information store on the zigpy database , when source_routing is enabled ?

closed 2022-11-05 by pipiche38