zigpy/bellows #543: Bellows overrides firmware settings with conservative values

tube0013, 2023-04-02:
Currently bellows overrides firmware settings for a lot of ezsp settings to a very conservative settings. I think historically this was to combat poor firmware. But with newer devices based on the MG12/MG21/MG24 these higher fw settings are ignored unless configured in configuration.yaml.

This applies to the [default settings from the Nabu Casa built firmware](https://github.com/NabuCasa/silabs-firmware/wiki/Zigbee-EmberZNet-NCP-firmware-configuration) which I also use with TubesZB EFR32 devices. 

For example fw built supporting 200 source routes is set to a much lower value:

```
2023-04-02 10:19:51.067 DEBUG (MainThread) [bellows.ezsp.protocol] Application frame received setConfigurationValue: [<EzspStatus.SUCCESS: 0>]
2023-04-02 10:19:51.068 DEBUG (MainThread) [bellows.ezsp.protocol] Send command setConfigurationValue: (<EzspConfigId.CONFIG_SOURCE_ROUTE_TABLE_SIZE: 26>, 16)
```

These defaults handicap the ncp unless a user is technically proficient enough to configure all the setting in configuration.yaml:

[image]

How bellows handles defaults ezsp settings vs actual fw settings needs to be re-evaluated - also considering the change in 7.x (I think) that runtime configuration can only decrease settings not increase them.

MattWestb, 2023-04-02:
Some of the custom setting is unrealistic for normal users like `200 source routes` also many users with 200 and more device is disabling source routing then many routers is not playing well with it and making problem and also direct children is making more problems then good in many cases.
The default config shall being OK also for users like we is have one this week with one EM357 on original EZSP 6.0.0.0 and if need tweekeing one EFR32MG2X with 3M RAM its one advance user case and shall being so but documented in the wiki or discussion.

tube0013, 2023-04-02:
99% of the zha user base isn't going to be checking the zigpy wiki for config settings.

I made a poor assumption a while back (removed all ezsp settings from config) and could not figure out why source routing enabled after years of stable use was now rubbish! I had made assumption that if I used the same settings in fw Nabu Casa was shipping with skyconnect and yellow I'd be okay.

closed 2023-04-18 by puddly

tube0013, 2023-04-18:
Thanks @puddly !

PR #550 (2023-04-18): Prefer firmware config defaults and only grow them when necessary
Default config is moved out of the protocol-specific definitions and into the main bellows application. With this change, two types of config are introduced:

 1. overrides: this config will always be set. For example, `CONFIG_TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S` will always be set to an exact value.
 2. minimum config: this config will only be changed if the current value is too low. Otherwise, the default value (set in firmware) is left alone.

Any of this behavior can be overridden by either overriding the config with a [... cut]