zigpy/bellows #563: EZSP v12 (EZSP PROTOCOL VERSION 12) support in bellows?

Hedda, 2023-06-09:
FYI, EZSP protocol version has been incremented to EZSP v12 in Silicom Labs Gecko SDK v4.3 (4.3.0.0) / Zigbee EmberZNet v7.3 (7.3.0.0):

https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.3.0

https://www.silabs.com/documents/public/release-notes/emberznet-release-notes-7.3.0.0.pdf

https://www.silabs.com/wireless/zigbee?tab=software#software

0x0C hexadecimal = 12 in decimal.

https://github.com/SiliconLabs/gecko_sdk/blob/gsdk_4.3/protocol/zigbee/app/util/ezsp/ezsp-protocol.h#L33

Silicon Labs EmberZNet 7.3.0.0 makes Silabs  Zigbee stack compatible Zigbee Protocol Specification Revision 23 / R23 (Zigbee PRO 2023), but users should be warned against flashing EZSP v12 firmware as of yet as it is not backwards compatible with EZSP v11 or older EZSP.

**Zigbee**
- Zigbee R23 compliance, with these Security enhancements among others:
  - Dynamic link key negotiation
  - Device interview to query devices before they are allowed to join
  - Trust Center Swap Out to replace an existing Trust Center with a new one
  - Frame Counter Synchronization
- Zigbee Direct Device (ZDD) support for:
  - Onboarding/commissioning
  - Communication to all Zigbee devices without a hub (Alpha), using Bluetooth LE
- Zigbee Smart Energy 1.4a compliance (Alpha)
- Enhancements to Zigbee Green Power (ZGP) APIs
- New Zigbee Security upgrade component for moving encryption keys from cleartext NVM3 tokens into secure storage

**Multiprotocol**
- Zigbee/OpenThread Concurrent Multiprotocol SoC sample app
- CPC GPIO expander module
- Zigbeed enhancements

https://www.silabs.com/blog/zigbee-pro-2023-spec-released-increases-security

https://csa-iot.org/newsroom/zigbee-pro-2023-improves-overall-security-while-simplifying-experience/

https://csa-iot.org/wp-content/uploads/2023/04/05-3474-23-csg-zigbee-specification-compressed.pdf

https://github.com/zigpy/zigpy/issues/1211

MattWestb, 2023-06-09:
>Silicon Labs EmberZNet 7.3.0.0 makes Silabs Zigbee stack compatible Zigbee Protocol Specification Revision 23 / R23 (Zigbee PRO 2023), but users should be warned against flashing EZSP v12 firmware as of yet as it is not backwards compatible with EZSP v11 or older EZSP.

Thais is the first part of R23 implementation and is default disabled in the GSDK so it shall not being any problem and if it was its falling back to old ZB3 paring so shall not being any problems only of you is finding one buggy R23 device paring with the coordinator it can.

The more important things is the new API that is being implanted and also is it written V12 is not backward compatible with older EZSP like all have being if configuring it right ?

Hedda, 2023-06-19:
FYI, [darkxst](https://github.com/darkxst) begun work adding support for multiple hardware manifests to allow override configs for automatic firmware builds:

https://github.com/darkxst/silabs-firmware-builder/blob/manifestjs/manifests

As such he is the first to have firmware builds for Silicon Labs EmberZNet 7.3 (though untested since they are automatically built):

https://github.com/darkxst/silabs-firmware-builder/tree/manifestjs/firmware_builds

MattWestb, 2023-06-19:
> FYI, [darkxst](https://github.com/darkxst) begun work adding support for multiple hardware manifests to allow override configs for automatic firmware builds:
> 
> https://github.com/darkxst/silabs-firmware-builder/blob/manifestjs/manifests
> 
> As such he is the first to have firmware builds for Silicon Labs EmberZNet 7.3 (though untested since they are automatically built):
> 
> https://github.com/darkxst/silabs-firmware-builder/tree/manifestjs/firmware_builds
> 
> PS: He is also offering expåerimental builds for the same EmberZNet version and hardware but with different baud rate speeds.

I was thinking this is the first posted ones https://github.com/NabuCasa/silabs-firmware/commit/8b4ee33e44140a6cad7e8f01b84ed6c1021b0529

puddly, 2023-06-22:
Implemented by #567.

closed 2023-06-22 by puddly