zigpy/bellows #592: [bellows.ezsp] NCP entered failed state. Requesting APP controller restart

Danieldiazi, 2023-11-14:
Hi

For the past several days, I have a lot of messages " [bellows.ezsp] NCP entered failed state. Requesting APP controller restart" and also "ERROR (MainThread) [bellows.uart] Lost serial connection: ConnectionResetError('Remote server closed connection')"

In my case, I have a Ethernet-connected gateway. Moreover, If the problem is with that device, I changed it from another new one, same model and yet I still receive those messages. 

Any solution or idea about this problem?

puddly, 2023-11-14:
What specific gateway? Can enable ZHA debug logging and post a log of this happening?

Danieldiazi, 2023-11-14:
> What specific gateway? Can enable ZHA debug logging and post a log of this happening?

Hi!

I've Silvercrest Zigbee Lidl (I followed this guide in the past: https://paulbanks.org/projects/lidl-zigbee/) 

Thanks for your answer

I will try to get log file

Danieldiazi, 2023-11-14:
> What specific gateway? Can enable ZHA debug logging and post a log of this happening?
And the debug log file:
[2023-11-14 095034.695 DEBUG - NCP.txt](https://github.com/zigpy/bellows/files/13350138/2023-11-14.095034.695.DEBUG.-.NCP.txt)

puddly, 2023-11-14:
Are you 100% sure nothing else (addons, etc.) is trying to connect to the gateway? Any other connection attempts will disconnect the active connection.

Danieldiazi, 2023-11-14:
> Are you 100% sure nothing else (addons, etc.) is trying to connect to the gateway? Any other connection attempts will disconnect the active connection.

Hi, I have container version of HA, and I've ZHA integration. Maybe is a check I have into "Uptime Kuma" ¿? With "uptime kuma" I check the TCP connection on 8888 port in order to get notified If gateway is down. I'll disable it.

My check is running each 5 minutes and I've this message error each 5 minutes. I may have shot myself in the foot...

puddly, 2023-11-14:
Disable the check and see what happens. The error says that the TCP server is disconnecting ZHA, which is consistent with what you describe.

Danieldiazi, 2023-11-14:
14 minutes later and I don't get any new message about that problem.  Yes, I think this check was the problem. Sorry.Thanks for you great support.

MattWestb, 2023-11-14:
Standard setting for the serial GW is only allowing one connection so opening the port with other program is disconnecting the first ones. Its possible overriding with some version in the config but better not having any other program making strange things with the Zigbee module.

Danieldiazi, 2023-11-14:
> Standard setting for the serial GW is only allowing one connection so opening the port with other program is disconnecting the first ones. Its possible overriding with some version in the config but better not having any other program making strange things with the Zigbee module.

Yes, When I've created the check using"Uptime Kuma" I didn't realize it would cause this problem. I replaced this TCP check and now I have a Ping check.  TCP was more specific, but no problem, The important is that GW and Zigbee network work fine!

Thanks

Danieldiazi, 2023-11-16:
After two days, the message has disappeared, confirming that the problem has been solved. Thank you for your great support and again, sorry for my mistake!. I close the issue.

closed 2023-11-16 by Danieldiazi