zigpy/bellows #608: cli fails with AttributeError `zigpy.config` has no attribute `CONF_FLOW_CONTROL`

kucharskim, 2024-01-25:
```
$ bellows -b 115200 -d /dev/ttyU0 form
Traceback (most recent call last):
  File "/tmp/XXX-env/bin/bellows", line 8, in <module>
    sys.exit(main())
  File "/tmp/XXX-env/lib/python3.10/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/tmp/XXX-env/lib/python3.10/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/tmp/XXX-env/lib/python3.10/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/tmp/XXX-env/lib/python3.10/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/tmp/XXX-env/lib/python3.10/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/tmp/XXX-env/lib/python3.10/site-packages/click/decorators.py", line 33, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/tmp/XXX-env/lib/python3.10/site-packages/bellows/cli/application.py", line 45, in form
    return util.app(inner, app_startup=False, extra_config=extra_config)(ctx)
  File "/tmp/XXX-env/lib/python3.10/site-packages/bellows/cli/util.py", line 75, in inner
    loop.run_until_complete(async_inner(*args, **kwargs))
  File "/usr/local/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/tmp/XXX-env/lib/python3.10/site-packages/bellows/cli/util.py", line 55, in async_inner
    zigpy_conf.CONF_FLOW_CONTROL: ctx.obj["flow_control"],
AttributeError: module 'zigpy.config' has no attribute 'CONF_FLOW_CONTROL'. Did you mean: 'CONF_DEVICE_FLOW_CONTROL'?
```

kucharskim, 2024-01-25:
I think below helps, still testing...
```
--- /tmp/XXX-env/lib/python3.10/site-packages/bellows/cli/util.py.orig  Thu Jan 25 08:55:02 2024
+++ /tmp/XXX-env/lib/python3.10/site-packages/bellows/cli/util.py       Thu Jan 25 09:25:02 2024
@@ -52,7 +52,7 @@
             zigpy_conf.CONF_DEVICE: {
                 zigpy_conf.CONF_DEVICE_PATH: ctx.obj["device"],
                 zigpy_conf.CONF_DEVICE_BAUDRATE: ctx.obj["baudrate"],
-                zigpy_conf.CONF_FLOW_CONTROL: ctx.obj["flow_control"],
+                zigpy_conf.CONF_DEVICE_FLOW_CONTROL: ctx.obj["flow_control"],
             },
             zigpy_conf.CONF_DATABASE: ctx.obj["database_file"],
         }
```

Hedda, 2024-03-04:
@kucharskim For reference, the answer to this will probably be similar/same to answering replies here -> https://github.com/zigpy/zigpy/discussions/1342

pepijndevos, 2024-03-30:
It seems it's really poorly documented how to use zigpy. Bellows CLI doesn't work, zigpy-cli doesn't have commands to actually do much beyond form a network and directs back to bellows to do packet capture, and there isn't much of any documentation how to use the zigpy library directly.

Hedda, 2024-03-30:
I think that it is today you need to look the ZHA integration component code inside the Home Assistent core, but it now seems like that might change in the future. Check out this discussion in the main zigpy project repository:

https://github.com/zigpy/zigpy/discussions/1207

as well as the related requests in this open issue that is asking for application examples:

https://github.com/zigpy/zigpy/issues/1087

WhiteHexagon, 2024-05-06:
I made the suggested change above, and the 'form' seems to work :) even allowing me to set a channel (because I had warning about busy channel 25).  Also the 'permit' and 'info' seem to work.  But the 'devices' never exits.  2 endpoints are listed IAS_CONTROL, and then CONTROLLER.  (this is a sonoff USB dongle plus stick, Model ZBDongle-E) .  For the Controller it shows input clusters: Basic(0) and at that point never exits.  I've been through the code best I can, and it seems like it might be scheduler locking, still looking...  Anyone got further than that please?  Oh and I tried the suggested zigpy cli but seems not to support any cli control of devices.

puddly, 2024-05-06:
As mentioned above, the bellows CLI isn't maintained at this point because it includes a lot of duplicate functionality from zigpy-cli, which uses the interfaces. Other than packet capture (which can be abstracted into a generic method within zigpy), everything else is radio-agnostic and should be supported within zigpy-cli.

If you want to make a PR adding the functionality you want to zigpy-cli, that would be much appreciated. The basic application startup sequence is shown in the `permit` command: https://github.com/zigpy/zigpy-cli/blob/ab58a1b0c3d6e4670ab6af105e0fadc3310da2aa/zigpy_cli/radio.py#L167

WhiteHexagon, 2024-05-06:
Probably it is already in zigpy-cli, I just cant seem to find it, or any examples.  Just basic stuff to read and write device attribures via zcl, and something equivalent to the bellows 'devices' command.

closed 2024-07-23 by puddly