zigpy/bellows #622: Backup running firmware (getting .ebl or .hex)

fakuivan, 2024-05-01:
Is it possible to dump the running firmware into something that can be restored to later? I'm messing around with a EM3581 and it'd be cool to be able to make a backup before flashing something onto it.

puddly, 2024-05-01:
Not without connecting a SWD debugger (unless you can figure out how to execute code from RAM via EZSP commands!). The bootloader doesn't provide this functionality either.

Is your EM3581 from a HUSBZB-1? I do have a full flash dump of an original HUSBZB-1 that can be converted into an EBL firmware with a bit of effort. There's little point in restoring it though unless you specifically want to run an ancient version of EmberZNet.

fakuivan, 2024-05-01:
This is for a cheap hub being sold as an orvibo rebrand thing. I have more info about it here: https://github.com/fakuivan/orvibo-gynoid-zigbee-hub-hack . I got so far as to strip the logic on the SOC and connecting it directly to HA via a serial to TCP relay.

Say I have a SWD debugger, jtag or whatever. What would be the steps for dumping it? Also is I wasn't able to find any router firmware, all the ones out there are just for controller functionality, is there any chance you know where to get that?

MattWestb, 2024-05-01:
Dis you find some photos if the connected Zigbee chip so we can getting the pin / pads its using ?
If yes then we can looking id Gary can cooking one firmware for it (EM8X need payed licence for compiler and much knowledge).

Dumping the chip (all flash consent also boot loader and saved settings) shall not being any problem can being done with one simple SWD probe and GDB that is free.

fakuivan, 2024-05-01:
I've added board pictures to the repo [here](https://github.com/fakuivan/orvibo-gynoid-zigbee-hub-hack/tree/f9231534dce83d522caec9e6c28aeb153c0e267c/res/images) the pins used for UART are PB1 and PB2, baud rate is 57600. If you want we can continue in https://github.com/fakuivan/orvibo-gynoid-zigbee-hub-hack/issues/1

closed 2024-05-01 by puddly