zigpy/bellows #627: setPreinstalledCbkeData283k1 does not work

kazkansouh, 2024-06-07:
Hello, thanks for making such a nice project :clap: 

I was playing around with the project while trying to set CBKE 283k1 data over EZSP and noticed the below call would fail with `EmberStatus.INVALID_CALL: 112`.

https://github.com/zigpy/bellows/blob/09cf7ce258084b2aeaac6fccaa2fc1fd66c242c4/bellows/ezsp/v8/commands.py#L514-L522

After some searching, came across the following page in Silabs community forum: https://community.silabs.com/s/article/the-difference-when-setting-the-device-s-ca-public-key-and-283k1-curve-ca-public

Once updating this definition as below I was able to use the `setValue` and `savePreinstalledCbkeData283k1` calls as outlined in the above link to set the CBKE data (and verified by joining it to a coordinator which required CBKE).

```python
    "savePreinstalledCbkeData283k1": (
        0x00ED,
        (),
        (t.EmberStatus,),
    ),
```

I am happy to raise a PR to make this change if it would be useful?

puddly, 2024-06-07:
Of course! I don't think I've ever used the ECC features of EmberZNet so this part of the command set exists untested. If you find any mistakes, please feel free to make a PR, it would be much appreciated.

puddly, 2024-07-23:
This was fixed (by accident) in https://github.com/zigpy/bellows/pull/636, I noticed the SDK source differed for this command.

closed 2024-07-23 by puddly