zigpy/bellows #640: KeyError: 'policyId' with bellows 0.40.0

pipiche38, 2024-07-24:
> source_traceback: Object created at (most recent call last):
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 232, in __step
>     result = coro.send(None)
>   File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 128, in start_zigpy_task
>     await worker_loop(self)
>   File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 384, in worker_loop
>     await process_incoming_command(self, command_to_send),
>   File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 390, in process_incoming_command
>     await dispatch_command(self, data)
>   File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 429, in dispatch_command
>     await _permit_to_joint(self, data)
>   File "/var/lib/domoticz/plugins/Domoticz-Zigbee/Classes/ZigpyTransport/zigpyThread.py", line 490, in _permit_to_joint
>     await app.permit(time_s=duration, node=target_router)
>   File "/usr/local/lib/python3.10/site-packages/bellows/zigbee/application.py", line 831, in permit
>     self.create_task(self._ezsp.pre_permit(time_s), "pre_permit")
>   File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 96, in create_task
>     task = asyncio.get_running_loop().create_task(target, name=name)
> Traceback (most recent call last):
>   File "/usr/lib/python3.10/asyncio/tasks.py", line 232, in __step
>     result = coro.send(None)
>   File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/v8/__init__.py", line 43, in pre_permit
>     await self.setPolicy(
>   File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/protocol.py", line 68, in command
>     data = self._ezsp_frame(name, *args, **kwargs)
>   File "/usr/local/lib/python3.10/site-packages/bellows/ezsp/protocol.py", line 52, in _ezsp_frame
>     data = t.serialize_dict(args, kwargs, tx_schema)
>   File "/usr/local/lib/python3.10/site-packages/bellows/types/__init__.py", line 20, in serialize_dict
>     return b"".join(t(params[k]).serialize() for k, t in schema.items())
>   File "/usr/local/lib/python3.10/site-packages/bellows/types/__init__.py", line 20, in <genexpr>
>     return b"".join(t(params[k]).serialize() for k, t in schema.items())
> KeyError: 'policyId'

puddly, 2024-07-24:
Fixed by https://github.com/zigpy/bellows/pull/639

pipiche38, 2024-07-24:
will wait for 40.1 or 41.0 no issue

closed 2024-07-24 by puddly