zigpy/bellows #654: Controller doesnt reconnect after serial connection lost.

ryanwinter, 2024-12-03:
Previously in home assistant 2024.10.3, I would see the following errors, but everything recovered fine:
```
2024-12-03 11:58:02.028 ERROR (bellows.thread_0) [bellows.uart] Lost serial connection: SerialException('device reports readiness to read but returned no data (device disconnected or multiple access on port?)')
2024-12-03 11:58:02.029 ERROR (MainThread) [bellows.ezsp] NCP entered failed state. Requesting APP controller restart
```

But now (home assistant 2024.11), it seems that if the serial connection is lost, it doesnt appear to reestablish? I think it might be related to #633 

This issue is also being tracked at in [Home Assistant Core](https://github.com/home-assistant/core/issues/130548).

puddly, 2024-12-03:
Can you attach a full ZHA debug log of it failing to reconnect?

erkr, 2024-12-04:
> Can you attach a full ZHA debug log of it failing to reconnect?

The are several logs in the references issue.

puddly, 2024-12-04:
@erkr Your logs seem to specifically omit the logging I'm interested in. I need to see if this occurs with other people because reconnection logic is working as far as I'm able to test with SiLabs radios.

erkr, 2024-12-04:
@puddly pity they don't provide what you need. I was able to return my ZBDongle-E and reverted to my Conbee2. Less range but stable

ryanwinter, 2024-12-07:
I'm still running ha 2024.10.3 without any issues, I'll try and roll forward again and capture more logs but unfortunately I'm away into early Jan.

ktrask, 2024-12-07:
I have the same issue as described in the bug, that the serial connection is not reconnecting. I hope my logs are helpful. This error is most noticable: `TypeError: object NoneType can't be used in 'await' expression`

Logs:

```
Failed to send request: ApplicationController is not running

Logger: py.warnings
Quelle: /usr/local/lib/python3.13/warnings.py:110
Erstmals aufgetreten: 09:01:55 (1 Vorkommnisse)
Zuletzt protokolliert: 09:01:55

/usr/local/lib/python3.13/asyncio/base_events.py:746: RuntimeWarning: coroutine 'SerialProtocol.disconnect' was never awaited self._ready.clear()

Logger: zigpy.application
Quelle: /usr/local/lib/python3.13/site-packages/zigpy/application.py:427
Erstmals aufgetreten: 08:32:43 (2 Vorkommnisse)
Zuletzt protokolliert: 08:35:57

Failed to disconnect from radio
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 425, in shutdown
    await self.disconnect()
  File "/usr/local/lib/python3.13/site-packages/bellows/zigbee/application.py", line 459, in disconnect
    await self._ezsp.disconnect()
  File "/usr/local/lib/python3.13/site-packages/bellows/ezsp/__init__.py", line 193, in disconnect
    await self._gw.disconnect()
TypeError: object NoneType can't be used in 'await' expression

Logger: bellows.thread
Quelle: components/zha/helpers.py:761
Erstmals aufgetreten: 08:32:43 (2 Vorkommnisse)
Zuletzt protokolliert: 08:35:57

Attempted to use a closed event loop

Logger: zigpy.application
Quelle: /usr/local/lib/python3.13/site-packages/zigpy/application.py:663
Erstmals aufgetreten: 08:32:43 (1 Vorkommnisse)
Zuletzt protokolliert: 08:32:43

Watchdog failure
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 661, in _watchdog_loop
    await self.watchdog_feed()
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 647, in watchdog_feed
    await self._watchdog_feed()
  File "/usr/local/lib/python3.13/site-packages/bellows/zigbee/application.py", line 925, in _watchdog_feed
    current_counters = await self._ezsp.read_counters()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bellows/ezsp/v4/__init__.py", line 191, in read_counters
    (res,) = await self.readCounters()
             ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bellows/ezsp/protocol.py", line 126, in command
    await self._gw.send_data(data)
  File "/usr/local/lib/python3.13/site-packages/bellows/uart.py", line 31, in send_data
    await self._transport.send_data(data)
  File "/usr/local/lib/python3.13/site-packages/bellows/ash.py", line 708, in send_data
    await asyncio.shield(
    ...<6 lines>...
    )
  File "/usr/local/lib/python3.13/site-packages/bellows/ash.py", line 656, in _send_data_frame
    await ack_future
RuntimeError: Connection has been closed

Logger: homeassistant
Quelle: /usr/src/homeassistant/homeassistant/runner.py:112
Erstmals aufgetreten: 08:32:43 (1 Vorkommnisse)
Zuletzt protokolliert: 08:32:43

Error doing job: Fatal write error on serial transport (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/serial/serialposix.py", line 621, in write
    n = os.write(self.fd, d)
OSError: [Errno 19] No such device

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/serial_asyncio_fast/__init__.py", line 310, in _write_data
    n = self._serial.write(data)
  File "/usr/local/lib/python3.13/site-packages/serial/serialposix.py", line 655, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 19] No such device
```

luca-mg, 2024-12-29:
Also rolled back to 2024.10.3