zigpy/bellows #661: Failure to initialize database

pedrocr, 2024-12-25:
I've tried the following example code to try and get events from my network:

```python
#!/usr/bin/env python3
import asyncio

# There are many different radio libraries but they all have the same API
from bellows.zigbee.application import ControllerApplication

class MainListener:
    """
    Contains callbacks that zigpy will call whenever something happens.
    Look for `listener_event` in the Zigpy source or just look at the logged warnings.
    """

    def __init__(self, application):
        self.application = application

    def device_joined(self, device):
        print(f"Device joined: {device}")

    def attribute_updated(self, device, cluster, attribute_id, value):
        print(f"Received an attribute update {attribute_id}={value}"
              f" on cluster {cluster} from device {device}")

async def main():
    app = ControllerApplication(ControllerApplication.SCHEMA({
        "database_path": "./zigbee.db",
        "device": {
            "path": "/dev/ttyUSB0",
        }
    }))

    listener = MainListener(app)
    app.add_listener(listener)

    await app.startup(auto_form=True)

    # Permit joins for a minute
    await app.permit(60)
    await asyncio.sleep(60)

    # Just run forever
    await asyncio.get_running_loop().create_future()

if __name__ == "__main__":
    asyncio.run(main())
```

This failed with this error:

```
$ ./zigpy_listener 
Traceback (most recent call last):
  File "./zigpy_listener", line 45, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "./zigpy_listener", line 25, in main
    app = ControllerApplication(ControllerApplication.SCHEMA({
  File "/usr/local/lib/python3.10/dist-packages/bellows/zigbee/application.py", line 84, in __init__
    super().__init__(config)
  File "/usr/local/lib/python3.10/dist-packages/zigpy/application.py", line 70, in __init__
    self._config = self.SCHEMA(config)
  File "/usr/local/lib/python3.10/dist-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.10/dist-packages/voluptuous/schema_builder.py", line 549, in validate_dict
    return base_validate(path, data.items(), out)
  File "/usr/local/lib/python3.10/dist-packages/voluptuous/schema_builder.py", line 330, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/local/lib/python3.10/dist-packages/voluptuous/schema_builder.py", line 779, in validate_callable
    return schema(data)
  File "/usr/local/lib/python3.10/dist-packages/voluptuous/schema_builder.py", line 205, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.10/dist-packages/voluptuous/schema_builder.py", line 549, in validate_dict
    return base_validate(path, data.items(), out)
  File "/usr/local/lib/python3.10/dist-packages/voluptuous/schema_builder.py", line 330, in validate_mapping
    cval = cvalue(key_path, value)
  File "/usr/local/lib/python3.10/dist-packages/voluptuous/schema_builder.py", line 590, in validate_sequence
    cval = validate(index_path, value)
  File "/usr/local/lib/python3.10/dist-packages/voluptuous/schema_builder.py", line 779, in validate_callable
    return schema(data)
  File "/usr/local/lib/python3.10/dist-packages/zigpy/config/validators.py", line 124, in cv_ota_provider
    provider_type = obj.get(zigpy.config.CONF_OTA_PROVIDER_TYPE)
AttributeError: 'Ledvance' object has no attribute 'get'
```

I found a similar thing in the zboss library here:

puddly, 2024-12-25:
Correct. You need to remove `ControllerApplication.SCHEMA` from your code.

pedrocr, 2024-12-25:
Thanks! That does work. Closing now and here's a minimal example that actually works and receives events:

```python
#!/usr/bin/env python3
import asyncio

# There are many different radio libraries but they all have the same API
from bellows.zigbee.application import ControllerApplication

class MainListener:
    """
    Contains callbacks that zigpy will call whenever something happens.
    Look for `listener_event` in the Zigpy source or just look at the logged warnings.
    """

    def __init__(self, application):
        self.application = application

    def device_joined(self, device):
        print(f"Device joined: {device}")

    def attribute_updated(self, device, cluster, attribute_id, value):
        print(f"Received an attribute update {attribute_id}={value}"
              f" on cluster {cluster} from device {device}")

async def main():
    app = ControllerApplication({
        "database_path": "./zigbee.db",
        "device": {
            "path": "/dev/ttyUSB0",
        }
    })

    listener = MainListener(app)
    app.add_listener(listener)

    await app.startup(auto_form=True)

    # Just run forever
    await asyncio.get_running_loop().create_future()

if __name__ == "__main__":
    asyncio.run(main())
```

pedrocr, 2024-12-25:
Thanks for your help

closed 2024-12-25 by pedrocr