zigpy/bellows #663: Unknown aplication frame received

ReinoutHeeck, 2025-01-22:
In Home Assistant UI all the Zigbee devices regularly become unavailable, this situation then rectifies itself after a couple of seconds.

The home-assistant.log mentions unkown application frames at that time (see below).

puddly, 2025-01-22:
Can you post the whole debug log? The warning alone doesn't say much, unfortunately.

ReinoutHeeck, 2025-01-22:
@puddy this is from my main log (that is riddled with messages from another integration that is misbehaving in my system).

I'll turn on debug logging on the Zigbee integration and post the results here -> stay tuned...

ReinoutHeeck, 2025-01-22:
@puddy here's the debug log, 

ignore the fritz integration errors - they are reported at https://github.com/home-assistant/core/issues/136127

earlier zigbee quirk warnings were already reported at https://github.com/zigpy/zha-device-handlers/issues/3750

Thank you!

[home-assistant_zha_2025-01-22T17-06-54.534Z.log](https://github.com/user-attachments/files/18509263/home-assistant_zha_2025-01-22T17-06-54.534Z.log)

ReinoutHeeck, 2025-01-22:
Hmm,  I saw my UI 'go grey' regarding all zigbee devices, hence stopped debug logging and uploaded the result here.

Now I'm reading a bit through it I cannot find the phrase 'Unknown application frame' in it :-/

I'll turn debug logging on again, see if it re-occurs overnight...

ReinoutHeeck, 2025-01-24:
I captured several 'Unknown application frame' events with debugging mode on. The main log file is 6.7GB now so I created some excerpts.

Here's a grep, the first (longer) occurrences were from before enabling debug logging:
```sh
grep 'Unknown application frame' home-assistant.log.1 
2025-01-21 20:17:49.804 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'004500040102070101000100001d74b9d57affff0c08a90a000025e40300000000' (b'4590ff004500040102070101000100001d74b9d57affff0c08a90a000025e40300000000').  This is a bug!
2025-01-21 20:17:50.046 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'004500040102070101000100001e88bed57affff0c08a90a000025e40300000000' (b'4590ff004500040102070101000100001e88bed57affff0c08a90a000025e40300000000').  This is a bug!
2025-01-21 20:17:50.233 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'004500040102070101000100001f88bed57affff0c08a90a000025e40300000000' (b'4590ff004500040102070101000100001f88bed57affff0c08a90a000025e40300000000').  This is a bug!
2025-01-21 20:17:50.448 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'004500040102070101000100002088bed57affff0c08a90a000025e40300000000' (b'4590ff004500040102070101000100002088bed57affff0c08a90a000025e40300000000').  This is a bug!
2025-01-21 20:17:50.493 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'003f06fdff00000100000000010000e7030000' (b'4590ff003f06fdff00000100000000010000e7030000').  This is a bug!
2025-01-22 06:32:35.122 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 18:59:18.030 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 19:04:54.226 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 19:10:28.962 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 19:16:04.578 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 19:45:26.909 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 19:51:02.232 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 20:31:33.090 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 21:24:37.770 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 21:35:49.312 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 21:48:23.341 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 22:06:33.167 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-22 23:42:56.451 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-23 01:08:09.886 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-23 02:01:14.278 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-23 02:20:47.772 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'005300' (b'0780ff005300').  This is a bug!
2025-01-24 01:33:51.498 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'0052000000' (b'0f80ff0052000000').  This is a bug!
2025-01-24 02:11:35.767 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'0052000000' (b'0f80ff0052000000').  This is a bug!
2025-01-24 04:04:45.674 WARNING (MainThread) [bellows.ezsp.protocol] Unknown application frame 0x00FF received: b'0052000000' (b'0f80ff0052000000').  This is a bug!
```

I split the occurrences up to create files below the 25MB upload limit:
```sh
grep 'Unknown application frame' home-assistant.log.1 -B 10000 -A 100 > home-assistant-matches.log
csplit -f home-assistant.log. home-assistant-matches.log '/--/' '{*}'
```

Note that all of the resulting files with debug enabled have exactly one line```'[bellows.uart] Connection lost: None``` preceding the Unknown frame event!

My Zigbee integration is re-initializing every couple of minutes since the last software upgrade and I recall having issues probably since the first January update. Now that I have debug information I understand it is due to connection loss with the coordinator radio (maybe that log line should be a more visible WARNING line instead of DEBUG?).

I find connection loss *every* couple of minutes strange since this is an ethernet connected coordinator and it worked fine before newyear. I'll see if downgrading to a December version stabilizes my HA behavior...

Here's  a couple of the files I created:

[home-assistant-02.log](https://github.com/user-attachments/files/18536592/home-assistant-02.log)

[home-assistant-03.log](https://github.com/user-attachments/files/18536595/home-assistant-03.log)

[home-assistant-04.log](https://github.com/user-attachments/files/18536598/home-assistant-04.log)

ReinoutHeeck, 2025-01-24:
Well this is embarrassing: after an unrelated server restarted Docker decided to revive an ancient HA container. It took me quite the sleuthing spree before I found this.

Two HA instances were fighting over the ethernet connected radio in my network the past month :-(

closed 2025-01-24 by ReinoutHeeck

puddly, 2025-01-24:
Thanks! I've updated the warning to hopefully save anyone else having this issue some time: https://github.com/zigpy/bellows/pull/667

ReinoutHeeck, 2025-01-25:
That patch looks awesome, much better than my suggestion to level up the 'connection lost' debug message.