zigpy/bellows #670: Exception when adding device using Home Assistant ZHA.

hbjorgo, 2025-03-10:
I get an exception when adding a Zigbee device to my network. When discovering the new device, it reaches the "Configuring" state in Home Assistant, but then it just hangs. The network already has multiple other devices that work fine. I try to add a [Frient SMSZB-120, HW: 5.2.0 Smoke Alarm](https://frient.com/products/intelligent-smoke-alarm/). As far as I can see, it is a Develco device. I'm running Home Assistant 2025.3.1 with the latest ZHA integration and Home Assistant Connect ZBT-1 Zigbee dongle.

```
Logger: bellows.ezsp
Source: /usr/local/lib/python3.13/site-packages/bellows/ezsp/__init__.py:533
First occurred: 3:12:16 PM (6 occurrences)
Last logged: 3:25:41 PM

Exception running handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/bellows/ezsp/__init__.py", line 531, in handle_callback
    handler(*args)
    ~~~~~~~^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/bellows/zigbee/application.py", line 493, in ezsp_callback_handler
    self._handle_frame(
    ~~~~~~~~~~~~~~~~~~^
        message_type=message_type,
        ^^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<6 lines>...
        message=message,
        ^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/bellows/zigbee/application.py", line 574, in _handle_frame
    self.packet_received(
    ~~~~~~~~~~~~~~~~~~~~^
        zigpy.types.ZigbeePacket(
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    ...<13 lines>...
        )
        ^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 1031, in packet_received
    return device.packet_received(packet)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/device.py", line 476, in packet_received
    hdr, _ = foundation.ZCLHeader.deserialize(data)
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/types/struct.py", line 319, in deserialize
    value, data = field_type.deserialize(data)
                  ~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/types/basic.py", line 199, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
```

And this one:

```
Logger: homeassistant
Source: /usr/src/homeassistant/homeassistant/runner.py:112
First occurred: 3:23:01 PM (1 occurrences)
Last logged: 3:23:01 PM

Error doing job: Task exception was never retrieved (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zigpy/device.py", line 381, in request
    return await req.result
           ^^^^^^^^^^^^^^^^
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line 68, in wrap_zigpy_exceptions
    yield
  File "/usr/local/lib/python3.13/site-packages/zha/zigbee/cluster_handlers/__init__.py", line 86, in wrapper
    return await RETRYABLE_REQUEST_DECORATOR(func)(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 96 lines cut ...]
        )
        ^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 1031, in packet_received
    return device.packet_received(packet)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/device.py", line 476, in packet_received
    hdr, _ = foundation.ZCLHeader.deserialize(data)
             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/types/struct.py", line 319, in deserialize
    value, data = field_type.deserialize(data)
                  ~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/types/basic.py", line 199, in deserialize
    raise ValueError(f"Data is too short to contain {byte_size} bytes")
ValueError: Data is too short to contain 1 bytes
2025-03-10 15:25:41.310 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7ff0576dcd60>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1741616741.291484, 'ieee': 00:15:bc:00:31:02:c5:40, 'min_update_delta': 30.0})
2025-03-10 15:25:41.311 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7ff0576dcd60>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1741616741.291484, 'ieee': 00:15:bc:00:31:02:c5:40, 'min_update_delta': 30.0}) completed
2025-03-10 15:25:41.315 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7ff0576dcd60>)
2025-03-10 15:25:41.316 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7ff0576dcd60>) completed
2025-03-10 15:25:41.318 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method execute of sqlite3.Connection object at 0x7ff0576dcd60>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1741616741.291484, 'ieee': 00:15:bc:00:31:02:c5:40, 'min_update_delta': 30.0})
2025-03-10 15:25:41.319 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method execute of sqlite3.Connection object at 0x7ff0576dcd60>, 'UPDATE devices_v13\n                    SET last_seen=:ts\n                    WHERE ieee=:ieee AND :ts - last_seen > :min_update_delta', {'ts': 1741616741.291484, 'ieee': 00:15:bc:00:31:02:c5:40, 'min_update_delta': 30.0}) completed
2025-03-10 15:25:41.321 DEBUG (Thread-3) [aiosqlite] executing functools.partial(<built-in method commit of sqlite3.Connection object at 0x7ff0576dcd60>)
2025-03-10 15:25:41.323 DEBUG (Thread-3) [aiosqlite] operation functools.partial(<built-in method commit of sqlite3.Connection object at 0x7ff0576dcd60>) completed
```

puddly, 2025-03-10:
Can you record a debug log of the whole joining process?

hbjorgo, 2025-03-10:
Here you are:
[home-assistant_zha_2025-03-10T16-23-18.517Z.log](https://github.com/user-attachments/files/19166722/home-assistant_zha_2025-03-10T16-23-18.517Z.log)